zephyrproject-rtos/zephyr #34262: Unable to find detailed documentation on pinmux driver development

Berte-Yu, 2021-04-14:
I‘m porting zephyr to a new soc(cortex-m3), 

When developing the pinmux driver, the detailed documentation cannot be found..

mnkp, 2021-04-29:
The existing pinmux driver is going to be deprecated. I think the best option for now when adding a new SoC to Zephyr is to keep pinmux driver private to the soc. I.e. place it in `soc/<arch>/<family>/` and do not provide the public pinmux driver. You can take Atmel SAM family as an example. The pinmux driver is located in `soc/arm/atmel_sam/common/soc_gpio.c`. It's used by all Atmel SAM drivers but there is no Atmel SAM pinmux driver in drirvers/pinmux/.

Berte-Yu, 2021-05-06:
Thanks,  I have ported the pinmax driver。

closed 2021-05-06 by Berte-Yu