zephyrproject-rtos/zephyr #34264: CI: twister: Add  merged report from all sub-builds to buildkite build artifacts

PerMac, 2021-04-14:
**Is your enhancement proposal related to a problem? Please describe.**
If one wants to check the status of some particular test run in the CI, they will have to check individual reports/artifacts from sub-builds (there are 20 of them). It would be more convenient to be able to have access to a single report with all the results

**Describe the solution you'd like**
After all sub-builds are executed, the resulting XML reports are merged into a single one and added to the build artifacts

**Describe alternatives you've considered**
Seems straight forward and I have no alternatives

galak, 2021-04-16:
Is there a tool we use that can merge junit xml files?

galak, 2021-04-16:
Looks like we can use junitparser

closed 2021-04-20 by galak