zephyrproject-rtos/zephyr #34265: BME280 Pressure calculation

ergo70, 2021-04-14:
case SENSOR_CHAN_PRESS:
		/*
		 * data->comp_press has 24 integer bits and 8
		 * fractional.  Output value of 24674867 represents
		 * 24674867/256 = 96386.2 Pa = 963.862 hPa
		 */
		val->val1 = (data->comp_press >> 8) / 1000U;
		val->val2 = (data->comp_press >> 8) % 1000 * 1000U +
			(((data->comp_press & 0xff) * 1000U) >> 8);
		break;

Gives 102 for val1 and 0.3 for val2 if the pressure is actually 1023 hPa, so for 4 digit pressures, the least significant digit is shifted into the fractional part val2. Is this intended behaviour?

jfischer-no, 2021-04-14:
kPA is taken as a basis here, see https://github.com/zephyrproject-rtos/zephyr/blob/master/include/drivers/sensor.h#L82
102300 Pa = 102,3 kPa

ergo70, 2021-04-14:
OK, thank you.

closed 2021-04-14 by jfischer-no