zephyrproject-rtos/zephyr #34268: The sample samples/bluetooth/mesh/ stucks in init process on disco_l475_iot

Verfuchst, 2021-04-14:
**Describe the bug**
I was trying to test the mesh sample `${ZEPHYR_BASE}/samples/bluetooth/mesh/`. After i build and flashed it on my board, it was working first by pressing the userbutton and turn off/on the led. After i rebuild and flashed it again on the board, without making changes, it stucks in the init flash process.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p auto -b disco_l475_iot1 samples/bluetooth/mesh/ -c
2. west flash

**Expected behavior**
Testing the mesh sample by pressing the userbutton and turn off/on led

**Logs and console output**
Logs with picocom 
```
*** Booting Zephyr OS build zephyr-v2.5.0-2155-gc7412bb0c46c  ***
Initializing...
[00:00:00.000,000] <inf> flash_stm32_qspi: MX2
```

**Environment (please complete the following information):**
 - OS: xUbuntu 20.04
 - Toolchain: Zephyr SDK 0.12.03,
 - c7412bb0c46cff96ac191814dadde36b45cc12f9

erwango, 2021-04-16:
@Verfuchst Did you make any specific modification before reflashing ?
I has a try on my side and it worked fine...

trond-snekvik, 2021-04-16:
Same for me. Are you able to try on a non-modified master, and pinpoint where things go wrong with a debugger? Does it get into the bluetooth enable callback?

Verfuchst, 2021-04-16:
I tried again on the master commit 36d96d2cd9b9e9f200691e34d053bb8cf52321ca 
[image]

erwango, 2021-04-16:
One thing to unlock your issue quickly would be to remove `		zephyr,flash-controller = &mx25r6435f;` so you don't use qspi anymore. Does not solve the point but could help.

Verfuchst, 2021-04-16:
I tried this with removing the line, but it did not changed anything. Now I'm also stuck in another code section.
[image]
I can try to get the same board next week and see if its something with the hardware.

carlescufi, 2021-04-22:
This happens before the Bluetooth subsystem is even initialized, so removing the Bluetooth labels.

Verfuchst, 2021-05-18:
Sorry for late respone, it wasnt possible to get as quick as possible the board, because of corona. But i tried the same scenario with the new same board even with the latest commit 4b4068c948ea4fd764df549aedb70d0f698491c3 and i still get the same situtation with stucking in the init process.
serial output:
```
*** Booting Zephyr OS build v2.6.0-rc1-181-g4b4068c948ea  ***
Initializing...
[00:00:00.000,000] <inf> flash_stm32_qspi: MX25
```
code section:
[image]

erwango, 2021-05-21:
Ok, that's really strange we're not seeing the same.

Can you have a try by removing use of qspi as flash controller on the board?
Basically, you need to:
-Remove qspi as flash controller
-Re-instantiate storage partition on flash 

This is equivalent to revert change (in the board dts file, no need to revert the full driver support)
https://github.com/zephyrproject-rtos/zephyr/pull/30864/files#diff-b334b8ae2850c29b634b8dd4678791bcd4731c3319f9fd026839e31b9a354832

Verfuchst, 2021-06-10:
Without the qspi i could run the sample, but even there, there was some weird acting. So it just felt like, its still something with the hardware and not your code. I will take a look at this, when i get the time. But for now, i would say we gonna close this issue, till someone has the same kinda problem. And if i get any news and i can say its 100% the hardware, i will leave a comment here.

erwango, 2021-06-10:
@Verfuchst Thanks for this feedback. I'm closing the issue then. Feel free to comment or reopen

closed 2021-06-10 by erwango