zephyrproject-rtos/zephyr #34269: LOG_MODE_MINIMAL BUILD error

hongshui3000, 2021-04-14:
**Describe the bug**
```
CONFIG_DEBUG=y -------------------------------------here
CONFIG_LOG=y
CONFIG_LOG_MODE_MINIMAL=y
#CONFIG_I2C_LOG_LEVEL_INF=y
```
Can not build successfully when CONFIG_DEBUG is enabled, and can build successfully when CONFIG_DEBUG=n

LOG:
```
zephyr/kernel/libkernel.a(fatal.c.obj): In function `k_sys_fatal_error_handler':
E:\work\zephyr\kernel/fatal.c:43: undefined reference to `log_1'
E:\work\zephyr\kernel/fatal.c:43: undefined reference to `log_from_user'
E:\work\zephyr\kernel/fatal.c:43: undefined reference to `log_0'
zephyr/kernel/libkernel.a(fatal.c.obj): In function `z_fatal_error':
E:\work\zephyr\kernel/fatal.c:106: undefined reference to `log_n'
E:\work\zephyr\kernel/fatal.c:106: undefined reference to `log_from_user'
E:\work\zephyr\kernel/fatal.c:106: undefined reference to `log_3'
E:\work\zephyr\kernel/fatal.c:121: undefined reference to `log_3'
E:\work\zephyr\kernel/fatal.c:121: undefined reference to `log_from_user'
E:\work\zephyr\kernel/fatal.c:121: undefined reference to `log_2'
zephyr/kernel/libkernel.a(sched.c.obj): In function `k_sched_unlock':
r\kernel/thread.c:477: undefined reference to `log_n'
```

nordic-krch, 2021-04-19:
@hongshui3000 can you be more specific? Which sample is failing to build? More details about configuration would be needed as well.

nordic-krch, 2021-05-12:
Can you point me to instruction how to setup this toolchain? What GCC version is it based? I though that it is esp toolchain and tried that but it compiles ok.

hongshui3000, 2021-05-12:
> Can you point me to instruction how to setup this toolchain? What GCC version is it based? I though that it is esp toolchain and tried that but it compiles ok.

XCC is based on old GCC 4.2.

nordic-krch, 2021-05-12:
I seems like linker/compiler behaves as if there were no garbage collection of unused stuff (-no-gc-sections) can you try to modify `subsys/logging/CMakeLists.txt` line 3 like:
`if(NOT CONFIG_LOG_MINIMAL OR CONFIG_NO_OPTIMIZATION) ` ?

hongshui3000, 2021-05-12:
@nordic-krch 
```
# SPDX-License-Identifier: Apache-2.0

if(NOT CONFIG_LOG_MINIMAL )
  zephyr_sources_ifdef(
    CONFIG_LOG
    log_list.c
    log_core.c
    log_msg.c
    log_output.c
  )

  zephyr_sources_ifdef(
    CONFIG_LOG2
    log_msg2.c
  )

  zephyr_sources_ifdef(
    CONFIG_LOG_BACKEND_UART
    log_backend_uart.c
  )

  zephyr_sources_ifdef(
    CONFIG_LOG_BACKEND_FS
    log_backend_fs.c
  )

  zephyr_link_libraries_ifdef(CONFIG_LOG_BACKEND_FS LITTLEFS)

  zephyr_sources_ifdef(
    CONFIG_LOG_CMDS
    log_cmds.c
  )

  zephyr_sources_ifdef(
    CONFIG_LOG_BACKEND_NATIVE_POSIX
    log_backend_native_posix.c
  )

  zephyr_sources_ifdef(
    CONFIG_LOG_BACKEND_XTENSA_SIM
    log_backend_xtensa_sim.c
  )

  zephyr_sources_ifdef(
    CONFIG_LOG_BACKEND_NET
    log_backend_net.c
  )

  zephyr_sources_ifdef(
    CONFIG_LOG_BACKEND_RTT
    log_backend_rtt.c
  )

  zephyr_sources_ifdef(
    CONFIG_LOG_BACKEND_SWO
    log_backend_swo.c
  )

  zephyr_sources_ifdef(
    CONFIG_LOG_MIPI_SYST_ENABLE
    log_output_syst.c
  )

  zephyr_sources_ifdef(
    CONFIG_LOG_BACKEND_ADSP
    log_backend_adsp.c
  )

  if(CONFIG_LOG_BACKEND_SPINEL)
    zephyr_library_include_directories(
	    ${ZEPHYR_BASE}/subsys/net/lib/openthread/platform/
    )
  endif()

  zephyr_sources_ifdef(
    CONFIG_LOG_BACKEND_SPINEL
    log_backend_spinel.c
  )

  if(CONFIG_LOG_DICTIONARY_SUPPORT)
    zephyr_sources(log_output_dict.c)
  endif()

else()
  zephyr_sources(log_minimal.c)
endif()

if(CONFIG_NO_OPTIMIZATIONS) ------------------------------------------->my fix
  if(CONFIG_LOG_MINIMAL)
    zephyr_sources(
      log_list.c
      log_core.c
      log_msg.c
      log_output.c
      log_msg2.c
    )
  else()
    zephyr_sources_ifndef(
      CONFIG_LOG
      log_list.c
      log_core.c
      log_msg.c
      log_output.c
    )

    zephyr_sources_ifndef(
      CONFIG_LOG2
      log_msg2.c
    )

    zephyr_sources(log_minimal.c)
  endif()
endif()--------------------------------------------------------------------------->my fix
```

nklayman, 2022-07-23:
I came across this issue on latest main branch of Zephyr with XCC based off GCC 4.2. Neither @hongshui3000's nor @nordic-krch's solutions worked for me, but I came up with this which appears to work alright:
```
// At the end of CMakeLists.txt
if(CONFIG_NO_OPTIMIZATIONS)
  if(CONFIG_LOG_MODE_MINIMAL)
    zephyr_sources(
      log_core.c
      log_msg.c
      log_mgmt.c
    )
  else()
    zephyr_sources(log_minimal.c)
  endif()
endif()
```

This worked with both minimal and non-minimal logging. This patch wasn't necessary with GCC (but it didn't break it), so maybe it should be conditional on that as well. Should I open a PR with this fix?

nordic-krch, 2022-07-25:
@nklayman please open PR, maybe you can wrap it with
`if(${ZEPHYR_TOOLCHAIN_VARIANT} STREQUAL "xcc")` and add a comment.

nklayman, 2022-07-25:
@nordic-krch I found out that building without logging enabled fails as well:
```
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /workspaces/zephyr/zephyr/build/zephyr/zephyr_pre0.map 
: && ccache /home/vscode/xcc/install/tools/RG-2017.8-linux/XtensaTools/bin/xt-xcc   zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/xtensa/core/offsets/offsets.c.obj  -Wl,-T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/workspaces/zephyr/zephyr/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/xtensa/core/libarch__xtensa__core.a  zephyr/arch/arch/xtensa/core/startup/libarch__xtensa__core__startup.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/posix/liblib__posix.a  zephyr/soc/xtensa/intel_adsp/common/libintel_adsp_common.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/timer/libdrivers__timer.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L/workspaces/zephyr/zephyr/build/zephyr  -lgcc  -lhal  zephyr/arch/common/libisr_tables.a  zephyr/soc/xtensa/intel_adsp/common/libintel_adsp_common.a  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N && cd /workspaces/zephyr/zephyr/build/zephyr && /usr/bin/cmake -E echo
zephyr/arch/arch/xtensa/core/libarch__xtensa__core.a(xtensa-asm2.c.obj):/workspaces/zephyr/zephyr/include/zephyr/sys/cbprintf.h:694: undefined reference to `z_impl_z_log_msg_runtime_vcreate'
zephyr/arch/arch/xtensa/core/libarch__xtensa__core.a(xtensa-asm2.c.obj):(.literal.xtensa_excint1_c+0x20): undefined reference to `z_log_minimal_printk'
zephyr/arch/arch/xtensa/core/libarch__xtensa__core.a(xtensa-asm2.c.obj): In function `z_log_msg_runtime_vcreate':
/workspaces/zephyr/zephyr/include/zephyr/logging/log_msg.h:75: undefined reference to `z_impl_z_log_msg_runtime_vcreate'
zephyr/arch/arch/xtensa/core/libarch__xtensa__core.a(xtensa-asm2.c.obj): In function `xtensa_excint1_c':
/workspaces/zephyr/zephyr/arch/xtensa/core/xtensa-asm2.c:262: undefined reference to `z_log_minimal_printk'
zephyr/soc/xtensa/intel_adsp/common/libintel_adsp_common.a(irq.c.obj): In function `z_log_msg_runtime_vcreate':
/workspaces/zephyr/zephyr/include/zephyr/logging/log_msg.h:75: undefined reference to `z_impl_z_log_msg_runtime_vcreate'
zephyr/kernel/libkernel.a(fatal.c.obj): In function `z_log_msg_runtime_vcreate':
/workspaces/zephyr/zephyr/include/zephyr/logging/log_msg.h:75: undefined reference to `z_impl_z_log_msg_runtime_vcreate'
zephyr/kernel/libkernel.a(thread.c.obj): In function `z_log_msg_runtime_vcreate':
/workspaces/zephyr/zephyr/include/zephyr/logging/log_msg.h:75: undefined reference to `z_impl_z_log_msg_runtime_vcreate'
zephyr/kernel/libkernel.a(sched.c.obj): In function `z_log_msg_runtime_vcreate':
/workspaces/zephyr/zephyr/include/zephyr/logging/log_msg.h:75: undefined reference to `z_impl_z_log_msg_runtime_vcreate'
ninja: build stopped: subcommand failed.
```

I'm not sure how to solve this, should I open a new issue that covers all the failing scenarios I've found? Happy to make a PR, but I think it makes sense to fix this as well first.

hongshui3000, 2022-07-26:
When using xcc to compile the code, if you choose not to optimize, then the log code needs to be fully compiled to avoid errors

closed 2022-10-17 by github-actions[bot]