zephyrproject-rtos/zephyr #34270: NVS read after consecutive restarts.

RenierVDWBluerain, 2021-04-14:
**Describe the bug**
I'm using NRF52840 dongle and are trying to get NVS flash to work. I've followed the examples on my custom code but are experiencing the following issue. When NVS write are completed in a button press interrupt I restart the board. On bootup NVS_read only reads the data successfully after the first restart. For consecutive restarts NVS_read no longer reads any data. Is this normal behaviour? If so how do I fix it so that it reads after any number of consecutive restarts.

Laczen, 2021-04-15:
@RenierVDWBluerain, thanks for creating the issue.

The behavior you are describing should not happen. Could you give some more detail on how this exactly happens (code, platform, ....).

Laczen, 2021-04-23:
I am closing this as there is insufficient information about the issue. If more information is available the author can reopen the issue.

closed 2021-04-23 by Laczen

Saranya-karan, 2024-12-16:
> **Describe the bug** I'm using NRF52840 dongle and are trying to get NVS flash to work. I've followed the examples on my custom code but are experiencing the following issue. When NVS write are [...]

is nvs working successfully in nrf52840 dongle?