zephyrproject-rtos/zephyr #34273: mqtt_publisher: Unable to connect properly on EC21 modem with bg9x driver

ycsin, 2021-04-14:
**Describe the bug**
I'm working on to port the bg9x driver for EC21. I've checked the TCP/IP AT commands of BG96 & EC21 and they are identical. I tried to run the `mqtt_publisher` example, but it doesn't work.

[image]

I traced a bit and found that it connected with my MQTT broker, but disconnected immediately after that. 

[image]

This is the reading of the AT commands going into the modem:

```
AT+QIOPEN=1,0,"TCP","server_ip_addr",1883,0,0
AT+QISEND=0,30
MQTT<zephyr_publisherAT+QICLOSE=0
```

I can see that the modem received something from the server, likely the CONNACK from the MQTT broker:
```
+QIURC: "recv",0
```

But somehow `offload_read` and `offload_recvfrom` never get called, so the application didn't progress further from connecting the broker.

**To Reproduce**
Steps to reproduce the behavior:
1. Build the `mqtt_publisher` sample on a board with EC21 modem
4. See error

**Expected behavior**
Publishes successfully to my MQTT broker.

**Impact**
Unable to progress with telemetry related task using MQTT and planned work to improve the driver.

**Logs and console output**
```
[00:00:02.000,000] <inf> modem_quectel_bg9x: Setting Modem Pins
[00:00:04.751,000] <inf> modem_quectel_bg9x: ... Done!
[00:00:04.751,000] <inf> modem_quectel_bg9x: Waiting for modem to respond
[00:00:13.792,000] <inf> modem_quectel_bg9x: Manufacturer: Quectel
[00:00:13.844,000] <inf> modem_quectel_bg9x: Model: EC21
[00:00:13.915,000] <inf> modem_quectel_bg9x: Revision: EC21EFAR06A01M4G
[00:00:13.969,000] <inf> modem_quectel_bg9x: IMEI: 867962045083204
[00:00:14.046,000] <inf> modem_quectel_bg9x: RSSI: -1000
[00:00:16.051,000] <inf> modem_quectel_bg9x: RSSI: -1000
[00:00:18.054,000] <inf> modem_quectel_bg9x: RSSI: -1000
[00:00:20.057,000] <inf> modem_quectel_bg9x: RSSI: -55
uart:~$ *** Booting Zephyr OS build zephyr-v2.5.0-2162-g365145aa971c  ***
[00:00:22.058,000] <inf> modem_quectel_bg9x: Network is ready.
[00:00:22.079,000] <inf> net_config: Initializing network
[00:00:22.079,000] <inf> net_mqtt_publisher_sample: attempting to connect:
[00:00:22.157,000] <inf> modem_quectel_bg9x: AT+QIOPEN: 0
[00:00:22.211,000] <inf> modem_quectel_bg9x: Data Receive Indication for socket: 0
[00:00:32.216,000] <inf> net_mqtt_publisher_sample: MQTT client disconnected -53
[00:00:32.286,000] <inf> modem_quectel_bg9x: AT+QIOPEN: 0
[00:00:32.328,000] <inf> modem_quectel_bg9x: Data Receive Indication for socket: 0
[00:00:42.334,000] <inf> net_mqtt_publisher_sample: MQTT client disconnected -53
[00:00:42.399,000] <inf> modem_quectel_bg9x: AT+QIOPEN: 0
[00:00:42.443,000] <inf> modem_quectel_bg9x: Data Receive Indication for socket: 0
[00:00:52.449,000] <inf> net_mqtt_publisher_sample: MQTT client disconnected -53
[00:00:52.452,000] <inf> modem_quectel_bg9x: RSSI: -61
[00:00:52.519,000] <inf> modem_quectel_bg9x: AT+QIOPEN: 0
[00:00:52.561,000] <inf> modem_quectel_bg9x: Data Receive Indication for socket: 0
```

**Environment (please complete the following information):**
 - OS: Windows 10
 - Toolchain gnuarmemb
 - Zephyr OS build zephyr-v2.5.0-2162-g365145aa971c

**Additional context**
So far I only found that it is required to increase the wait for the AT+QICLOSE in `socket_close()` or else it fails on EC21

ycsin, 2021-04-14:
@bwasim May I have you input please?

ycsin, 2021-04-14:
Okay I think the `sock->is_waiting` is always false because the driver works on a polling basis?

ycsin, 2021-04-14:
I'm guessing there's some issue with the `modem_socket_poll` and `modem_socket_data_ready`, particularly some synchronization issue with the `&cfg->sem_poll`.

jukkar, 2021-04-15:
@rlubos any ideas for the MQTT issue?

rlubos, 2021-04-15:
Well it's really likely there's some issue with `modem_socket_poll`, at least that's what the application log would suggest. If you look at the implementation of the `try_to_connect()` function in the sample:
https://github.com/zephyrproject-rtos/zephyr/blob/master/samples/net/mqtt_publisher/src/main.c#L364
the application will:
1. `mqtt_connect()`: Create a socket and send the `CONNECT` message to the broker (which likely succeeds given that you can see that the broker recognized the device)
2.  Setup fds for polling and calls `poll()` in the `wait()` function
3.  In case no data is received `mqtt_abort()` is called and the whole process repeated (which is consistent with the logs, -53 is `ECONNABORTED` assuming no newlib is used)

I only don't understand why is does it time out after 10 seconds instead of default 2 seconds, did you modify the `APP_CONNECT_TIMEOUT_MS`?

Why exactly would `modem_socket_poll` fail I don't know, I've never used this driver nor I have HW to debug it. I'd start with verifying that this line is actually executed:
https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/modem/modem_socket.c#L285
and then whether `modem_socket_data_ready()` is actually called:
https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/modem/modem_socket.c#L354

bwasim, 2021-04-15:
`modem_socket_poll` would fail if the quectel-bg9x zephyr driver (or the quectel modem) didn't receive data or failed to report data reception.

ycsin, 2021-04-15:
@rlubos I might have edited the timeout to 10 seconds, can't remember, experimented with a few things.

I did put breakpoints to two of these functions, but if I put break point to both of them, then only `modem_socket_poll` runs, `modem_socket_data_ready` likely can't run as when the packet comes in and the CPU is halted?

If I only breakpoint at `modem_socket_data_ready`, then I remember everytime this function runs, `sock->is_polled` is always false, but have to check again tomorrow.

ycsin, 2021-04-15:
> `modem_socket_poll` would fail if the quectel-bg9x zephyr driver (or the quectel modem) didn't receive data or failed to report data reception.

It did report data reception
```
[00:00:22.211,000] <inf> modem_quectel_bg9x: Data Receive Indication for socket: 0
```
and the AT commands shows that it received something
```
+QIURC: "recv",0
```
I read that you tested this driver with mqtt pub/sub, did you try the `mqtt_publisher` sample or something else?

bwasim, 2021-04-15:
You're receiving a URC indicating that the modem has received data for socket 0. Once this is done, the `modem_socket_poll` should unblock (with success) and your application would trigger a read from the modem. This will eventually call up `on_cmd_sockread_common` in quectel-bg9x.c which performs the actual read from zephyr. Seems that that's not happening.  

Can you debug your main application (mqtt_publisher) and check what the call to `wait` returns and if `mqtt_input` is actually invoked ? https://github.com/zephyrproject-rtos/zephyr/blob/master/samples/net/mqtt_publisher/src/main.c#L381

This driver has been tested been mqtt publisher multiple times before.

ycsin, 2021-04-16:
> `mqtt_input`

`wait()` returned 0, the `if (nfds > 0)` in `wait()` is not true, so it just returned the `int ret = 0`

ycsin, 2021-04-16:
@rlubos 
```
[00:00:22.065,000] <inf> net_mqtt_publisher_sample: attempting to connect:
[00:00:22.129,000] <inf> modem_quectel_bg9x: AT+QIOPEN: 0
[00:00:22.136,000] <inf> modem_socket: modem_socket_poll: reset sem_poll
[00:00:22.136,000] <inf> modem_socket: modem_socket_poll: if (sock)
[00:00:22.136,000] <inf> modem_socket: modem_socket_poll: is_polled = true
[00:00:22.136,000] <inf> modem_socket: modem_socket_poll: Waiting for sem_poll
[00:00:22.207,000] <inf> modem_quectel_bg9x: Data Receive Indication for socket: 0
[00:00:22.207,000] <inf> modem_socket: modem_socket_data_ready: give sem_poll
[00:00:22.207,000] <inf> net_mqtt_publisher_sample: wait ret: 0
[00:00:32.212,000] <inf> net_mqtt_publisher_sample: MQTT client disconnected -53
[00:00:32.286,000] <inf> modem_quectel_bg9x: AT+QIOPEN: 0
```

Looks like something is wrong after it got the sem, I need to add more log to see what's going on better

ycsin, 2021-04-16:
> You're receiving a URC indicating that the modem has received data for socket 0. Once this is done, the `modem_socket_poll` should unblock (with success) and your application would trigger a read from the modem. This will eventually call up `on_cmd_sockread_common` in quectel-bg9x.c which performs the actual read from modem. Seems that that's not happening.
> 
> Can you debug your main application (mqtt_publisher) and check what the call to `wait` returns and if `mqtt_input` is actually invoked ? https://github.com/zephyrproject-rtos/zephyr/blob/master/samples/net/mqtt_publisher/src/main.c#L381
> 
> This driver has been tested been mqtt publisher multiple times before.

The `wait()` returned 0 as `z_sock_poll()` returned 0, and because of that `mqtt_input()` didn't run

ycsin, 2021-04-16:
Log:
```
[00:00:22.061,000] <inf> net_mqtt_publisher_sample: attempting to connect:
[00:00:22.061,000] <inf> net_mqtt_publisher_sample: Attempt mqtt_connect
[00:00:22.117,000] <inf> modem_quectel_bg9x: AT+QIOPEN: 0
[00:00:22.124,000] <inf> net_mqtt_publisher_sample: prepare_fds
[00:00:22.124,000] <inf> modem_socket: modem_socket_poll: reset sem_poll
[00:00:22.124,000] <inf> modem_socket: modem_socket_poll: if (sock)
[00:00:22.124,000] <inf> modem_socket: modem_socket_poll: is_polled = true
[00:00:22.124,000] <inf> modem_socket: modem_socket_poll: Waiting for sem_poll
[00:00:22.171,000] <inf> modem_quectel_bg9x: Data Receive Indication for socket: 0
[00:00:22.171,000] <inf> modem_socket: modem_socket_data_ready: give sem_poll
[00:00:22.171,000] <inf> modem_socket: modem_socket_poll: Took sem_poll
[00:00:22.171,000] <inf> modem_socket: Packet size: 0
[00:00:22.171,000] <inf> modem_socket: modem_socket_poll: found_count 0
[00:00:22.171,000] <inf> net_mqtt_publisher_sample: wait ret: 0
[00:00:22.711,000] <inf> net_mqtt_publisher_sample: MQTT client disconnected -53
```

I think the problem is here:
```
		/*
		 * Handle user check for ZSOCK_POLLOUT events:
		 * we consider the socket to always be writable.
		 */
		if (fds[i].events & ZSOCK_POLLOUT) {
			fds[i].revents |= ZSOCK_POLLOUT;
			found_count++;
            LOG_INF("modem_socket_poll: %d POLLOUT", i);
		} else if ((fds[i].events & ZSOCK_POLLIN) &&
			   (sock->packet_sizes[0] > 0U)) {
			fds[i].revents |= ZSOCK_POLLIN;
			found_count++;
            LOG_INF("modem_socket_poll: %d POLLIN", i);
		}

        LOG_INF("Packet size: %u", sock->packet_sizes[0]);
```

Somehow the packet size is always 0, therefore `modem_socket_poll` always returns 0

bwasim, 2021-04-16:
The URC indicating data reception doesn't inform about the amount of data received. Its just tells the system that data is available for this socket. I would assume that that's why you get `sock->packet_sizes[0]=0`.

bwasim, 2021-04-16:
Seems that the quectel-bg9x driver is broken since this commit: https://github.com/zephyrproject-rtos/zephyr/commit/7d67991a91f21aee5fde7a6830522c1beefbcadd

The quectel-bg9x driver URC doesn't give information about the amount of data received, and hence modem_socket_poll fails.

ycsin, 2021-04-16:
> The URC indicating data reception doesn't inform about the amount of data received. Its just tells the system that data is available for this socket. I would assume that that's why you get `sock->packet_sizes[0]=0`.
> 
> I need to look through the code to understand why `modem_socket_poll` returns 0. Does it have some correlation with `packet_sizes`?
> 
> @ycsin , can you check if `modem_socket_poll` fails because `packet_sizes` is 0. If so, we may have to update the driver to inform the stack about the amount of data received or just fix up the `modem_socket_poll` function. This ([7d67991](https://github.com/zephyrproject-rtos/zephyr/commit/7d67991a91f21aee5fde7a6830522c1beefbcadd)) commit seems a bit suspicious, but I'm not 100% sure yet..

Yep, because the `fds[0].events` is set to `ZSOCK_POLLIN` in [mqtt_publisher's main.c](https://github.com/zephyrproject-rtos/zephyr/blob/master/samples/net/mqtt_publisher/src/main.c#L119), for `modem_socket_poll` to return a non zero value, this condition must be fufilled in the [second for loop](https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/modem/modem_socket.c#L325)

```
((fds[i].events & ZSOCK_POLLIN) && (sock->packet_sizes[0] > 0U))
```

We have can somehow get the packet size during the URC, which is configurable with AT command for BG95 according to one of the user on slack, but EC21 doesn't have this AT command. The other possible method which is shared among quectel modem is by sending AT+QIRD=<connectID>,0 on URC, which will query the total length, read length and unread length.

I think modifying the `modem_socket_data_ready` to set the a flag of the socket to true apart from just giving `sem_poll`, then in the second for loop of `modem_socket_poll` query which socket's flag is set to true to determine a socket received data (instead of packet size) might also work, but I'm not sure about the implication of this.

bwasim, 2021-04-16:
Yeah, I think that the 2nd check for `packet_sizes` done here (https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/modem/modem_socket.c#L324) is redundant. I can understand why the first check (https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/modem/modem_socket.c#L296) helps is needed if the data is already available, but the 2nd check shouldn't be required. Can you just try removing the 2nd check ?
```
		/*
		 * Handle user check for ZSOCK_POLLOUT events:
		 * we consider the socket to always be writable.
		 */
		if (fds[i].events & ZSOCK_POLLOUT) {
			fds[i].revents |= ZSOCK_POLLOUT;
			found_count++;
		} else if (fds[i].events & ZSOCK_POLLIN) {
			fds[i].revents |= ZSOCK_POLLIN;
			found_count++;
		}
```
This should fix up your problem.

Before making this change to the mainline, we should loop @XavierChapron to understand why the 2nd condition was added.

ycsin, 2021-04-16:
> Yeah, I think that the 2nd check for `packet_sizes` done here (https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/modem/modem_socket.c#L324) is redundant. I can understand why the [...]

In fact I tried this, but then the modem will now attempt to read the sock even if there is no new incomming data URC. That resulted in some "length problem/error"

bwasim, 2021-04-16:
> In fact I tried this, but then the modem will now attempt to read the sock even if there is no new incoming data URC. That resulted in some "length problem/error"

I think you're running into a different problem here that has nothing to do with socket polling (probably a protocol error)? If so, we should look at this problem separately. Once the poll function returns successfully, the `offload_recvfrom` function gets invoked with a certain required `len`. Is it possible that in your case, the `len` provided by the app and the length of data returned by the modem mismatch?

ycsin, 2021-04-16:
> Once the poll function returns successfully, the `offload_recvfrom` function gets invoked with a certain required `len`. Is it possible that in your case, the `len` provided by the app and the length of data returned by the modem mismatch?

Because after this change the `modem_socket_poll` will always returns non zero value, since `(fds[i].events & ZSOCK_POLLIN)` is always valid(for every fds available) as this is set from the `main.c` application. And then the `offload_recvfrom ` gets invoked and do `AT+QIRD` even without new incoming data.

Unfortunately I didn't save the AT transactions of the modem, will have to try again during weekend or when I go back to work.

XavierChapron, 2021-04-19:
> Yeah, I think that the 2nd check for `packet_sizes` done here (https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/modem/modem_socket.c#L326) is redundant. I can understand why the first check (https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/modem/modem_socket.c#L296) helps if the data is already available, but the 2nd check shouldn't be required. Can you just try removing the 2nd check ?
> 
> ```
> 		/*
> 		 * Handle user check for ZSOCK_POLLOUT events:
> 		 * we consider the socket to always be writable.
> 		 */
> 		if (fds[i].events & ZSOCK_POLLOUT) {
> 			fds[i].revents |= ZSOCK_POLLOUT;
> 			found_count++;
> 		} else if (fds[i].events & ZSOCK_POLLIN) {
> 			fds[i].revents |= ZSOCK_POLLIN;
> 			found_count++;
> 		}
> ```
> 
> This should fix up your problem.
> 
> Before making this change to the mainline, we should loop-in @XavierChapron to understand why the 2nd condition was added. Addition of this condition enforces all modem drivers to provide data length along with URC. If we end up deciding that this is mandatory, I would just go with the 1st approach you mentioned, i.e.
> 
> > The other possible method which is shared among quectel modem is by sending AT+QIRD=,0 on URC, which will query the total length, read length and unread length.

Hi @bwasim , if I understand correctly, your are talking about the `sock->packet_sizes[0] > 0U` condition?
I didn't introduced this condition, [it was there](https://github.com/zephyrproject-rtos/zephyr/commit/f5c45c29468158b5745352ac0c7cc3a841041774#diff-36990db8e7451f52524903d2da613a9d53d54d18d7c96f8d726741a97c34b9faR273) since the first version.

I'm not familiar with the modem_socket.c code, but the `sock->packet_sizes` array seems used everywhere. Having null values doesn't seem's to be a good idea.

ycsin, 2021-04-20:
Now I fixed most of the driver side issue and it published qos 0 message successfully

Log:
```
[00:00:22.108,000] <inf> net_mqtt_publisher_sample: attempting to connect:
[00:00:22.108,000] <inf> net_mqtt_publisher_sample: Attempt mqtt_connect
[00:00:22.174,000] <inf> modem_quectel_bg9x: AT+QIOPEN: 0
[00:00:22.242,000] <inf> modem_quectel_bg9x: Data Receive Indication for socket: 0 length: 4
[00:00:22.242,000] <inf> net_mqtt_publisher_sample: wait ret: 1
[00:00:22.242,000] <inf> net_mqtt_publisher_sample: mqtt_input
[00:00:22.251,000] <inf> net_mqtt_publisher_sample: MQTT client connected!
[00:00:22.251,000] <inf> net_mqtt_publisher_sample: try_to_connect: 0 <OK>
[00:00:22.256,000] <inf> net_mqtt_publisher_sample: mqtt_ping: 0 <OK>
[00:00:22.301,000] <inf> modem_quectel_bg9x: Data Receive Indication for socket: 0 length: 2
[00:00:22.305,000] <inf> net_mqtt_publisher_sample: PINGRESP packet
[00:00:22.769,000] <inf> net_mqtt_publisher_sample: mqtt_publish: 0 <OK>
[00:00:23.283,000] <inf> net_mqtt_publisher_sample: mqtt_publish: 0 <OK>
[00:00:23.797,000] <inf> net_mqtt_publisher_sample: mqtt_publish: 0 <OK>
[00:00:24.302,000] <inf> net_mqtt_publisher_sample: mqtt_ping: 0 <OK>
[00:00:24.816,000] <inf> net_mqtt_publisher_sample: mqtt_publish: 0 <OK>
[00:00:25.330,000] <inf> net_mqtt_publisher_sample: mqtt_publish: 0 <OK>
[00:00:25.424,000] <inf> modem_quectel_bg9x: Socket Close Indication for socket: 0
[00:00:25.465,000] <inf> modem_quectel_bg9x: Socket Closed: 0
[00:00:25.830,000] <dbg> net_sock.z_impl_zsock_close: (0x20001440): close: ctx=0x20000fac, fd=0
[00:00:25.830,000] <inf> net_mqtt_publisher_sample: MQTT client disconnected -57
[00:00:25.830,000] <inf> net_mqtt_publisher_sample: mqtt_publish: -57 <ERROR>
[00:00:25.830,000] <inf> net_mqtt_publisher_sample: mqtt_disconnect: -57 <ERROR>
[00:00:25.830,000] <inf> net_mqtt_publisher_sample: Bye!
```

Modem AT:
```
AT+QIOPEN=1,0,"TCP","x.x.x.x",1883,0,0
OK

+QIOPEN: 0,0
AT+QISEND=0
> MQTT<zephyr_publisher
SEND OK

+QIURC: "recv",0
AT+QIRD=0,0
+QIRD: 4,0,4

OK
AT+QIRD=0,2
+QIRD: 2

OK
AT+QIRD=0,2
+QIRD: 2

OK
AT+QISEND=0
> À
SEND OK

+QIURC: "recv",0
AT+QIRD=0,0
+QIRD: 6,4,2

OK
AT+QIRD=0,2
+QIRD: 2
Ð

OK
AT+QISEND=0
> 0sensors
SEND OK
AT+QISEND=0
> DOORS:OPEN_QoS0
SEND OK
AT+QISEND=0
> 2
SEND OK
AT+QISEND=0
> DOORS:OPEN_QoS1
SEND OK
AT+QISEND=0
> 4
SEND OK
AT+QISEND=0
> DOORS:OPEN_QoS2
SEND OK
AT+QISEND=0
> À
SEND OK
AT+QISEND=0
> 0sensors
SEND OK
AT+QISEND=0
> DOORS:OPEN_QoS0
SEND OK
AT+QISEND=0
> 2
SEND OK
AT+QISEND=0
> DOORS:OPEN_QoS1
SEND OK

+QIURC: "closed",0
AT+QICLOSE=0
OK
```

It publishes to my server and I subscribe to the `sensors` topic on my laptop, but somehow I only received `DOORS:OPEN_QoS0` once and no QoS 1 or 2 messages received. I modified the `mqtt_publisher` to run `publisher();` only once. From the example code I think it should ping then publish each of QoS 0/1/2 message once, but from the console log and AT commands, seems like it tried to: PING -> PUB QoS 0 -> PUB QoS 1 -> PUB QoS 2 -> PING -> PUB QoS 0 -> PUB QoS 1

@rlubos Could you spot anything obvious?

ycsin, 2021-04-20:
> I'm not familiar with the modem_socket.c code, but the `sock->packet_sizes` array seems used everywhere. Having null values doesn't seem's to be a good idea.

@XavierChapron I think the `modem_socket.c` is fine, the issue should be at the driver. The bg9x driver has nothing to set the initial inbound packet size

closed 2021-07-05 by github-actions[bot]