zephyrproject-rtos/zephyr #34275: drivers: led_pwm: Improper label assignment

MarekPieta, 2021-04-14:
**Describe the bug**
led_pwm driver uses label property assigned directly to the node that is compatible with "pwm-leds". The dts binding assumes that the label is assigned to its child node. Because of that the driver ignores the assigned label and it always uses the default one.

I think that the label should be assigned directly to the node that is compatible with "pwm-leds" (this node could be used to represent a logical LEDs with color channels as child nodes).

https://github.com/zephyrproject-rtos/zephyr/blob/188cb2cb7c44148849889670bedfb3a2e259ad3c/dts/bindings/led/pwm-leds.yaml#L4-L18

**Expected behavior**
Driver implementation should be consistent with the DTS binding.

**Environment (please complete the following information):**
Zephyr SHA: 188cb2cb7c44148849889670bedfb3a2e259ad3c

mbolivar-nordic, 2021-05-12:
@simonguinot what do you think about this one?

simonguinot, 2021-05-13:
Hi @mbolivar-nordic,

I think @MarekPieta  is right. It is correct to have a label attribute for the LED child nodes. It can be used to identify/name a LED and also to retrieve its child node number. But it is not correct to say it can be used to retrieve the LED PWM device (using device_get_binding()). So this comment should be removed. In addition a label property should be added to PWM LED parent node binding, with the "device_get_binding()" comment in the description.

Looking back at the led_pwm driver, I think I made a poor choice using DT_INST_PROP_OR(id, label, "LED_PWM_"#id) to set the device name. I should have used DEVICE_DT_NAME() instead though DEVICE_DT_INST_DEFINE(). If the label property is not available, this falls back the device name to  DT_NODE_FULL_NAME, which is smarter than an arbitrary string. Most of the time the pwm-leds compatible nodes don't have a label property. And it would be better for the application to be able to use DEVICE_DT_NAME() too to retrieve the device name, instead of having to guess an arbitrary string.

And to finish, note that the very same issue can be found in dts/bindings/gpio/gpio-leds.yaml.

Please let me know if you want me to fix this.

mbolivar-nordic, 2021-05-13:
> Please let me know if you want me to fix this.

That'd be great if you have time; thanks!

commit 0530ece6 (2021-05-19): dts: bindings: fix label properties for pwm-leds node

This patch adds a label property to the PWM LED parent node and fixes
the label property description of the child node. Indeed it is the label
of a parent node which can be used as device_get_binding() parameter to
retrieve the LED device, and not the one of a child node.

Fixes #34275
closed 2021-05-19 by carlescufi