zephyrproject-rtos/zephyr #34295: TensorFlow Lite Micro

laurenmurphyx64, 2021-04-14:
## Origin

[TensorFlow](https://github.com/tensorflow/tensorflow)

## Purpose

[TensorFlow](https://www.tensorflow.org/) is an end-to-end open-source platform for machine learning. With support for the [TensorFlow Lite for Microcontrollers](https://www.tensorflow.org/lite/microcontrollers) C/C++ library, Zephyr apps will be able to do  machine learning.

## Mode of integration

I suggest we bring TensorFlow in as a module for flexibility and ability to integrate changes from upstream. The name of the repository could be ``tensorflow`` or ``tensorflow-lite-micro`` (to make it clear only Micro is supported directly).

## Pull Request

PR #34057 

## Description

As mentioned, the TFL Micro library brings machine learning to microcontrollers.

Unless they're willing to do legwork similar to the linked PR on their own, developers of TF apps on Zephyr build inside the TF tree with a Makefile generated from a template that invokes CMake and Make. As a result, they aren't able to interact directly with `CMakeLists.txt`, `prj.conf`` and other build settings / options in the normal way.

By maintaining TFL Micro as a Zephyr module, we greatly lower the barrier for entry for Zephyr developers to use ML and make Zephyr a more viable RTOS for ML-based products.

# Dependencies

The TensorFlow Lite Micro library depends on [Flatbuffers](https://github.com/google/flatbuffers), [Gemmlowp](https://github.com/google/gemmlowp) and [Ruy](https://github.com/google/ruy), which are all ``Apache-2.0`` licensed and maintained by Google. The PR adds two samples to the tree, one of which asks the user to emulate a board on Renode using a setup adapted from  [an Antmicro demo](https://github.com/antmicro/litex-vexriscv-tensorflow-lite-demo/), which is also licensed ``Apache-2.0``.

Since it's mostly a mathematical library for use by the application, no OS code should depend on it. It in turn doesn't need much OS support aside from a C++ 11 compatible compiler, debug logging, etc. (See their [Porting to a new platform guide](https://github.com/tensorflow/tensorflow/blob/master/tensorflow/lite/micro/docs/new_platform_support.md) for more details.)

## Revision

[TensorFlow 2.4.1](https://github.com/tensorflow/tensorflow/releases/tag/v2.4.1), released January 21st (latest release)

## License

``Apache-2.0``

carlescufi, 2021-04-23:
+1 from me, I think we should name the repo as its upstream counterpart, i.e. `tensorflow`

carlescufi, 2021-05-03:
This can be closed now.

closed 2021-05-03 by carlescufi