zephyrproject-rtos/zephyr #34309: unable to connect to azure iot hub via mqtt protocol

moncef-bibani, 2021-04-15:
**Describe the bug**
I am trying to connect the nucleo-f429zi using the Ethernet interface to my azure-iot-hub. The samples project is here: https://docs.zephyrproject.org/latest/samples/net/cloud/mqtt_azure/README.html
i created an iot-hub and i added a device. The sample project requires these parameters:

Username: {iothubhostname}/{device_id}/?api-version=2018-06-30
Device ID, SaS token and Azure cloud server address and azure cloud server port.
I set up these parameters in the menuconfig of the project, except the azure cloud server address and port. I let the mqtt-port with the default value 8883 and i set up the server address like the user-name. Because i could not find any description of the azure cloud server address in my azure IoT hub. 

**To Reproduce**
Steps to reproduce the behavior:
1. cd zephyrproject/zephyr
2. west build -b nucleo_f429zi samples/net/cloud/mqtt_azure -- -DCONF="prj.conf"
3. west build -t menuconfig
4. west flash

**Impact**
the sample project cannot connect to the iot-hub. I guess because i didn´t set up correctly the ip address of the azure server. please the logs in the picture below.

**Logs and console output**
[image]

**Environment (please complete the following information):**
 - OS: Linux Ubuntu
 - Toolchain Zephyr SDK

jukkar, 2021-04-15:
The -116 errno is ETIMEDOUT, so the connection cannot be established. Difficult to know what is going on, there are lot of moving parts involved when setting the Azure configuration parameters etc. I do not have Azure credentials to try this.

jukkar, 2021-04-15:
BTW, the current timeout (from your screenshot) seems to be 3 sec. Could you try to increase it as I remember seeing some complaints that the cloud connection establishment can take a long time.

moncef-bibani, 2021-04-15:
i tried again with esp-01. i could connect the nucleo to my wifi. But this time, i get another error. see picture below:
[image]

moncef-bibani, 2021-04-15:
> The -116 errno is ETIMEDOUT, so the connection cannot be established. Difficult to know what is going on, there are lot of moving parts involved when setting the Azure configuration parameters etc. I do not have Azure credentials to try this.

i skiped the field for the ip-address because it is not mandatory for this sample. But i got another error. thank you for your Response.

closed 2021-04-19 by moncef-bibani