zephyrproject-rtos/zephyr #34315: BMI270 configuration file sending to I2C seems to be not handling the last part of the configuration properly.

aelsam, 2021-04-15:
**Describe the bug**
The BMI270 driver's initiation code is not sending to the I2C sensor the same way as the Bosch code sample.  The BMI270_WR_LEN is written in regardless of the configuration size.    The bosch code handles this with a remaining value.  And not send off data pass the config_file max length. 

**To Reproduce**
Check the write_config_file function in bmi270.c

**Expected behavior**
nothing good will happen if you write the random data that's past the configuration_file's max length. 

**Impact**
Instability

**Additional context**
Current Zephyr driver: BMI270.c

static int8_t write_config_file(struct bmi270_data *dev)
{
	int8_t ret = 0;
	uint16_t index = 0;
	uint8_t addr_array[2] = { 0 };

	/* Disable loading of the configuration */
	for (index = 0; index < sizeof(bmi270_config_file);
	     index += BMI270_WR_LEN) {
		/* Store 0 to 3 bits of address in first byte */
		addr_array[0] = (uint8_t)((index / 2) & 0x0F);

		/* Store 4 to 11 bits of address in the second byte */
		addr_array[1] = (uint8_t)((index / 2) >> 4);

		ret = reg_write(BMI270_REG_INIT_ADDR_0, addr_array, 2, dev);

		k_usleep(BMI270_INTER_WRITE_DELAY_US);

		if (ret == 0) {
			ret = reg_write(BMI270_REG_INIT_DATA,
					(bmi270_config_file + index),
					BMI270_WR_LEN, dev);
			k_usleep(BMI270_INTER_WRITE_DELAY_US);
		}
	}

	return ret;
}

************
BOSCH Sample: bmi2.c (Note it has remaining value handling)

static int8_t write_config_file(struct bmi2_dev *dev)
{
    /* Variable to define error */
    int8_t rslt;

    /* Variable to update the configuration file index */
    uint16_t index = 0;

    /* config file size */
    uint16_t config_size = dev->config_size;

    /* Variable to get the remainder */
    uint8_t remain = (uint8_t)(config_size % dev->read_write_len);

    /* Variable to get the balance bytes */
    uint16_t bal_byte = 0;

    /* Variable to define temporary read/write length */
    uint16_t read_write_len = 0;

    /* Disable advanced power save mode */
    rslt = bmi2_set_adv_power_save(BMI2_DISABLE, dev);
    if (rslt == BMI2_OK)
    {
        /* Disable loading of the configuration */
        rslt = set_config_load(BMI2_DISABLE, dev);
        if (rslt == BMI2_OK)
        {
            if (!remain)
            {
                /* Write the configuration file */
                for (index = 0; (index < config_size) && (rslt == BMI2_OK); index += dev->read_write_len)
                {
                    rslt = upload_file((dev->config_file_ptr + index), index, dev->read_write_len, dev);
                }
            }
            else
            {
                /* Get the balance bytes */
                bal_byte = (uint16_t) config_size - (uint16_t) remain;

                /* Write the configuration file for the balancem bytes */
                for (index = 0; (index < bal_byte) && (rslt == BMI2_OK); index += dev->read_write_len)
                {
                    rslt = upload_file((dev->config_file_ptr + index), index, dev->read_write_len, dev);
                }

                if (rslt == BMI2_OK)
                {
                    /* Update length in a temporary variable */
                    read_write_len = dev->read_write_len;

                    /* Write the remaining bytes in 2 bytes length */
                    dev->read_write_len = 2;

                    /* Write the configuration file for the remaining bytes */
                    for (index = bal_byte;
                         (index < config_size) && (rslt == BMI2_OK);
                         index += dev->read_write_len)
                    {
                        rslt = upload_file((dev->config_file_ptr + index), index, dev->read_write_len, dev);
                    }

                    /* Restore the user set length back from the temporary variable */
                    dev->read_write_len = read_write_len;
                }
            }

            if (rslt == BMI2_OK)
            {
                /* Enable loading of the configuration */
                rslt = set_config_load(BMI2_ENABLE, dev);

                /* Wait till ASIC is initialized */
                dev->delay_us(150000, dev->intf_ptr);
                if (rslt == BMI2_OK)
                {
                    /* Enable advanced power save mode */
                    rslt = bmi2_set_adv_power_save(BMI2_ENABLE, dev);
                }
            }
        }
    }

    return rslt;
}

jfischer-no, 2021-04-15:
@BoschSensortec :arrow_up: FYI

closed 2021-07-11 by github-actions[bot]

abebeos, 2023-09-15:
Closing bugs due to staleness is a sloppy process.

I would expect more precision from an RTOS project.