zephyrproject-rtos/zephyr #34324: RTT is not working on STM32

ghlualven, 2021-04-16:
**Describe the bug**
I would like to use logging RTT backend with Segger RTTViwer but it does not work. My Segger J-Link is not able to find the RTT control block. Using the same setup with another board (nRF52840dk) does work without any problem.

**To Reproduce**
Steps to reproduce the behavior:
1. Use Blinky sample as the base
2. use the following prj.conf
```
CONFIG_DEBUG=y

# RTT
CONFIG_USE_SEGGER_RTT=y

# console
CONFIG_RTT_CONSOLE=y
CONFIG_UART_CONSOLE=n

# shell
CONFIG_SHELL=y
CONFIG_SHELL_BACKENDS=y
CONFIG_SHELL_BACKEND_SERIAL=y
CONFIG_SHELL_PROMPT_UART="uart:~$ "

# logging
CONFIG_LOG=y
CONFIG_LOG_BACKEND_RTT=y
CONFIG_LOG_BACKEND_UART=n
CONFIG_SHELL_LOG_BACKEND=n

CONFIG_LOG_PROCESS_TRIGGER_THRESHOLD=10
CONFIG_LOG_PROCESS_THREAD=y
CONFIG_LOG_PROCESS_THREAD_SLEEP_MS=500
CONFIG_LOG_PROCESS_THREAD_STACK_SIZE=2048
CONFIG_LOG_BUFFER_SIZE=1024
CONFIG_LOG_DETECT_MISSED_STRDUP=y
#LOG Maximum string define
CONFIG_LOG_STRDUP_MAX_STRING=100
CONFIG_LOG_STRDUP_BUF_COUNT=4
```

3. add LOG_xxx in main loop
```
void main(void)
{
	struct device *dev;
	bool led_is_on = true;
	int ret;

    printk("Blinky sample on board: %s\n", CONFIG_BOARD);

	dev = device_get_binding(LED0);
	if (dev == NULL) {
		return;
	}

	ret = gpio_pin_configure(dev, PIN, GPIO_OUTPUT_ACTIVE | FLAGS);
	if (ret < 0) {
		return;
	}

	while (1) {
        // printk("led: %d\n", led_is_on);
        // LOG_DBG("led: %d\n", led_is_on);
        LOG_INF("led: %d\n", led_is_on);
        // LOG_WRN("led: %d\n", led_is_on);
        // LOG_ERR("led: %d\n", led_is_on);
		gpio_pin_set(dev, PIN, (int)led_is_on);
		led_is_on = !led_is_on;
		k_msleep(SLEEP_TIME_MS);
	}
}
```
4. Build and flash the application to board nucleo_l476rg  successfully (the ST-Link on board was already replaced with JLink firmware)
5. Open JLinkRTTViewer and connect successfully with SWD, RTT Control Block select "Auto" or "Address" 0x20000920
```
.bss._SEGGER_RTT
                0x0000000020000920       0xa8 zephyr/libzephyr.a(SEGGER_RTT.c.obj)
                0x0000000020000920                _SEGGER_RTT
```
6. No log output in Terminal at all. 
7. I also use J-Mem, it connected ok, but all memory shows "00".
8. I tried on another board STM32F3 Discovery, the same problem.
9. I also tried to use a real J-Link Plus probe, it has the same problem.

**Expected behavior**
RTT logging should work to display in JLinkRTTViewer. Seems the memory is protected from being readout from SWD. The same program and settings works fine on nRF52840dk.

**Environment (please complete the following information):**
 - OS: MacOS
 - Toolchain: gnuarmemb gcc-arm-none-eabi-9-2019-q4-major
 - Commit SHA or Version used: 2.5-branch 6c2ae13e9a85f41dc421c206f28baf6eb71ba48e
 - JLink: SEGGER J-Link V6.98e

erwango, 2021-04-26:
@ghlualven Unfortunately there are not much RTT users on STM32. Have you tried with another STM32 reference (not STM32F3 based) ?

ghlualven, 2021-04-26:
Yes, I tried on nucleo_l476rg, nucleo_f411re, L443VC, all the same problem.

mHendelja, 2021-04-27:
I also tried on STM32L433RC and custom board build around same MCU, same problem with RTT. 
RTT displaced only Zephyr crash log when I made segmentation fault.

erwango, 2021-04-27:
ok, thanks for reporting. We'll see what we can do.

erwango, 2021-04-27:
@ghlualven, @mHendelja  Looking to https://wiki.segger.com/STM32, it seems some specvific steps are required for STM32 devices. Have you tried those ?

wthidden, 2021-04-27:
I've tried all the suggestions on the segger page with no joy.

ghlualven, 2021-04-28:
I also have tried all the suggestions on the segger page, no lucky.

mHendelja, 2021-04-29:
@erwango did all steps and some additional settings, nothing enables RTT.

erwango, 2021-04-29:
@wthidden, @mHendelja Thanks for reporting. Unfortunately, I don't know much about RTT and I'm not a user, so my ability to help will be limited. I'll try to seek for information internally.

wthidden, 2021-04-29:
What was the reasoning on giving this bug a  low priority?

erwango, 2021-04-29:
> What was the reasoning on giving this bug a low priority?

One subsystem on one vendor.

ghlualven, 2021-04-30:
I hope this would be fixed soon.

erwango, 2021-05-05:
@wthidden You're stating in https://github.com/zephyrproject-rtos/zephyr/pull/34828 that this is working on a F429 board.
Did it fix you're issue or there are still cases where it does not work ?

wthidden, 2021-05-07:
PR #34828 fixed my issue with not having RTT and SystemView working on the F429 boards. SystemView sometimes is not able to gather data, but after a board power cycle SystemView seems fine.

erwango, 2021-05-07:
@wthidden Thanks for this feedback.
@ghlualven, @mHendelja Can you have a try on your side ?

ghlualven, 2021-05-08:
Still not work on my side.
On one STM32L443 board, there have no any output.
On one STM32L073VZ board, there have some outout, but the MCU stop working right after RTT connected.
```
*** Booting Zephyr OS build zephyr-v2.5.0-3693-g2f01479b564c  ***

Blinky sample on board: ASD V1

[00:00:00.001,000] <inf> main: led: 1

[00:00:01.001,000] <inf> main: led: 0

[00:00:02.001,000] <inf> main: led: 1

[00:00:03.001,000] <inf> main: led: 0

[00:00:04.001,000] <inf> main: led: 1

[00:00:05.002,000] <inf> main: led: 0

[00:00:06.002,000] <inf> main: led: 1

[00:00:07.002,000] <inf> main: led: 0

[00:00:08.002,000] <inf> main: led: 1

[00:00:09.002,000] <inf> main: led: 0

[00:00:10.003,000] <inf> main: led: 1

[00:00:11.003,000] <inf> main: led: 0

[00:00:12.003,000] <inf> main: led: 1

[00:00:13.003,000] <inf> main: led: 0

[00:00:14.003,000] <inf> main: led: 1

[00:00:15.004,000] <inf> main: led: 0

[00:00:16.004,000] <inf> main: led: 1

[00:00:17.004,000] <inf> main: led: 0

[00:00:18.004,000] <inf> main: led: 1

*** Booting Zephyr OS build zephyr-v2.5.0-3693-g2f01479b564c  ***

Blinky sample on board: ASD V1

[00:00:00.001,000] <inf> main: led: 1

[00:00:01.001,000] <inf> main: led: 0

[00:00:02.001,000] <inf> main: led: 1

[00:00:03.001,000] <inf> main: led: 0

[00:00:04.001,000] <inf> main: led: 1

[00:00:05.002,000] <inf> main: led: 0

[00:00:06.002,000] <inf> main: led: 1

[00:00:07.002,000] <inf> main: led: 0

[00:00:08.002,000] <inf> main: led: 1

[00:00:09.002,000] <inf> main: led: 0

[00:00:10.003,000] <inf> main: led: 1

[00:00:11.003,000] <inf> main: led: 0

[00:00:12.003,000] <inf> main: led: 1

[00:00:13.003,000] <inf> main: led: 0

[00:00:14.003,000] <inf> main: led: 1

[00:00:15.004,000] <inf> main: led: 0

[00:00:16.004,000] <inf> main: led: 1

[00:00:17.004,000] <inf> main: led: 0

[00:00:18.004,000] <inf> main: led: 1

*** Booting Zephyr OS build zephyr-v2.5.0-3693-g2f01479b564c  ***

Blinky sample on board: ASD V1

[00:00:00.001,000] <inf> main: led: 1

*** Booting Zephyr OS build zephyr-v2.5.0-3693-g2f01479b564c  ***

Blinky sample on board: ASD V1

[00:00:00.001,000] <inf> main: led: 1
```

erwango, 2021-05-10:
@qianfan-Zhao, @GeorgeCGV you contributed to RTT support on STM32 thanks to following PR:
https://github.com/zephyrproject-rtos/zephyr/pull/8508
https://github.com/zephyrproject-rtos/zephyr/pull/13668

I know this is long time back, but as such, you're the reference RTT users on STM32.
Would you mind having a look to the reported issue or provide some information on the way you used RTT on STM32 devices (Segger version, Zephyr DV, boards, ...) ? Thanks in advance

manoj153, 2021-06-01:
Issue: systemview is not displaying any events and halts code execution.
Is there any link between this issue to my observation? 
I had not been able to make the system view work properly with **nucleo_h723zg**

I have enabled the suggested .conf as per: https://docs.zephyrproject.org/latest/guides/tracing/index.html#tracing

```
# enable to use thread names
CONFIG_THREAD_NAME=y
CONFIG_SEGGER_SYSTEMVIEW=y
CONFIG_USE_SEGGER_RTT=y
CONFIG_TRACING=y
# enable for post-mortem tracing
CONFIG_SEGGER_SYSVIEW_POST_MORTEM_MODE=n
```

Log from SV:

```
15:54:19 - JLink: Device "STM32H723ZG" selected.
15:54:19 - JLink: Found SW-DP with ID 0x6BA02477
15:54:19 - JLink: Found SW-DP with ID 0x6BA02477
15:54:19 - JLink: DPIDR: 0x6BA02477
15:54:19 - JLink: Scanning AP map to find all available APs
15:54:19 - JLink: AP[3]: Stopped AP scan as end of AP map has been reached
15:54:19 - JLink: AP[0]: AHB-AP (IDR: 0x84770001)
15:54:19 - JLink: AP[1]: AHB-AP (IDR: 0x84770001)
15:54:19 - JLink: AP[2]: APB-AP (IDR: 0x54770002)
15:54:19 - JLink: Iterating through AP map to find AHB-AP to use
15:54:19 - JLink: AP[0]: Core found
15:54:19 - JLink: AP[0]: AHB-AP ROM base: 0xE00FE000
15:54:19 - JLink: CPUID register: 0x411FC272. Implementer code: 0x41 (ARM)
15:54:19 - JLink: Found Cortex-M7 r1p2, Little endian.
15:54:19 - JLink: FPUnit: 8 code (BP) slots and 0 literal slots
15:54:19 - JLink: CoreSight components:
15:54:19 - JLink: ROMTbl[0] @ E00FE000
15:54:19 - JLink: ROMTbl[0][0]: E00FF000, CID: B105100D, PID: 000BB4C7 ROM Table
15:54:19 - JLink: ROMTbl[1] @ E00FF000
15:54:19 - JLink: ROMTbl[1][0]: E000E000, CID: B105E00D, PID: 000BB00C SCS-M7
15:54:20 - JLink: ROMTbl[1][1]: E0001000, CID: B105E00D, PID: 000BB002 DWT
15:54:20 - JLink: ROMTbl[1][2]: E0002000, CID: B105E00D, PID: 000BB00E FPB-M7
15:54:20 - JLink: ROMTbl[1][3]: E0000000, CID: B105E00D, PID: 000BB001 ITM
15:54:20 - JLink: ROMTbl[0][1]: E0041000, CID: B105900D, PID: 001BB975 ETM-M7
15:54:20 - JLink: ROMTbl[0][2]: E0043000, CID: B105900D, PID: 004BB906 CTI
15:54:20 - JLink: Cache: Separate I- and D-cache.

15:54:20 - JLink: I-Cache L1: 32 KB, 512 Sets, 32 Bytes/Line, 2-Way
15:54:20 - JLink: D-Cache L1: 32 KB, 256 Sets, 32 Bytes/Line, 4-Way
15:54:30 - Warning: Did not record SystemView TRACE START event, yet.
```

erwango, 2021-06-02:
@manoj153 

> @erwango please advise if I should open a new issue for this.

Hard to say for now. Let's keep it here and see how it evolves

erwango, 2021-06-02:
@ghlualven, @mHendelja

I'd like to see if the following could have an impact on your issue.
Would you mind having a try to add the following line in `soc/arm/st_stm32/stm32l4/soc.c`:

```
#include "stm32l4xx_ll_system.h"
```
Then, in stm32l4_init:
```
LL_DBGMCU_EnableDBGSleepMode();
```

Thanks

ghlualven, 2021-06-03:
@erwango I tried the change, but no lucky.

erwango, 2021-06-03:
> @erwango I tried the change, but no lucky.

ok, thanks.

manoj153, 2021-06-20:
> @manoj153
> 
> > @erwango please advise if I should open a new issue for this.
> 
> Hard to say for now. Let's keep it here and see how it evolves

> Issue: systemview is not displaying any events and halts code execution.
> Is there any link between this issue to my observation?
> I had not been able to make the system view work properly with **nucleo_h723zg**
> 
> I have enabled the suggested .conf as per: https://docs.zephyrproject.org/latest/guides/tracing/index.html#tracing
> 
> ```
> # enable to use thread names
> CONFIG_THREAD_NAME=y
> CONFIG_SEGGER_SYSTEMVIEW=y
> CONFIG_USE_SEGGER_RTT=y
> CONFIG_TRACING=y
> # enable for post-mortem tracing
> CONFIG_SEGGER_SYSVIEW_POST_MORTEM_MODE=n
> ```
> 
> Log from SV:
> 
> ```
> 15:54:19 - JLink: Device "STM32H723ZG" selected.
> 15:54:19 - JLink: Found SW-DP with ID 0x6BA02477
> 15:54:19 - JLink: Found SW-DP with ID 0x6BA02477
> 15:54:19 - JLink: DPIDR: 0x6BA02477
> 15:54:19 - JLink: Scanning AP map to find all available APs
> 15:54:19 - JLink: AP[3]: Stopped AP scan as end of AP map has been reached
> 15:54:19 - JLink: AP[0]: AHB-AP (IDR: 0x84770001)
> 15:54:19 - JLink: AP[1]: AHB-AP (IDR: 0x84770001)
> 15:54:19 - JLink: AP[2]: APB-AP (IDR: 0x54770002)
> 15:54:19 - JLink: Iterating through AP map to find AHB-AP to use
> 15:54:19 - JLink: AP[0]: Core found
> 15:54:19 - JLink: AP[0]: AHB-AP ROM base: 0xE00FE000
> 15:54:19 - JLink: CPUID register: 0x411FC272. Implementer code: 0x41 (ARM)
> 15:54:19 - JLink: Found Cortex-M7 r1p2, Little endian.
> 15:54:19 - JLink: FPUnit: 8 code (BP) slots and 0 literal slots
> 15:54:19 - JLink: CoreSight components:
> 15:54:19 - JLink: ROMTbl[0] @ E00FE000
> 15:54:19 - JLink: ROMTbl[0][0]: E00FF000, CID: B105100D, PID: 000BB4C7 ROM Table
> 15:54:19 - JLink: ROMTbl[1] @ E00FF000
> 15:54:19 - JLink: ROMTbl[1][0]: E000E000, CID: B105E00D, PID: 000BB00C SCS-M7
> 15:54:20 - JLink: ROMTbl[1][1]: E0001000, CID: B105E00D, PID: 000BB002 DWT
> 15:54:20 - JLink: ROMTbl[1][2]: E0002000, CID: B105E00D, PID: 000BB00E FPB-M7
> 15:54:20 - JLink: ROMTbl[1][3]: E0000000, CID: B105E00D, PID: 000BB001 ITM
> 15:54:20 - JLink: ROMTbl[0][1]: E0041000, CID: B105900D, PID: 001BB975 ETM-M7
> 15:54:20 - JLink: ROMTbl[0][2]: E0043000, CID: B105900D, PID: 004BB906 CTI
> 15:54:20 - JLink: Cache: Separate I- and D-cache.
> 
> 15:54:20 - JLink: I-Cache L1: 32 KB, 512 Sets, 32 Bytes/Line, 2-Way
> 15:54:20 - JLink: D-Cache L1: 32 KB, 256 Sets, 32 Bytes/Line, 4-Way
> 15:54:30 - Warning: Did not record SystemView TRACE START event, yet.
> ```
> 
> [image]
> 
> @erwango please advise if I should open a new issue for this.

How is it evolving @erwango in regards to this specific issue?
found someone mentioned similar here:
https://lists.zephyrproject.org/g/users/topic/help_using_systemview_on_an/81836728?p=,,,20,0,0,0::recentpostdate%2Fsticky,,,20,2,20,81836728

GeorgeCGV, 2021-08-18:
> @qianfan-Zhao, @GeorgeCGV you contributed to RTT support on STM32 thanks to following PR:
> #8508
> #13668
>
> I know this is long time back, but as such, you're the reference RTT users on STM32.
> Would you [...]

@erwango, the goal was to fix #10739, as debugging over the J-Link worked way better for me (*used flashed ST-LINK On-Board*).
If I remember correctly, the SystemView didn't work for me as well. However, I didn't dive into that back in the day.

ABOSTM, 2021-08-18:
I spend some time investigating this issue, and here are my findings.
My current status:
RTT is working on:
* nucleo_f429zi
* nucleo_l4r5zi
* stm32f3_disco

RTT is not working on:
* nucleo_f411re
* nucleo_l476rg

During my investigations, I figure out that even on "not working" boards (nucleo_l476rg mainly used to debug), it is possible to make it work:
* RTT is working when a debugger is attached, and I break CPU.
It means that RTTviewer did not get live update of messages while cpu is running, but a soon as I break CPU, RTTviewer get all messages (including those 'send' while running)
It means also that when debugging step by step it works fine (because CPU is implicitly stopped after each step) … which doesn't help to debug ☹
So it seems all mechanisms (Zephyr RTT port) to write data to RAM are working fine. But RTT viewer did succeed to access it while CPU is running.

* When adding RTT trace within zephyr kernel initialization (so before scheduling), RTTviewer get those traces provided that either I reduce system clock (for example default reset MSI) or I add some active waiting loop after traces (while loop).
My feeling is that in both cases we let enough time for RTTviewer to get the messages before starting zephyr scheduler.
But adding traces in 'main' function (so after scheduling is started), those traces are not shown in RTTviewer with the same conditions. So Scheduling seems to be the key point.
In other words, without zephyr scheduling RTT is working fine.

* When adding RTT trace in 'main' function (after scheduling), RTTviewer get those traces on condition that I replace k_msleep() by an active loop (either k_busy_wait() or while loop),
(no need to reduce system clock).
And if I mixes both k_msleep() and active loop, RTT may work depending on values for k_msleep and while loop.
In other words, without zephyr scheduling RTT is working fine.

Note: the state of CONFIG_MPU has no effect
  and CONFIG_PM also has no effect

I compare nucleo_l476rg and nucleo_l4r5zi  , both have the same compilation switch/kconfig enabled,
except of course dedicated switches like CONFIG_SOC, CONFIG_BOARD, CONFIG_FLASH_SIZE, CONFIG_SRAM_SIZE, CONFIG_NUM_IRQS. And that is all.
They have the same system clock.
I took the  exact nucleo_l476rg binary (not working), and I flashed it to nucleo_l4r5zi  … RTT is working fine on nucleo_l4r5zi  !!!

So it seems that there is an hardware difference between l476rg and l4r5zi, (probably similar to diff between f411re and f429zi),
which, in combination with zephyr scheduling causes RTT viewer to no accessing RAM while CPU is running. 
Any help from Zephyr scheduler experts is welcome .

ABOSTM, 2021-08-19:
A bit more investigation shows that it is not linked to zephyr scheduler, but simply to WFI instruction. 
Problem vanished when WFI is replaced (for test purpose) by a NOP.

jaybuckeye06, 2021-09-05:
Thank @ABOSTM for the extensive experiments. wfi -> nop leads me to think about the sleep / standby mode. 

STM32 has a debug register DBGMCU. It contains bits controlling low power behavior, c.f. [RM0008](https://www.st.com/resource/en/reference_manual/cd00171190-stm32f101xx-stm32f102xx-stm32f103xx-stm32f105xx-and-stm32f107xx-advanced-arm-based-32-bit-mcus-stmicroelectronics.pdf) and [AN4989](https://www.st.com/resource/en/application_note/dm00354244-stm32-microcontroller-debug-toolbox-stmicroelectronics.pdf).

[image]

I have just done some experiments on Nucleo_F103RB. Previously there was no output until the target is halted. Now there is at least some incomplete output when the target is running.

[image]

I will continue to explore it and would appreciate any suggestions.

jaybuckeye06, 2021-09-06:
Further found out that DMA1 EN bit should be set in RCC_AHBENR register (@ 0x4002 1014). The description is in RM0008 rev. 21 p.g. 111.

[image]

I am only having a nucleo_f103rb in my hands at this moment. Could someone confirm this on other boards? 

---

JLinkEXE
```
CoreSight components:
ROMTbl[0] @ E00FF000
ROMTbl[0][0]: E000E000, CID: B105E00D, PID: 001BB000 SCS
ROMTbl[0][1]: E0001000, CID: B105E00D, PID: 001BB002 DWT
ROMTbl[0][2]: E0002000, CID: B105E00D, PID: 000BB003 FPB
ROMTbl[0][3]: E0000000, CID: B105E00D, PID: 001BB001 ITM
ROMTbl[0][4]: E0040000, CID: B105900D, PID: 001BB923 TPIU-Lite
Cortex-M3 identified.
J-Link>r
Reset delay: 0 ms
Reset type NORMAL: Resets core & peripherals via SYSRESETREQ & VECTRESET bit.
Reset: Halt core after reset via DEMCR.VC_CORERESET.
Reset: Reset device via AIRCR.SYSRESETREQ.
J-Link>W4 0xE0042004,7
Writing 00000007 -> E0042004
J-Link>W4 0x40021014,0x15
Writing 00000015 -> 40021014
J-Link>g
J-Link>
```

Output:
```
SEGGER J-Link V7.50a - Real time terminal output
J-Link STLink V21 compiled Aug 12 2019 10:29:20 V1.0, SN=774339459
Process: JLinkExe
[00:00:00.000,000] <err> main: LOG_ERR: 0

[00:00:01.000,000] <err> main: LOG_ERR: 1

[00:00:02.000,000] <err> main: LOG_ERR: 2

[00:00:03.000,000] <err> main: LOG_ERR: 3

[00:00:04.000,000] <err> main: LOG_ERR: 4

[00:00:05.000,000] <err> main: LOG_ERR: 5
```

erwango, 2021-09-06:
>  Further found out that DMA1 EN bit should be set in RCC_AHBENR register (@ 0x4002 1014). The description is in RM0008 rev. 21 p.g. 111., although it does not mention its relationship with AHB-AP.

@jaybuckeye06 Woul you min opening a dedicated issue for this specific point ?

mass-anis, 2021-09-15:
@jaybuckeye06 tested it on a nucleol433 and it works.
setting **DBG_SLEEP**, **DBG_STOP**, and **DBG_STANDBY** to 1 in the **DBGMCU_CR** and enabling one of the DMA channels in **RCC_AHB1ENR**  seems to do the trick

ABOSTM, 2021-09-15:
@jaybuckeye06 , @mass-anis ,
Thanks for your help, that is a good catch, enabling clock of DMA (either DMA1 or DMA2) works !!!
As far as DBGMCU bits, it is not mandatory for STM32L476, but it is mandatory for STM32F103.

So I made a PR #38561 to activate DMA clock and setting one of DBGMCU bit.
I hope it will be sufficient for all STM32 boards.

erwango, 2021-09-15:
@ghlualven, @mHendelja, @wthidden, @manoj153 would you mind testing #38561 ?

closed 2021-09-21 by cfriedt

Coi-l, 2022-12-28:
Hello, sorry to awaken an old issue, but I am seeing similar problems on a H7 board and I can't sort it out.
I have a STM32H7A3 Nucleo-144 board with an STM32H7A3ZIT6Q on it.
I have a reference blinky-app with some printouts running on this, I have the same app running on a blackpill board without issues and RTT logging working properly. I can flash the board via the segger. It just seems to be the logging that is the issue.

I can not seem to get the logging running on the nucleo board, the led blinks properly.
I followed what @jaybuckeye06 did and tried to enable the debug bits and a dma clock with the following sequence:
r
W4 0xE00E1004, 7 #debug bits
W4 0x58024538, 3 #Enable two dma clocks (tried with just one as well)
g

But no luck. Maybe I'm missing something trivial in what is going on here, but I'm a bit out of my depth as well.
Any suggestions would be appreciated.

Thanks in advance

PR #38561 (2021-09-21): soc: stm32: SEGGER RTT requires some extra configuration
soc: stm32: SEGGER RTT requires some extra configuration

On some STM32 boards, for unclear reason,
RTT feature is working with realtime update only when
  * one of the DMA is clocked
and sometimes also
  * one of the DBGMCU bit STOP/STANDBY/SLEEP is set
 
Fixes #34324
commit 436ba560 (2021-09-21): soc: stm32: SEGGER RTT requires some extra configuration

On some STM32 boards, for unclear reason,
RTT feature is working with realtime update only when
  * one of the DMA is clocked
and sometimes also
  * one of the DBGMCU bit STOP/STANDBY/SLEEP is set
Fixes #34324