zephyrproject-rtos/zephyr #34325: HAL microchip: Missing Wake

albertofloyd, 2021-04-16:
**Describe the bug**
None of the wake event (GIRQ21 and GIRQ22) are exposed in the HAL 

KSC_INT, SMB-I2C _WAKE_ONLY, ESPI_WAKE_ONLY, PS2_0A_WK, etc

**Expected behavior**
Bit definitions to be added in HAL.

**Impact**
Magic numbers to be used inside SoC power module making this error prone or relying in old documentation.

scottwcpg, 2021-04-16:
The modules/hal/microchip/mec/mec1501/component/keyscan.h has defines for keyscan GIRQ21 NVIC value 135 and GIRQ21 bit position 21, and a define for the GIRQ mask value.  There's a type in the NVIC, MCHP_KSAN_NVIC instead of MCHP_KSCAN_NVIC.  I will fix.  Is this sufficient for key scan definitions?
PS/2 component header ps2_ctrl.h is missing the GIRQ21 wake bits.  Will add.
I will add the GIRQ22 bit definitions to the the ecia.h component header so they are in one place.

albertofloyd, 2021-04-16:
Sounds like a good plan.
Note most interrupt values and aggregator bits are directly used in device tree (no macros required)
I'm referring to LPM wake up-only GIRQs values   (SMB, PS2, KSCINT & eSPI wake)

Current code 
void soc_deep_sleep_non_wake_en(void)
{
#ifdef CONFIG_ESPI_XEC
	GIRQ22_REGS->SRC = 0xfffffffful;
	GIRQ22_REGS->EN_SET = (1ul << 9); /* Better to have a macro */
#endif
}

Code to be added better having a HAL macro than self-defined in the module itself. 
/* When MEC15xx drivers are power-aware this should be move there */
#define MCHP_PS20B_WAKE_ONLY_BIT  BIT(19)
void soc_deep_sleep_wake_en(void)
{
#ifdef CONFIG_PS2_XEC_0
	/* Enable PS2_0B_WK */
	GIRQ21_REGS->SRC = MCHP_PS20B_WAKE_ONLY_BIT  ;
	GIRQ21_REGS->EN_SET = MCHP_PS20B_WAKE_ONLY_BIT  ;
#endif
#ifdef CONFIG_PS2_XEC_1
	/* Enable PS2_1B_WK */
	GIRQ21_REGS->SRC = (1ul << 21);    	
	GIRQ21_REGS->EN_SET = (1ul << 21);
#endif
}

closed 2021-04-28 by nashif