zephyrproject-rtos/zephyr #34342: No output on SWO pin (STM32L4)

andresmanelli, 2021-04-16:
**Describe the bug**
I'm not quite sure if this is a bug, but by default the SWO pin is not enabled to output data. I thing the debugger could do this, but I feel like asking the question though.

As a workaround, I added this line:

```diff
diff --git a/subsys/logging/log_backend_swo.c b/subsys/logging/log_backend_swo.c
index 203896ccd0..8e2e3da5cc 100644
--- a/subsys/logging/log_backend_swo.c
+++ b/subsys/logging/log_backend_swo.c
@@ -98,6 +98,8 @@ static void log_backend_swo_init(void)
         ITM->TCR  = 0x0001000D;
         /* Enable stimulus port used by the logger */
         ITM->TER  = 1 << ITM_PORT_LOGGER;
+        /* Enable the SWO pin */
+        DBGMCU->CR |= 0x20;
 }

 static void log_backend_swo_panic(struct log_backend const *const backend)
```

**To Reproduce**
 - Enable SWO logging as backend
 - Log simple messages in loop
 - No output is present on pin

**Expected behavior**
SWO pin outputs the logging data

**Environment (please complete the following information):**
 - OS: ArchLinux
 - Toolchain Zephyr SDK
 - BlackMagicProbe (Bumpy)
 - Version 9f858ac1b6bcf7f9262906f6719d51d5be4404ac

**Additional context**
Testing SWO with STM32L4

galak, 2021-04-20:
@andresmanelli can you send a PR with this.

andresmanelli, 2021-04-20:
@galak submitted #34436

FRASTM, 2021-04-21:
If so, maybe using the `LL_DBGMCU_SetTracePinAssignment` of the stm32l4xx_ll_system.h  could help.

PR #34436 (2021-05-11): logging: Enable SWO pin for STM32 SoCs 
Set TRACE_MODE to asynchronous and enable trace output pin.

Fixes #34342
commit 53d00dc2 (2021-05-11): logging: Enable SWO pin for STM32 SoCs

Set TRACE_MODE to asynchronous and enable trace output pin.
Add soc_config.c in stm32 soc direcotry.

Fixes #34342
closed 2021-05-11 by galak