zephyrproject-rtos/zephyr #34355: LittleFS sample code catch an "undefined symbol `ITCM_ADDR' referenced in expression" in linker step

icatorze, 2021-04-16:
I'm trying littlefs subsys sample on a mimxrt1020_evk with relocation to ITCM zone. After a menuconfig i can build code but linker step aborts with:
undefined symbol `ITCM_ADDR' referenced in expression.

There's no evidence of ITCM_ADDR anymore, because this address comes from DTS file now. I searched all code and scripts but can not found who is still using this symbol. 

After I manually edit generated files: linker_relocated.ld and linker_zephyr_prebuilt.cmd and force a build, result binary run sucessfully on board.

**To Reproduce**
`west build -d build\littlefssample -b mimxrt1020_evk zephyr\samples\subsys\fs\littlefs -DDTC_OVERLAY_FILE=boards/mimxrt1020_evk.overlay
`
My mimxrt1020_evk.overlay and  mimxrt1020_evk.conf are attached. (Please remove .txt extension)

[mimxrt1020_evk.conf.txt](https://github.com/zephyrproject-rtos/zephyr/files/6327457/mimxrt1020_evk.conf.txt)
[mimxrt1020_evk.overlay.txt](https://github.com/zephyrproject-rtos/zephyr/files/6327458/mimxrt1020_evk.overlay.txt)

galak, 2021-04-16:
@JordanYates can you take a look at this.  Not sure if its related to your recent work.

JordanYates, 2021-04-16:
Yes, this was reported in #34185 and fixed in #34331.

galak, 2021-04-19:
Considering this resolved.  @icatorze please reopen if you still have issues.

closed 2021-04-19 by galak