zephyrproject-rtos/zephyr #34368: Cmake's Python path breaks after using west build --pristine

chepo92, 2021-04-18:
I have two python 3 installations, but cmake fails to find the correct one after using --pristine build option 

**To Reproduce**
Steps to reproduce the behavior:

1. After correctly building a hello world example i used --pristine option to build another example
` west build -b sam4e_xpro samples/subsys/usb/cdc_acm --pristine`
2. which caused the following error:

```
(base) C:\Users\Axel\zephyrproject\zephyr>west -v build -b sam4e_xpro samples/subsys/usb/cdc_acm --pristine
ZEPHYR_BASE=C:\Users\Axel\zephyrproject\zephyr (origin: configfile)
-- west build: generating a build system
cmake version 3.20.1 is OK; minimum version is 3.13.1
Running CMake: 'C:\Program Files\CMake\bin\cmake.EXE' '-DWEST_PYTHON=c:\users\axel\.platformio\penv\scripts\python.exe' '-BC:\Users\Axel\zephyrproject\zephyr\build' '-SC:\Users\Axel\zephyrproject\zephyr\samples\subsys\usb\cdc_acm' -GNinja -DBOARD=sam4e_xpro
Including boilerplate (Zephyr base): C:/Users/Axel/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: C:/Users/Axel/zephyrproject/zephyr/samples/subsys/usb/cdc_acm
-- Zephyr version: 2.5.99 (C:/Users/Axel/zephyrproject/zephyr)
-- Found Python3: C:/ProgramData/Anaconda3/python.exe (found suitable exact version "3.7.1") found components: Interpreter
CMake Error at C:/Users/Axel/zephyrproject/zephyr/cmake/west.cmake:48 (message):
  Unable to import west.version from 'C:/ProgramData/Anaconda3/python.exe':

  Traceback (most recent call last):

    File "<string>", line 1, in <module>

  ModuleNotFoundError: No module named 'west'

  Please install with:

      C:/ProgramData/Anaconda3/python.exe -m pip install west

  Or verify these installations:

    The Python version used by west is:  c:\users\axel\.platformio\penv\scripts\python.exe (real path C:/Users/Axel/.platformio/penv/Scripts/python.exe)
    The Python version used by CMake is: C:/ProgramData/Anaconda3/python.exe
Call Stack (most recent call first):
  C:/Users/Axel/zephyrproject/zephyr/cmake/app/boilerplate.cmake:157 (include)
  C:/Users/Axel/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  C:/Users/Axel/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:35 (include_boilerplate)
  CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' '-DWEST_PYTHON=c:\users\axel\.platformio\penv\scripts\python.exe' '-BC:\Users\Axel\zephyrproject\zephyr\build' '-SC:\Users\Axel\zephyrproject\zephyr\samples\subsys\usb\cdc_acm' -GNinja -DBOARD=sam4e_xpro
```

**Expected behavior**
Cmake should pick c:\users\axel\.platformio\penv\scripts\python.exe as indicated instead  it auto selects C:/ProgramData/Anaconda3/python.exe

I made it work by changing manually the cmakecache: `PYTHON_PREFER_EXECUTABLE:FILEPATH=c:\users\axel\.platformio\penv\scripts\python.exe ` after the error is throw, but I have to do it for every new build

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain (gnuarm?)

mbolivar-nordic, 2021-04-21:
Thanks for the verbose logs, @chepo92 !

> -- Found Python3: C:/ProgramData/Anaconda3/python.exe (found suitable exact version "3.7.1") found components: Interpreter

@tejlmand that part looks weird, since earlier it says:

> Running CMake: 'C:\Program Files\CMake\bin\cmake.EXE' '-DWEST_PYTHON=c:\users\axel\.platformio\penv\scripts\python.exe' '-BC:\Users\Axel\zephyrproject\zephyr\build' '-SC:\Users\Axel\zephyrproject\zephyr\samples\subsys\usb\cdc_acm' -GNinja -DBOARD=sam4e_xpro

Seems like there's a bug in the code that should be using WEST_PYTHON for everything  if it is set, no?

chepo92, 2021-04-21:
Yeah, I was searching for similar issues and there is a findpython script in python.cmake that i think is producing this behaviour, https://github.com/zephyrproject-rtos/zephyr/blob/master/cmake/python.cmake

mbolivar-nordic, 2021-04-21:
I'm also surprised you're not hitting https://github.com/zephyrproject-rtos/zephyr/blob/e1a6820e4c470ad5693eef469a0f4b61f868d8ec/cmake/west.cmake#L48

tejlmand, 2021-04-22:
> I'm also surprised you're not hitting

He does:
> ```
>    CMake Error at C:/Users/Axel/zephyrproject/zephyr/cmake/west.cmake:48 (message):
>      Unable to import west.version from 'C:/ProgramData/Anaconda3/python.exe':
> ```

tejlmand, 2021-04-22:
Fixed here: https://github.com/zephyrproject-rtos/zephyr/pull/34479

PR #34479 (2021-04-27): cmake: use WEST_PYTHON as a preferred python installation if defined.
Fixes: #34368

Now added WEST_PYTHON to list of preferred pythons to check.
WEST_PYTHON is added after PYTHON_PREFER so that a user may still
request a specific python is used by supplying
`-DPYTHON_PREFER=<python>` even when building using `west build`.
commit 5db987f9 (2021-04-27): cmake: use WEST_PYTHON as a preferred python installation if defined.

Fixes: #34368

Now added WEST_PYTHON to list of preferred pythons to check.
WEST_PYTHON is added after PYTHON_PREFER so that a user may still
request a specific python is used by supplying
`-DPYTHON_PREFER=<python>` even when building using `west build`.
closed 2021-04-27 by nashif