zephyrproject-rtos/zephyr #34369: Driver esp for wifi got a dead lock.

skyarm, 2021-04-18:
**Describe the bug**
  driver esp for wifi got a dead lock:
  when bg_thread_main call the net_init, then net_init call net_if_init:
  void net_if_init(void)
 {
   ...
    k_mutex_lock(&lock, K_FOREVER); 
    **//net_if_init is the owner of the lock.**
   ...
    init_iface(iface);  //init_iface call esp_iface_init
   }
   void esp_iface_init {
     esp_reset(..) // esp_reset be called.
  }
  
   esp_reset(..) {
     modem_cmd_send(...) //now esp_tx thread  get reset response.
     ret = k_sem_take(&dev->sem_if_up, ESP_INIT_TIMEOUT);  //wait for the sem_if_up sem until timeout.
   }
   
  esp_rx call function on_cmd_ready() {
   ...
     **//net_if_ipv4_addr_rm try to become the owner of the lock,  but the net_if_init is the owner of lock already.**
     net_if_ipv4_addr_rm(dev->net_iface, &dev->ip);
    ...
     k_work_submit_to_queue(&dev->workq, &dev->init_work);  
    **//k_work_submit_to_queue will not be called until k_sem_take(&dev->sem_if_up, ESP_INIT_TIMEOUT) timeout.
   //So k_sem_take(&dev->sem_if_up, ESP_INIT_TIMEOUT) can not take the sem, So it will deadlocked.
   //Fortunately k_sem_take will timeout, But esp_rx thread, esp_init_work thread and main thread are waiting for k_sem_take   timeout, until the k_sem_take timeout,  all the initialization is stopped.**
   ...
  }
  net_if_ipv4_addr_rm(dev->net_iface, &dev->ip) {
  ...
     k_mutex_lock(&lock, K_FOREVER); 
     **//So net_if_ipv4_addr_rm Cann't get the lock, until  k_sem_take(&dev->sem_if_up, ESP_INIT_TIMEOUT) timeout**
  ...
 }

skyarm, 2021-04-18:
I try to fixed the bug, But there is no way. Or update the base library of subsys/net_if.c, So like eswifi, Cann't call net_if_xxx function in other thread.  Or do not wait for the result of esp_iface_init in main thread.

skyarm, 2021-04-20:
The driver of esp wifi is so complex, I suggest that the UART commutation do not work in other thread, like eswifi driver.

skyarm, 2021-04-20:
> The driver of esp wifi is so complex, I suggest that the UART commutation do not work in other thread, like eswifi driver.

A UART interface wifi modem use three thread to initial the driver, It's so complex.

mniestroj, 2021-04-20:
>>The driver of esp wifi is so complex, I suggest that the UART commutation do not work in other thread, like eswifi driver.
>
>A UART interface wifi modem use three thread to initial the driver, It's so complex.

@skyarm Feel free to send PRs with improvements. Any contribution is welcome!

PR #34410 (2021-05-06): drivers: wifi: esp: do not wait at the end of ESP chip reset
esp_reset() is called from net_if init function, which holds net_if lock
after commit 24b49f439970 ("net: if: Add locking"). At the end of
esp_reset() there is a blocking wait on `sem_if_up` semaphore. This
semaphore can be release only by esp_init_work(). esp_init_work()
however blocks on net_if operations, because net_if init function (which
invokes esp_reset() underneath) is still holding net_if lock. As a
result there is a deadlock, because esp_reset() and esp_init_work() are
both waiting on each other.

Remove waiting for `sem_i [... cut]
closed 2021-05-06 by galak