zephyrproject-rtos/zephyr #34372: Occational CPU Lockups on high BLE throughput

caco3, 2021-04-19:
**Describe the bug**
We have an application based on a nrf52840. We see occational `CPU Lockups` while sending BLE notifications every few ms.

It seems to happen more oftnen when `CONFIG_ASSERT=y` is enabled, but also happens without that.
No log message is output, the MCU simply restarts and the `CPU Lockup` flag is set.

The code is used for a measurement device. We run an external ADC at 10 kSamples/s and read its data through SPI DMA transfers. After 30 samples (=> 333 kHz) we trigger an Interrupt Handler which copies the data into another buffer.
Then we we release a semaphore. A Kernel Thread is waiting for this semaphore so it can grab the data from the buffer.
If there is enough data for a BLE Notification, it gets created and provided to `bt_gatt_notify()`.

What is the best way to trace this further down?

**Environment :**
 - OS: Ubuntu 20.04,
 - Toolchain: zephyr-sdk-0.12.2
 - Zephyr Project (Tag `2.5.0`)

caco3, 2021-04-20:
I can enforce the `CPU Lockup` with following code in the main loop:
```
#include <logging/log_ctrl.h>

while (1) {
        if (k_uptime_get() > 5000) {
            log_panic();
        }
}
```

caco3, 2021-04-20:
I might have found the culprit:

I have my own logger backend defined but I haven't defined a custom panic handler:
```
static uint8_t bt_output_buf[MAX_MESSAGE_SIZE];

static int putMessageToQueue(uint8_t *data, size_t length, void *ctx) {

}

LOG_OUTPUT_DEFINE(log_output_bt, putMessageToQueue, bt_output_buf, MAX_MESSAGE_SIZE);

static void put(const struct log_backend *const backend, struct log_msg *msg) {
    uint32_t flags;
    flags = LOG_OUTPUT_FLAG_LEVEL | LOG_OUTPUT_FLAG_TIMESTAMP | LOG_OUTPUT_FLAG_CRLF_NONE;
    if (IS_ENABLED(CONFIG_LOG_BACKEND_FORMAT_TIMESTAMP)) {
        flags |= LOG_OUTPUT_FLAG_FORMAT_TIMESTAMP;
    }

    log_output_msg_process(&log_output_bt, msg, flags);
}

const struct log_backend_api log_backend_bt_api = {
    .put = put,
    .put_sync_string = NULL,
    .put_sync_hexdump = NULL,
    .panic = NULL,  // <-- If set to NULL, it will crash on log_panic();
    .init = NULL,
    .dropped = NULL,
};

LOG_BACKEND_DEFINE(log_backend_bt, log_backend_bt_api, true);
```

When I create a panic handler and assign it, it no longer crashes on a `log_panic()` call.

Can somebody confirm this?

carlescufi, 2021-04-20:
>Can somebody confirm this?

I am hoping @nordic-krch can confirm that.

nordic-krch, 2021-04-22:
yes, logger expects that panic handler is part of the backend and calls in unconditionally.

carlescufi, 2021-04-22:
OK, closing this then.

closed 2021-04-22 by carlescufi

caco3, 2021-04-23:
Thanks for the clarification.

Sadly I still experience CPU Lockup Resets, even when I disable my custom Log Backend.
Is there any way to trace down why those Lockups get triggered?
I need to nail it down more and create another bug report here.

sslupsky, 2021-05-15:
@carlescufi you might want to reopen this issue, it seems the original problem of the kernel locking up is not resolved.

I've been experiencing this problem for a couple years now.  When there is a panic, it seems like the panic causes a hard fault which causes another panic and then the kernel gets stuck pending on a semaphore.

carlescufi, 2021-06-11:
@sslupsky I see there's #35316 now instead, so let's track this issue there.