zephyrproject-rtos/zephyr #34375: drivers: can: Can configure fails when  CONFIG_CAN_FD_MODE is enabled

kumarnac, 2021-04-19:
The can_configure API fails when  CONFIG_CAN_FD_MODE is enabled. 
Below API passes the speed for the FD mode as 0 which returns an ERROR computed by can_set_bitrate and  can_calc_timing_data

`static inline int can_configure(const struct device *dev, enum can_mode mode, uint32_t bitrate)
{
if (bitrate > 0) {
       int err = can_set_bitrate(dev, bitrate, 0);
      if (err != 0) {
       return err;
      }
 }
return can_set_mode(dev, mode);
}
`

**To Reproduce**
Always reproducible

**Expected behavior**
API should add a provision for specifying the FAST bit rate  or possibly use the  ARBITRATION bit rate for  DATA  phase(if API changes is not possible)

**Impact**
can_configure  API cannot be used if  CONFIG_CAN_FD_MODE mode is enabled.

alexanderwachter, 2021-04-24:
The can configure function is legacy and not made for CAN-FD. Please use the new API instead:
`can_set_bitrate(...)` and `can_set_mode(...)`

kumarnac, 2021-04-26:
The  can_configure   API needs to  return error when  CAN-FD is enabled and would require changed the API description.

commit 8d5a0664 (2021-12-02): drivers: can: fix can_configure() when CAN-FD is enabled

Currently, can_configure() pass a hard-coded 0 for the data bitrate
(which is only used for CAN-FD), breaking this API for CAN-FD enabled
applications.

Instead pass in the provided bitrate for both arbitration phase and data
phase.

Fixes: #34375
commit 05318b53 (2021-12-10): drivers: can: fix can_configure() when CAN-FD is enabled

Currently, can_configure() pass a hard-coded 0 for the data bitrate
(which is only used for CAN-FD), breaking this API for CAN-FD enabled
applications.

Instead pass in the provided bitrate for both arbitration phase and data
phase.

Fixes: #34375