zephyrproject-rtos/zephyr #34397: Missing libpython3.8.so.1.0 when running debug

zhudsonPHY, 2021-04-19:
**Describe the bug**
west debug fails with error 
```
/home/user/zephyr-sdk-0.12.4/arm-zephyr-eabi/bin/arm-zephyr-eabi-gdb: error while loading shared libraries: libpython3.8.so.1.0: cannot open shared object file: No such file or directory
FATAL ERROR: command exited with status 127: /home/user/zephyr-sdk-0.12.4/arm-zephyr-eabi/bin/arm-zephyr-eabi-gdb /home/user/app/build/zephyr/zephyr.elf -ex 'target remote :2331' -ex 'monitor halt' -ex 'monitor reset' -ex load
```
I followed the getting started guide and the application development steps to create a very simple hello world program outside of the zephyr base directory. It compiles and flashes fine however when trying to get debug functioning I get the above error. I am using a jlink and mimxrt1060_evk.

Based on some stack exchange posts of other people missing that library I installed libpython3.8-dev and that solved the problem. Perhaps this should be added to the list of dependencies? Or did I miss something in the documentation?

**To Reproduce**
Steps to reproduce the behavior:
1. west build
2. west flash
3. west debug

**Expected behavior**
GDB starts with no errors 

**Environment (please complete the following information):**
 - OS: Linux Ubuntu 18.04
 - Toolchain: Zephyr SDK, West

galak, 2021-04-20:
You need to do:
`sudo apt get install python3.8-dev`

galak, 2021-04-20:
This is a known requirement.  You can also utilize the `/home/user/zephyr-sdk-0.12.4/arm-zephyr-eabi/bin/arm-zephyr-eabi-gdb-no-py` version

jfischer-no, 2021-04-20:
Unfortunately, exactly this Python version is no longer available on some distributions.

zhudsonPHY, 2021-04-20:
> You need to do:
> `sudo apt get install python3.8-dev`

If specifically python3.8 is required the getting started documentation should be changed to reflect that. Currently it just calls out installing python3 which on 18.04 installs python3.6.9 

I appreciate that your solution solves my problem, and I had even already found a solution before I posted. The issue here is more with the documentation.

closed 2021-04-21 by galak