zephyrproject-rtos/zephyr #34403: Logging disable function causes Zephyr hard lockup

madscientist159, 2021-04-20:
**Describe the bug**
After the new logging v2 patches were applied to our development tree, existing, funcitonal code using `z_shell_log_backend_disable()` has ceased to function.

Whereas before this call would disable logging on the affected shell:
`z_shell_log_backend_disable(shell->log_backend);`

it now causes a complete hard lockup of the entire system immediately upon execution.   No debug messages are printed, none of the shells are responsive, and all network communication is halted.

What have you tried to diagnose or workaround this issue?

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
Logging should be disabled on the specified shell and the system should continue to function normally in all other respects.

**Impact**
Showstopper.  Will need to revert the patches in question to restore our terminal application to working status.

nordic-krch, 2021-04-20:
Can you add more details? Platform? I assume that logging v2 is not used (no change in logging configuration). From which context it is disabled?

I tried to add shell command that disables and enables shell log backend and that runs fine on `qemu_cortex_m3` with shell over uart.

madscientist159, 2021-04-20:
@nordic-krch This is on a custom LiteX platform [1], but I would expect the same problem to show up on the RISC-V LiteX demo if you enable the telnet shell.  That's likely a difference from your test environment, both the UART and telnet shells are enabled on this platform.

The call is from the shell thread itself.  See https://gitlab.raptorengineering.com/kestrel-collaboration/kestrel-firmware/zephyr-firmware/-/blob/master/kestrel/src/shell.c#L134 for the exact call.

[1] https://gitlab.raptorengineering.com/kestrel-collaboration/kestrel-firmware/zephyr-rtos/-/tree/master/boards/powerpc/litex_kestrel

madscientist159, 2021-04-20:
@nordic-krch Here are the logging-related configuration options from the build:

```
CONFIG_SOC_LOG_LEVEL_INF=y
CONFIG_SOC_LOG_LEVEL=3
CONFIG_ARCH_LOG_LEVEL_INF=y
CONFIG_ARCH_LOG_LEVEL=3
CONFIG_MPU_LOG_LEVEL_INF=y
CONFIG_MPU_LOG_LEVEL=3
CONFIG_KERNEL_LOG_LEVEL_INF=y
CONFIG_KERNEL_LOG_LEVEL=3
CONFIG_UART_CONSOLE_LOG_LEVEL_INF=y
CONFIG_UART_CONSOLE_LOG_LEVEL=3
CONFIG_ETHERNET_LOG_LEVEL_DEFAULT=y
CONFIG_ETHERNET_LOG_LEVEL=3
CONFIG_GPIO_LOG_LEVEL_INF=y
CONFIG_GPIO_LOG_LEVEL=3
CONFIG_SPI_LOG_LEVEL_INF=y
CONFIG_SPI_LOG_LEVEL=3
CONFIG_FLASH_LOG_LEVEL_INF=y
CONFIG_FLASH_LOG_LEVEL=3
CONFIG_FS_LOG_LEVEL_INF=y
CONFIG_FS_LOG_LEVEL=3
CONFIG_LOG=y
CONFIG_LOG_MODE_IMMEDIATE=y
CONFIG_LOG_IMMEDIATE=y
CONFIG_LOG_RUNTIME_FILTERING=y
CONFIG_LOG_DEFAULT_LEVEL=3
CONFIG_LOG_OVERRIDE_LEVEL=0
CONFIG_LOG_MAX_LEVEL=4
CONFIG_LOG_FUNC_NAME_PREFIX_DBG=y
CONFIG_LOG_PRINTK=y
CONFIG_LOG_DOMAIN_ID=0
CONFIG_LOG_CMDS=y
CONFIG_LOG2_USE_VLA=y
CONFIG_NET_BUF_LOG_LEVEL_INF=y
CONFIG_NET_BUF_LOG_LEVEL=3
CONFIG_NET_L2_ETHERNET_LOG_LEVEL_DEFAULT=y
CONFIG_NET_L2_ETHERNET_LOG_LEVEL=3
CONFIG_NET_ARP_LOG_LEVEL_DEFAULT=y
CONFIG_NET_ARP_LOG_LEVEL=3
CONFIG_NET_IPV6_LOG_LEVEL_DEFAULT=y
CONFIG_NET_IPV6_LOG_LEVEL=3
CONFIG_NET_ICMPV6_LOG_LEVEL_DEFAULT=y
CONFIG_NET_ICMPV6_LOG_LEVEL=3
CONFIG_NET_IPV6_NBR_CACHE_LOG_LEVEL_DEFAULT=y
CONFIG_NET_IPV6_NBR_CACHE_LOG_LEVEL=3
CONFIG_NET_IPV4_LOG_LEVEL_DEFAULT=y
CONFIG_NET_IPV4_LOG_LEVEL=3
CONFIG_NET_ICMPV4_LOG_LEVEL_DEFAULT=y
CONFIG_NET_ICMPV4_LOG_LEVEL=3
CONFIG_NET_DHCPV4_LOG_LEVEL_DEFAULT=y
CONFIG_NET_DHCPV4_LOG_LEVEL=3
CONFIG_NET_TCP_LOG_LEVEL_DEFAULT=y
CONFIG_NET_TCP_LOG_LEVEL=3
CONFIG_NET_TCP_BACKLOG_SIZE=1
CONFIG_NET_UDP_LOG_LEVEL_DEFAULT=y
CONFIG_NET_UDP_LOG_LEVEL=3
CONFIG_NET_MGMT_EVENT_LOG_LEVEL_DEFAULT=y
CONFIG_NET_MGMT_EVENT_LOG_LEVEL=3
CONFIG_NET_STATISTICS_LOG_LEVEL_DEFAULT=y
CONFIG_NET_STATISTICS_LOG_LEVEL=3
CONFIG_NET_LOG=y
CONFIG_NET_PKT_LOG_LEVEL_DEFAULT=y
CONFIG_NET_PKT_LOG_LEVEL=3
CONFIG_NET_CORE_LOG_LEVEL_DEFAULT=y
CONFIG_NET_CORE_LOG_LEVEL=3
CONFIG_NET_IF_LOG_LEVEL_DEFAULT=y
CONFIG_NET_IF_LOG_LEVEL=3
CONFIG_NET_TC_LOG_LEVEL_DEFAULT=y
CONFIG_NET_TC_LOG_LEVEL=3
CONFIG_NET_UTILS_LOG_LEVEL_DEFAULT=y
CONFIG_NET_UTILS_LOG_LEVEL=3
CONFIG_NET_CONTEXT_LOG_LEVEL_DEFAULT=y
CONFIG_NET_CONTEXT_LOG_LEVEL=3
CONFIG_NET_CONN_LOG_LEVEL_DEFAULT=y
CONFIG_NET_CONN_LOG_LEVEL=3
CONFIG_NET_ROUTE_LOG_LEVEL_DEFAULT=y
CONFIG_NET_ROUTE_LOG_LEVEL=3
CONFIG_NET_HTTP_LOG_LEVEL_DEFAULT=y
CONFIG_NET_HTTP_LOG_LEVEL=3
CONFIG_NET_CONFIG_LOG_LEVEL_DEFAULT=y
CONFIG_NET_CONFIG_LOG_LEVEL=3
CONFIG_NET_SOCKETS_LOG_LEVEL_DEFAULT=y
CONFIG_NET_SOCKETS_LOG_LEVEL=3
CONFIG_NET_CONNECTION_MANAGER_LOG_LEVEL_DEFAULT=y
CONFIG_NET_CONNECTION_MANAGER_LOG_LEVEL=3
CONFIG_SHELL_LOG_LEVEL_INF=y
CONFIG_SHELL_LOG_LEVEL=3
CONFIG_SHELL_BACKEND_SERIAL_LOG_MESSAGE_QUEUE_TIMEOUT=100
CONFIG_SHELL_BACKEND_SERIAL_LOG_MESSAGE_QUEUE_SIZE=10
CONFIG_SHELL_BACKEND_SERIAL_LOG_LEVEL_DEFAULT=y
CONFIG_SHELL_BACKEND_SERIAL_LOG_LEVEL=5
CONFIG_SHELL_TELNET_LOG_MESSAGE_QUEUE_TIMEOUT=100
CONFIG_SHELL_TELNET_LOG_MESSAGE_QUEUE_SIZE=10
CONFIG_SHELL_TELNET_INIT_LOG_LEVEL_DEFAULT=y
CONFIG_SHELL_TELNET_INIT_LOG_LEVEL=5
CONFIG_SHELL_TELNET_LOG_LEVEL_INF=y
CONFIG_SHELL_TELNET_LOG_LEVEL=3
CONFIG_SHELL_LOG_BACKEND=y
```

madscientist159, 2021-04-20:
@galak What additional information do you need from me at this point?  I've provided links to everything and the full configuration.  Would be nice to get this feature working again.

nordic-krch, 2021-04-21:
Further questions:
- is there any change in optimization? Maybe optimization is off?
- do you stack guard enabled, is there any on that platform?
- if not stack guard can you try to increase shell thread stack?
- if optimization is off, can you confirm that `CONFIG_LOG2_ALWAYS_RUNTIME` is set

The thing is that logging v2 code is supposed to be compiled out when not enabled. However, it is using compile time switches (IS_ENABLED) rather than ifdefs and I've seen that some compilers are not handling that properly and that leads to increased stack usage when optimization is off. Option `CONFIG_LOG2_ALWAYS_RUNTIME` fixes that and should be enabled by default when `NO_OPTIMIZATIONS` is set.

madscientist159, 2021-04-21:
@nordic-krch

- No change in optimization.
- Stack guard is enabled.  It does not trip.
- Already tried increasing stack size -- no effect

Also tested the following:
- Confirmed reverting the logger v2 patchset [1] restored all functionality
- Apparent race condition was noted with the v2 patchset applied.  If two consoles are printing large amounts of text at the same time, the system will deadlock at some point.
- When the system goes down, it's in an IRQ-critical section -- the timer interrupt is disabled at the time of lockup.

nordic-krch, 2021-04-22:
It is very strange. I've been looking at the code and comparing code size of logging and shell before and after log v2 and it differs only by few bytes as expceted. Can you try to revert commit which adapts shell to log2: https://github.com/zephyrproject-rtos/zephyr/pull/31535/commits/b3bcad5101d724eab3346001b48d83e8dab5caaa ?

Then if that does not help go back to https://github.com/zephyrproject-rtos/zephyr/pull/31535/commits/jkthis is refactor of the logging before introducing log2. One thing there, after commit checkout you will need to remove  line `subsys/logging/log_core.c:639` which must have been accidentally added during upmerges. 

I've also noticed that you are using immediate mode (`LOG_MODE_IMMEDIATE`) it is a synchronous, blocking mode and logging and shell is called from any context with shell transport being called with irq locked. Not sure if telnet supports it but that should come out before as well.

carlescufi, 2021-04-29:
@madscientist159 is this still an issue with the latest master?

madscientist159, 2021-04-29:
@carlescufi I'll check later tonight.  It looks like the only change was a stack increase?

carlescufi, 2021-05-06:
@madscientist159 ping

madscientist159, 2021-05-09:
@carlescufi After thorough testing, it looks like this bug is resolved in latest master.  Thanks all for the quick diagnosis and fix!

madscientist159, 2021-05-09:
@carlescufi Unfortunately it looks like the binary being tested was a hybrid of the new and old code due to the way Zephyr's build system references the source trees...a clean build is still showing the crash.

madscientist159, 2021-05-10:
I was able to gather some potentially useful debug information over the weekend.

First, I am unable to replicate the issue in QEMU, using the same CPU architecture (ppc64le).  The major caveat there is, of course, that QEMU does not emulate the LiteX Ethernet hardware, and that QEMU runs a couple of orders of magnitude faster than the LiteX soft CPU.  There are also some Kconfig differences needed to deal with the QEMU virtual hardware.

Second, it appears that on the actual hardware we're basically getting stuck in `memcpy` on the following call chain:

```
msg_from_fifo: (subsys/shell/shell_log_backend.c:52) / flush_expired_messages: (subsys/shell/shell_log_backend.c:79)
k_msgq_get: (kernel/msg_q.c:220)
memcpy
```

nordic-krch, 2021-05-11:
@madscientist159 thanks for investigation. So it turned out that the problem was not related to logging overhaul. `msgq` should not be accessed when immediate mode is in use and that was the case when shell log backend was disabled and `fifo_flush` was called.

In #35084 i made a cleanup to  have fifo resetting done in the same place for v1 and v2. For v1 deferred mode it was moved to shell log backend enable function.

PR #35084 (2021-05-13): shell: log_backend: Fix fifo cleanup
When shell log backend was disabled fifo used only in
deferred mode was unconditionally flushed which lead to
errors in immediate mode where fifo was not present. On
the other hand, in case of LOG2_MODE_DEFERRED fifo (mpsc_pbuf)
was initialized in enable stage.

In order to clean things up, deferred v1 fifo flushing and
deferred v2 mpsc pbuf initialization were moved to fifo_reset
function which is called when shell log backend is enabled.

Fixes #34403.
closed 2021-05-13 by carlescufi