zephyrproject-rtos/zephyr #34409: mDNS response on link local when using DHCPv4 and AutoIP/Static IP

n-lohrmann, 2021-04-20:
**Describe the bug**
When using device that is configured for both DHCPv4 and AutoIP/Static IP, all mDNS responses have link local as source. These responses are not received by host on DHCP network. 

**To Reproduce**
CONFIG_NET_DHCPV4=y                         
CONFIG_NET_IPV4_AUTO=y
CONFIG_NET_IF_UNICAST_IPV4_ADDR_COUNT=2 
CONFIG_DNS_SD=y
CONFIG_MDNS_RESPONDER=y
CONFIG_MDNS_RESPONDER_DNS_SD=y

**Expected behavior**
Expect the source to be set as the address set in the query response, or best matching from source of mDNS query.

**Logs and console output**
[image]

**Environment (please complete the following information):**
Zephyr v2.5.0
Commit: fe7c2efca800a0cf1bccf23aefe08b3c4beb88bf

**Additional context**
Added a little workaround, that sets the local IP 
[mdns.txt](https://github.com/zephyrproject-rtos/zephyr/files/6341772/mdns.txt)

jukkar, 2021-04-21:
@n-lohrmann Are you able to receive another mDNS requests after this? If you have net-shell enabled, what does `net conn` show after your fix?
The fix looks okeish but might cause some unintended side effects because the local address is used in bind. The default bind is done at mDNS init so it might be possible to alter the address as you have done. Just to want to make sure everything works as expected after this change.

n-lohrmann, 2021-04-21:
> @n-lohrmann Are you able to receive another mDNS requests after this? If you have net-shell enabled, what does `net conn` show after your fix?
> The fix looks okeish but might cause some unintended [...]

Yup, you are right, it only works for one response, then stops replying. 
[net-conn-mdns.txt](https://github.com/zephyrproject-rtos/zephyr/files/6350810/net-conn-mdns.txt)

jukkar, 2021-04-21:
Yep, the workaround you had overwrites the local listening address which now prevents the incoming traffic to 224.0.0.251, so this is no good.
What IP addresses you have assigned to your network interface. Can you attach output of `net iface` command here.

To me it looks like we should get proper src address from `net_if_ipv4_select_src_addr()`, so if you have ll address and normal IP address, the normal one should be preferred, and now that does not seem to happen.

I have not heard people using IPv4 auto addresses with mDNS, this looks like a corner case and you were the unlucky person to have this combination.

jukkar, 2021-05-10:
@n-lohrmann I have a fix proposal in #35047, can you try if it works for you and report back.

n-lohrmann, 2021-05-10:
> @n-lohrmann I have a fix proposal in #35047, can you try if it works for you and report back.

Looks good to me :+1:

PR #35047 (2021-05-11): net: if: Do not check IPv4 multicast addr when selecting src addr
When destination address is a multicast address, select the source
address from non link local address first.

This means that for example if we are responding to mDNS query,
and we have both normal IPv4 and LL (169.254.x.y) address set for
the interface, we are now able to select the normal address instead
of the LL one.

Fixes #34409
commit 6485e7d0 (2021-05-11): net: if: Do not check IPv4 multicast addr when selecting src addr

When destination address is a multicast address, select the source
address from non link local address first.

This means that for example if we are responding to mDNS query,
and we have both normal IPv4 and LL (169.254.x.y) address set for
the interface, we are now able to select the normal address instead
of the LL one.

Fixes #34409
closed 2021-05-11 by jukkar