zephyrproject-rtos/zephyr #34413: Improve __used attribute to actually keep requested function/variable

nordic-krch, 2021-04-20:
**Is your feature request related to a problem? Please describe.**
There is `__used` specified in `include/toolchain/gcc.h` which is setting attribute `used`. The idea is to keep marked function/variable in the binary even when it is not used. However, linker still removes it (thanks to garbage collection?). 

**Describe the solution you'd like**
Similarly as `__used`, `__func_used` could be added which adds code to dedicated section which is kept. Something like:
```
#define __func_used __attribute__((used)) __attribute__((section(".used_code")))
```

**Additional context**
So far for rpc `--no-gc-sections` was used to keep rpc function but that acts globally and generates issues as zephyr has this approach where api is available but code is not compiled in and relies on `IS_ENABLED`. See #26848.

tejlmand, 2021-04-21:
not convinced that the right thing to do here is to actually place all of those functions in same output section.

Of course it has the benefit to those users who don't care in which output section the code is placed.
They won't have to specify an new section in the linker script, which is often the case today when users specify custom sections.

An alternative that users can do in their CMake file, is to tell the linker that a particular symbol should be kept, even if it appears unused, by doing:
```
zephyr_ld_options("-Wl,--undefined=<symbol>")
```
But of course, this is a GNU ld centric solution, and also keeps this knowledge a bit separated from the actual code.
But it has the advantage of not relying on specific sections in the linker script.

nordic-krch, 2021-04-21:
What about following approach:
- create section, e.g. `keep_functions`
- then macro:
```
define __func_used(name) \ 
   static void *name##_ptr __attribute__ ((section(".keep_functions"))) \                           
    __attribute__((used)) = (void *)name 
```
- then use if after the function, like:
```
void foo(void)
{
    printk("test");
}
__func_used(foo);
```

nordic-krch, 2021-04-27:
@tejlmand @nashif any opinions?

tejlmand, 2021-04-27:
> What about following approach:
> 
>     * create section, e.g. `keep_functions`
> 
>     * then macro:

Not sure how that changes my main concern ?
> not convinced that the right thing to do here is to actually place all of those functions in same output section.

Of course, as mentioned https://github.com/zephyrproject-rtos/zephyr/issues/34413#issuecomment-823834066
I also see so benefits.

Today's approach is clear in the sense that developers have to make a conscious choice that this particular function should go into a specific section that has `KEEP`, if needed.
What about variables ?

nordic-krch, 2021-04-27:
> Not sure how that changes my main concern ?

With this approach, function code is together with rest of the code and only pointer lands in the section so it can be kept hidden in shared section.

manoj153, 2021-09-01:
Hey guys, just want to know this attr `__used` usage is portable across all SoC?

tejlmand, 2021-09-02:
> Hey guys, just want to know if this attr `__used` usage is portable across all SoC?

The __used is setting an attribute, so it's dependent on the toolchain, not the SoC.

manoj153, 2021-09-02:
> > Hey guys, just want to know if this attr `__used` usage is portable across all SoC?
> 
> The __used is setting an attribute, so it's dependent on the toolchain, not the SoC.

understand, therefore diffrent SoC might use different toolchain, thus this is the best / portable way to do it at the moment?

closed 2022-07-01 by carlescufi