zephyrproject-rtos/zephyr #34419: significant build time increase with new logging subsystem

galak, 2021-04-20:
We are seeing large increases in CI times associated with the new logging subsystem:

Comparing https://buildkite.com/zephyr/zephyr-daily/builds/304#_ [before logging changes merged] to https://buildkite.com/zephyr/zephyr-daily/builds/305#_.

We can take a random build like 6/400 and see an increase from 33m41s to 42m28s.

Looking at git commits this seems to be between 67fade04a0cf (before logging) and f1bb20f6b43b (logger builds)

Looking at micro level if we take a test like samples/net/vlan and build file subsys/net/ip/tcp2.c.  We see a huge increase in C-preprocessor time (on my system going from 0.12s to 2.8s).  [This file was picked since it showed such a large increase]

carlescufi, 2021-04-20:
The increase is visible also when building locally.

nashif, 2021-04-21:
@nordic-krch  do we have an idea what is going on?

nordic-krch, 2021-04-21:
currently, both versions are compiled since `IS_ENABLED` compile-time switch is used also logging v2 is more extensively using preprocess (analyzing each argument and determining its type and size). Tomorrow i will be looking into speeding things up.

nordic-krch, 2021-05-19:
I've added another (final?) speed up to logging. Seems that there are also other factors in build time increase.

I've been testing with:
```
 ccache -C; time scripts/twister -p frdm_k64f -b -c -s tests/kernel/common/kernel.common
```
And getting following results:

Before | After | Without logging v2 | v2.5.0
------------ | ------------- | ------------ | ------------
35.17s | 32,6 | 29.6 | 23.25

It means that something else contributed to the build time increase from v2.5.0.

galak, 2021-05-19:
How are you "disabling" logging v2?

nordic-krch, 2021-05-19:
by deleting content of this if statement:
https://github.com/zephyrproject-rtos/zephyr/blob/db4359cfd2492cf658f18cf0cd49736d7a6788b6/include/logging/log_core.h#L312

PR #34489 (2021-04-22): logging: Tweaks to speed up compilation
Logging attempts to do most of the work at compile time instead
of preprocessor by using conditions which can be resolved at
compile time (e.g. if (IS_ENABLED(...))). Apparently, such extensive
use significantly loads the compiler since all paths are compiled
even though cut from final binary. Patch reduces it by replacing
some compile time switch with preprocessor.

Handling of function name prefix has been moved to preprocessor.
Immediate logging v2 has also been moved to preprocessor.

Fixes #34419.
PR #35431 (2021-05-19): logging: Add empty Z_LOG_MSG2_SIMPLE_CREATE version
Add empty Z_LOG_MSG2_SIMPLE_CREATE when CONFIG_LOG_SPEED=n. It
allows to reduce compilation time by removing code at preprocessor
stage. Previously functionality was the same, compiler was removing
unused code but because logging is widely use that had a visible
impact on compile time.

This optimization speeds up compilation and cuts ~40% of logging compile time. I don't see a place where logging v2 compilation could be further improved.

I've been testing with:
```
 ccache -C; time scripts/twister -p frdm_k64f -b -c -s tests/kernel/common/k [... cut]
closed 2021-05-19 by galak