zephyrproject-rtos/zephyr #34461: Unable to use PWM pins with STM Nucleo H743ZI

manuaatitya, 2021-04-21:
# Unable to use PWM with STM Nucleo H743ZI

Hi, I'm trying out to control a servo motor attached to the STM Nucleo H743Zi board running on Zephyr RTOS but the PWM pins seems to be configured improperly and I get no signal from the pins.

The issue I'm facing is very similar to [this issue](https://github.com/zephyrproject-rtos/zephyr/issues/26607) however, the solution provided here doesn't seem to work.

I'm attaching my [source code](https://pastebin.com/G1U1Ty88) here for your reference.

**Environment**
 - OS: Ubuntu 16.04 
 - Toolchain  Zephyr SDK

**A few important points regarding my source code**
- it comes with a preconfigured PWM pin called PWM_12 as seen from the [dts file](https://drive.google.com/file/d/1MMEBw6EpnEb3pvXXeHAoFTbZ7n96eubV/view?usp=sharing) of the board.
- I'm not using any overlays for the project and using the default configuration.
- [Drive link](https://drive.google.com/drive/folders/1Vf9VEkpT_a9_drTIr7VKIisSCv4wlobR?usp=sharing) with all the device tree files for the board for easy access.

The code actually builds without errors but on testing with an oscilloscope, there's no PWM pin output from the pin.

Any help to resolve this issue will be appreciated.

FRASTM, 2021-04-21:
I check with the samples/basic/blinky_pwm/ on a nucleo_h723zg which is a like the nucleo h743zi with similar PWM config. The PWM output of the Timer12 channel1 is on pin **pb14**. On the nucleo board  (MB1364) the PB14 is the red led. Enabling the PWM of the the TIMER 12 channel 1, will directly control the red led.

IshanBhatnagar14, 2021-04-21:
Nope , nothing like that is happening , are you sure , we do not see any red led blinking ?

manuaatitya, 2021-04-21:
Hi @FRASTM, the issue is there is not output from the PWM pin and hence we do not see any blining/flashing of leds, 

Now I also tried out the blinky_pwm_led example as you suggested, but I get an error on build saying zephyr/include/drivers/pwm.h:374 "undefined reference to __device_ord_24'

FRASTM, 2021-04-22:
what is your zephyr branch ? With the latest (ecf89143da0df6f9bd7346149415830372798430) I can compile w/o any error
_west build -p auto -b nucleo_h743zi samples/basic/blinky_pwm_

-- west build: making build dir ./zephyrproject/zephyr/build pristine
-- west build: generating a build system
Including boilerplate (Zephyr base): ./zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: ./zephyrproject/zephyr/samples/basic/blinky_pwm
-- Zephyr version: 2.5.99 (./zephyrproject/zephyr)
-- Found Python3: /usr/bin/python3.6 (found suitable exact version "3.6.9") found components: Interpreter 
-- Found west (found suitable version "0.7.2", minimum required is "0.7.1")
-- Board: nucleo_h743zi
-- Cache files will be written to: ./.cache/zephyr
-- Using toolchain: zephyr 0.11.3 (./zephyr-sdk-0.11.3)
-- Found dtc: ./zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: ./zephyrproject/zephyr/boards/arm/nucleo_h743zi/nucleo_h743zi.dts
-- Generated zephyr.dts: ./zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: ./zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: ./zephyrproject/zephyr/build/zephyr/include/generated/device_extern.h
Parsing ./zephyrproject/zephyr/Kconfig
Loaded configuration './zephyrproject/zephyr/boards/arm/nucleo_h743zi/nucleo_h743zi_defconfig'
Merged configuration './zephyrproject/zephyr/samples/basic/blinky_pwm/prj.conf'
Configuration saved to './zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to './zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 9.2.0
-- The CXX compiler identification is GNU 9.2.0
-- The ASM compiler identification is GNU
-- Found assembler: ./zephyr-sdk-0.11.3/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: ./zephyrproject/zephyr/build
-- west build: building application
[1/145] Preparing syscall dependency handling
[138/145] Linking C executable zephyr/zephyr_prebuilt.elf
[145/145] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       25028 B         2 MB      1.19%
            SRAM:        4736 B       512 KB      0.90%
            DTCM:          0 GB       128 KB      0.00%
     BACKUP_SRAM:          0 GB         4 KB      0.00%
        IDT_LIST:          0 GB         2 KB      0.00%

manuaatitya, 2021-04-24:
Hi @FRASTM, it seems that the codebase I was not using was not in the latest zephyr branch. When reverted to the latest branch, I'm able to build the app.
The issue is resolved now.
Thanks for the help.

I'm closing the issue.

closed 2021-04-24 by manuaatitya