zephyrproject-rtos/zephyr #34473: Add Requirements repository with infrastructure and placeholder requirements

laurenmurphyx64, 2021-04-21:
Initialize the Zephyr Project GitHub repository for requirements with placeholder requirements and put in applicable infrastructure.

carlescufi, 2021-04-22:
@laurenmurphyx64 what exactly do you mean by this? There is a Docker image and a Getting Started Guide, so I am unsure what specifically you are referring to.

laurenmurphyx64, 2021-04-22:
> @laurenmurphyx64 what exactly do you mean by this? There is a Docker image and a Getting Started Guide, so I am unsure what specifically you are referring to.

@carlescufi Sorry, maybe documentation isn't the best tag to put it under. A [separate requirements catalog repository](https://github.com/zephyrproject/reqmgmt) is part of our safety collateral for functional safety certification. @AmberHibberd asked for a card for 2.6 so I made one.

nashif, 2021-04-28:
> @laurenmurphyx64 what exactly do you mean by this? There is a Docker image and a Getting Started Guide, so I am unsure what specifically you are referring to.

discussed with @carlescufi , this is unrelated to python requirements file.

nashif, 2021-04-28:
repo created.

closed 2021-04-28 by nashif