zephyrproject-rtos/zephyr #34474: MPS2-AN385 SRAM does not match what the documentation page says

AfflatusX, 2021-04-21:
**Describe the bug**
We're using MPS2-AN385 + QEMU for unit tests, and noticed when building for that board, both FLASH and SRAM reports 4MB for region size. According to https://docs.zephyrproject.org/2.5.0/boards/arm/mps2_an385/doc/index.html it has 8MB SRAM. Is the discrepency expected? 

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b mps2_an385 samples/net/sockets/echo_server
2. build reports:

           FLASH:      127596 B         4 MB      3.04%
            SRAM:       72120 B         4 MB      1.72%

**Expected behavior**
It should match the spec mentioned in the document.

**Impact**
We need more than 4MB SRAM, this is stopping us from using the board for unit testing purpose.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK
 - Commit SHA or Version used - 2.3.0

ioannisg, 2021-04-22:
will take a look

ioannisg, 2021-04-23:
I took a look briefly, into the QEMU implementation of AN385, 386, here: https://qemu.readthedocs.io/en/latest/system/arm/mps2.html

It looks like with respect to SRAM (memory in the 0x20000000 area) QEMU implements the AN385 as is.

Now, in the AN385 (https://developer.arm.com/documentation/dai0385/d/)  it is stated that in the respective area only 4MB are implemented (even though 8MB are reserved).

So it seems to me that only 4M should be available to use.

ioannisg, 2021-04-23:
I think the 8MB in the documentation refers to both ZBTSRAM blocks (1 and 2,3), each have 4 MB available.
I'll send a patch to correct the documentation in Zephyr.

@gromero @galak let me know if you have any comments.

closed 2021-04-23 by MaureenHelm

gromero, 2021-04-23:
Hi @ioannisg. Sorry to be late for the party : )

Your understanding about the 8MB in the docs referring actually to 4MB of flash (ZBTSRAM 1) + 4MB used as RAM (ZBTSRAM 2_3) and as it's sort of displayed by Zephyr compile statistics as posted by @AfflatusX  above, where FLASH = ZBTSRAM 1 and SRAM = ZBTSRAM 2_3, looks correct to me, so I agree with the fix in  #34521 ("boards: arm: mps2_an385: correct flash and SRAM sizes").

I just have two comments about it.

In the note `(in ZBTSRAM 1, starting at address 0x00400000)` I understand that the correct starting address is actually `0x0`.

By inspecting the DTS for that board I think that it can be improved by:

a) Renaming `sram0` to `sram2_3` and `flash0` to `sram1`, then updating `zephyr,sram` and `zephyr,flash` accordingly, similarly to what @galak suggested to me for the AN521: https://github.com/zephyrproject-rtos/zephyr/pull/32643

b) Adding the description for the 16MB PSRAM memory region so it can be available by the OS when it checks the devicetree (take also a look at the DTS for the AN521 board where similar region is available too).

Cheers,
Gustavo

PR #34521 (2021-04-23): mps2_an385: fix documentation about SRAM size, and some doc fixes in ARM boards
Fixes #34474