zephyrproject-rtos/zephyr #34482: net_tunnel_virtual:frdm_k64f: build failure

hakehuang, 2021-04-22:
**Describe the bug**
Build failure for tests/net/virtual cases

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=frdm_k64f
3. make
4. See error

**Expected behavior**
build pass

**Impact**
virtual network testing

**Logs and console output**

`

/build/src/workspace/test_jenkins_master/zephyr/tests/net/virtual/build_283b1e9/zephyr/misc/generated/extra_kconfig_options.conf:2: warning: malformed string literal in assignment to NET_TEST_TUNNEL_MY_ADDR (defined at /build/src/workspace/test_jenkins_master/zephyr/tests/net/virtual/Kconfig:14). Assignment ignored.

/build/src/workspace/test_jenkins_master/zephyr/tests/net/virtual/build_283b1e9/zephyr/misc/generated/extra_kconfig_options.conf:3: warning: malformed string literal in assignment to NET_TEST_TUNNEL_NAME (defined at /build/src/workspace/test_jenkins_master/zephyr/tests/net/virtual/Kconfig:20). Assignment ignored.

/build/src/workspace/test_jenkins_master/zephyr/tests/net/virtual/build_283b1e9/zephyr/misc/generated/extra_kconfig_options.conf:4: warning: malformed string literal in assignment to NET_TEST_TUNNEL_PEER_ADDR (defined at /build/src/workspace/test_jenkins_master/zephyr/tests/net/virtual/Kconfig:8). Assignment ignored.

error: Aborting due to Kconfig warnings

Parsing /build/src/workspace/test_jenkins_master/zephyr/tests/net/virtual/Kconfig

Loaded configuration '/build/src/workspace/test_jenkins_master/zephyr/boards/arm/frdm_k64f/frdm_k64f_defconfig'

Merged configuration '/build/src/workspace/test_jenkins_master/zephyr/tests/net/virtual/prj.conf'

Merged configuration '/build/src/workspace/test_jenkins_master/zephyr/tests/net/virtual/build_283b1e9/zephyr/misc/generated/extra_kconfig_options.conf'

CMake Error at /build/src/workspace/test_jenkins_master/zephyr/cmake/kconfig.cmake:264 (message):

  command failed with return code: 1

Call Stack (most recent call first):

  /build/src/workspace/test_jenkins_master/zephyr/cmake/app/boilerplate.cmake:555 (include)

  /build/src/workspace/test_jenkins_master/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)

  /build/src/workspace/test_jenkins_master/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:35 (include_boilerplate)

  CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!

make: *** No targets specified and no makefile found.  Stop.

script returned exit code 2
`

**Environment (please complete the following information):**
 - OS: (e.g. Linux, )
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: zephyr-v2.5.0-2384-gfacd0f57db

galak, 2021-04-22:
Tried this with zephyr-v2.5.0-2532-ga379196b48ed and seems to build fine.  So maybe the issue has been fixed.

galak, 2021-04-22:
@hakehuang closing this, if you can still reproduce please re-open.

closed 2021-04-22 by galak