zephyrproject-rtos/zephyr #34492: Logging still broken with SOF

lyakh, 2021-04-22:
**Description**
When built with Zephyr, SOF logs, as generated by the logtool, appear badly corrupted.

**To Reproduce**
Build and run SOF with Zephyr, print logs per
```
sudo tools/logtool.py -e sof
```

**Expected behavior**
Properly formatted consistent logs should be printed.

**Impact**
Debugging becomes impossible.

**Logs and console output**
Example output:
> 23086196: pipe reset
> 23092510: dai_get type 3 index 1 new sref 2
> 23093828: dai_put type 3 index 1 new sref 1
> d080002
> 23096241: dai_config() dai type = 3 index = 1
> 23097545: dai_config(), channel = -1
> ef 3
> 23111526: dai_get type 3 index 1 new sref 2
> x4001
> 23112817: dai_put type 3 index 1 new sref 1
> , sscr1 = 0xd0723115214: dai_config() dai type = 3 index = 1
> 00000, sscr3 = 0x07070000, ssioc = 0x00000020
> 23116497: dai_config(), channel = -1
> 0004002, sspsp2 = 0x00

andyross, 2021-04-22:
This certainly looks like messages being interleaved across the two CPUs.  Did this regress with the recent LOG2 work?  Logging has always been atomic.  Pull in @nordic-krch and @npitre who know more about the new code.

A quick review gets a little confused by all the layers of macros, but the basic architecture doesn't seem to be any different.  Space in the output buffer is allocated by what looks like a correctly-spinlocked call, filled, and then finalized later.   I don't see much opportunity to get this wrong.

I'll see if I can put together a non-SOF exerciser.

nordic-krch, 2021-04-23:
In the config above logv2 is not used (`CONFIG_LOG_MODE_DEFERRED=y`). Can you bisect or start by trying commit before logging overhaul (https://github.com/zephyrproject-rtos/zephyr/commit/67fade04a0cf465d335f9392b86ad9db386796d9) ?

lyakh, 2021-04-23:
@andyross Sorry, for a confusion: this happens with a single core.
@nordic-krch this has been broken for a rather long time, for months. I mentioned it several times during Zephyr-SOF syncs and I had the impression that this was known an worked upon and almost fixed

andyross, 2021-04-23:
Were you seeing this prior to SMP work even?  That sounds like something entirely different.  Maybe this is getting mixed up with the pre-existing SOF trace output in some module or another (similar to the header collision with the cache pointer API, etc...)?  I'll see about making this happen on x86_64 so @nordic-krch has something to test against, but if it's not SMP we may need to look in the platform layer or SOF.

lyakh, 2021-04-26:
A complete .config for a cAVS 1.5 / Apollolake SOF build: 
[apl.config.txt](https://github.com/zephyrproject-rtos/zephyr/files/6379021/apl.config.txt)

lyakh, 2021-05-10:
@marc-hb additional information: with a test, that (supposedly) generated a DMIC IRQ flood I got a log as printed by logtool.py:
```
, OUTSTA347542234: task cancel 0x9e058ba8 0x1fffa0c0U
036
, OUTSTA347543716: dmic_irq_handler(), OUTSTAT0 = 0x28000000, OUTSTAT1 = 0x80000000
top(), dmic_active_fifos = -32062
, OUTSTA347545748: dmic_irq_handler(): full fifo A or PDM overrun
A347547181: dmic_stop(), dmic_active_fifos = -32088
```
those lines are obviously corrupted, they are truncated, but - this sequence of lines is then repeated **exactly** continuously (only with counters incrementing). Which means the corruption isn't random but perhaps is related to some string length constraints. You know, that before actual log lines logtool delivers multiple 0-bytes. So, maybe those together with the actual lines exceed some length. Could you try to figure out where those 0-bytes are coming from?

marc-hb, 2021-05-12:
@lyakh I always saw the same kind of garbage with `intel_adsp_cavs15/tools/logtool.py` and never with `intel_adsp_cavs15/tools/adsplog.py`. What do you see with the latter?

marc-hb, 2021-05-12:
@lyakh confirmed that #35193 "works for him" (tm)

PR #35193 (2021-05-12): intel_adsp_cavs15/ logtool.py: fix support for multiple NULLs
Stop printing garbage when strings are terminated with more than one
zero byte.

Thanks Andy for the diagnosis (and the better adsplog.py)

Fixes #34492
commit d3ae4e9a (2021-05-12): intel_adsp_cavs15/ logtool.py: fix support for multiple NULLs

Stop printing garbage when strings are terminated with more than one
zero byte.

Thanks Andy for the diagnosis (and the better adsplog.py)

Fixes #34492
closed 2021-05-12 by nashif