zephyrproject-rtos/zephyr #34500: thingy52 lis2dh12 sensor values too large

pabigot, 2021-04-22:
Running `samples/sensors/lis2dh` on a thingy52_nrf52832 produces this:

```
*** Booting Zephyr OS build zephyr-v2.5.0-2446-g9693fc8fee68  ***
Polling at 0.5 Hz
#1 @ 337 ms: x 2.566368 , y 8.809920 , z 3.868704
#2 @ 2342 ms: x 2.451456 , y 8.656704 , z 3.983616
#3 @ 4347 ms: x 2.489760 , y 8.695008 , z 3.907008
```

I can guarantee the sensor is not experiencing 10 gravities of acceleration.  Running an independent application based on https://github.com/STMicroelectronics/STMems_Standard_C_drivers on the same hardware confirms these calculations are 10x higher than the sensor is producing.

avisconti, 2021-04-23:
@pabigot
OK, I'll take a look at it. Just for my curiosity, where is the the graviity vector Can you also add the values returned by STdC drivers keeping the h/w setup in the same orientation? Thanks!

pabigot, 2021-04-23:
The sensor orientation is on its side at an angle due to cables propping it up.  Earth gravity is mostly associated with the Y axis; the X and Z components are correct.  The magnitude of the vector is one gravity (9.99 according to Zephyr).

A subset of `lis2dh12_STdC/example/lis2dh12_self_test.c` augmented with output produced this for the values of the averaged `acceleration_mg` (in mg, truncated to integer):
```
X: 257
Y: 912
Z: 404
```
which is essentially the same.

avisconti, 2021-04-23:
I need to leave for the week. Now. I took a quick look and I suspect that there might be the following mistake:
```
    int32_t converted_val;
```
instead of
```
   int64_t converted_val;
```

avisconti, 2021-04-23:
BTW, I really want to replace this driivers using the STdC.

pabigot, 2021-04-23:
Looking into it it's PEBCAK: Zephyr's producing values in m/s^2, which is roughly 10 times the value in `g` that I was assuming.  That the STdC sample produces values in mg was what threw me.  Sorry.

closed 2021-04-23 by pabigot

ahoyahoy, 2025-01-19:
Dear @avisconti,
I saw that you had plans to create an example for using the STdC driver. I couldn’t find anything like that—are those plans still on? I’m struggling with how to set it up. Thanks!

avisconti, 2025-01-20:
> Dear [@avisconti](https://github.com/avisconti), I saw that you had plans to create an example for using the STdC driver. I couldn’t find anything like that—are those plans still on? I’m struggling with how to set it up. Thanks!

Well, I had that plan in my mind at that time, but then I changed a bit my mind, as I had plenty of things to do and this one seemed to me not 100% necessary (the current driver seems to work well). Can you specify a bit better what you are struggling with?

ahoyahoy, 2025-01-26:
@avisconti 
Thanks for your reply and for taking the time to engage.

Unfortunately, I’m a complete beginner with C and embedded development. I’m trying to set up LIS2DH12 interrupts, but I’ve noticed the Zephyr driver implements only a limited subset of features. For example, it supports SENSOR_TRIG_DATA_READY, SENSOR_TRIG_DELTA, and SENSOR_TRIG_TAP, but not SENSOR_TRIG_FIFO_WATERMARK.

Similarly, it doesn’t implement parameter settings like SENSOR_ATTR_LOWER_THRESH, SENSOR_ATTR_UPPER_THRESH, or SENSOR_ATTR_HYSTERESIS, even if the sensor supports them. The only implemented attributes are (as far as I can tell): SENSOR_ATTR_SLOPE_TH, SENSOR_ATTR_SLOPE_DUR, and SENSOR_ATTR_SAMPLING_FREQUENCY.

So, I thought I’d try setting these thresholds and interrupts directly via registers. Unfortunately, I couldn’t find an example for doing that, and my attempts haven’t been successful.

Then I looked at the official STdC driver—it seems much more readable and feature-complete. But I hit another roadblock trying to create the necessary SPI read and write functions. I just couldn’t get it working.

Would it be possible to extend the LIS2DH driver in Zephyr to support these features? Alternatively, could you provide an example of how to directly write to the sensor’s registers? :)))))

I also noticed the new example stream implementation using FIFO, but I think it isn’t applicable to LIS2DH.

Thanks so much for your guidance!

ahoyahoy, 2025-02-09:
Hi @avisconti 😉

I made some progress.
https://github.com/sayahoyahoy/lis2-pid-example/tree/main

As a beginner, I have no clue about manually controlling SPI.
Thus, I use lis2dh_raw_read and lis2dh_raw_write from the Zephyr driver.

I created a fake device to fill bus_cfg.

I used raw read/write in platform_write functions, as mentioned in the README.
I read lis2dh12_device_id_get(), which returned 0x33... so I assumed it can communicate with the sensor.

I must set .cs to NULL:

```
.config = {
	.operation = SPI_OP_MODE_MASTER | SPI_WORD_SET(8) | SPI_TRANSFER_MSB, // | SPI_MODE_CPOL | SPI_MODE_CPHA,
	.frequency = 10000000, // 10 MHz
	.slave = 0,
	.cs = NULL,
	// .cs = &my_spi_cs,
};
```
Otherwise, I get:

```
[00:00:00.506,591] <err> os: ***** BUS FAULT *****
[00:00:00.506,591] <err> os:   Instruction bus error
[00:00:00.506,622] <err> os: r0/a1:  0x200004c0  r1/a2:  0x00000001  r2/a3:  0x00000001
[00:00:00.506,622] <err> os: r3/a4:  0x21650000 r12/ip:  0x00000000 r14/lr:  0x000036f7
[00:00:00.506,652] <err> os:  xpsr:  0x40000000
[00:00:00.506,652] <err> os: Faulting instruction address (r15/pc): 0x21650000
[00:00:00.506,683] <err> os: >>> ZEPHYR FATAL ERROR 27: Unknown error on CPU 0
[00:00:00.506,713] <err> os: Current thread: 0x20000be8 (unknown)
[00:00:01.730,743] <err> os: Halting system
I have no clue how to set it correctly.
```

However, measuring doesn't work and returns only weird values:
raw: 13107 13107 13107

I tried something with manually controlling the CS pin, but I have no idea.

So here's a specific question—if you have a moment, could you look into it?
How do I properly set up the SPI so that it communicates correctly? :)))