zephyrproject-rtos/zephyr #34503: ehl_crb: tests/application_development/libcxx/application_development.cpp.libcxx.exceptions timeout

yerabolu, 2021-04-22:
**Describe the bug**
tests/application_development/libcxx/application_development.cpp.libcxx.exceptions Failed Timeout.

What have you tried to diagnose or workaround this issue?

**To Reproduce**
Steps to reproduce the behavior:
1. twister -W -p ehl_crb --device-testing --device-serial /dev/ttyUSB0 -T  tests/application_development/libcxx/ --west-flash="/home/ztest/EHL_X86_PXE.sh" -j 1
2. device timeout error in device.log

**Expected behavior**
Test is expected to Pass and test also fails on 

**Logs and console output**
```  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
^M  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0^M100   100    0   100    0     0   1851      0 --:--:-- --:--:-- --:--:--  1851
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
^M  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0^M100   100    0   100    0     0   1886      0 --:--:-- --:--:-- --:--:--  1886
```

**Environment (please complete the following information):**
 - OS: Fedora 29
 - Toolchain: SDK 12.4
 - Commit SHA e1a6820e4c470ad5693eef469a0f4b61f868d8ec

galak, 2021-04-27:
The output looks like an issue with flashing the test and not the test itself.

nashif, 2021-04-28:
duplicate of #32448

closed 2021-04-28 by nashif