zephyrproject-rtos/zephyr #34505: mimxrt1050_evk： failed to run testcases tests/net

Zhaoningx, 2021-04-23:
**To Reproduce**

Hi, I tried to run testcases tests/net on mimxrt1050_evk, sam_e70b_xplained and k64f platforms, but almost case are failed,

**Steps to reproduce the behavior:**

twister -p mimxrt1050_evk --device-testing --device-serial /dev/ttyACM0 -T tests/net

**see error**

Booting Zephyr OS zephyr-v1.14.2-intel-rc3-ww02.2 ***** 
there are a lot of testcase error log.

W: You have 1 IPv6 net_if addresses but 2 network interfaces
W: Consider increasing CONFIG_NET_IF_MAX_IPV6_COUNT value.
W: You have 1 IPv4 net_if addresses but 2 network interfaces
W: Consider increasing CONFIG_NET_IF_MAX_IPV4_COUNT value.
*** Booting Zephyr OS build zephyr-v2.5.0-2468-ge1a6820e4c47  ***
Running test suite test_net_shell_usability
===================================================================
START - test_setup
Cannot add 2001:db8::1 to interface 0x80000f28
Assertion failed at WEST_TOPDIR/zephyr/tests/net/shell/src/main.c:195: test_setup: (0 is false)
exiting
FAIL - test_setup in 0.14 seconds
===================================================================

...
===================================================================
START - test_ethernet_set_qav_kernel
Assertion failed at WEST_TOPDIR/zephyr/tests/net/socket/net_mgmt/src/main.c:463: test_ethernet_set_qav: (ret not equal to 0)
Cannot set Qav parameters
FAIL - test_ethernet_set_qav_kernel in 0.14 seconds
===================================================================
START - test_ethernet_set_qav_user
Assertion failed at WEST_TOPDIR/zephyr/tests/net/socket/net_mgmt/src/main.c:463: test_ethernet_set_qav: (ret not equal to 0)
Cannot set Qav parameters
FAIL - test_ethernet_set_qav_user in 0.14 seconds
===================================================================
START - test_ethernet_get_qav_kernel
Assertion failed at WEST_TOPDIR/zephyr/tests/net/socket/net_mgmt/src/main.c:490: test_ethernet_get_qav: (ret not equal to 0)
Cannot get Qav parameters (-1)
FAIL - test_ethernet_get_qav_kernel in 0.15 seconds
===================================================================
START - test_ethernet_get_qav_user
Assertion failed at WEST_TOPDIR/zephyr/tests/net/socket/net_mgmt/src/main.c:490: test_ethernet_get_qav: (ret not equal to 0)
Cannot get Qav parameters (-1)
FAIL - test_ethernet_get_qav_user in 0.15 seconds
===================================================================
...

===================================================================
START - test_mgmt
Starting Network Management API test
- Request Net MGMT
- Sending event 1 times, without a receiver, without info
Throwing event 0x97AB1234 1 times
- Sending event 2 times, without a receiver, without info
Throwing event 0x97AB1234 2 times
- Sending event 1 times, with a receiver, without info
Throwing event 0x97AB1234 1 times
Received event 0x97AB1234
Received 0x97AB1234 1 times
- Sending event 2 times, with a receiver, without info
Throwing event 0x97AB1234 2 times
Received event 0x97AB1234
Received event 0x97AB1234
Received 0x97AB1234 2 times
- Sending event 1 times, without a receiver, with info
Throwing event 0x97AB1234 1 times
- Sending event 2 times, without a receiver, with info
Throwing event 0x97AB1234 2 times
- Sending event 1 times, with a receiver, with info
Throwing event 0x97AB1234 1 times
Received event 0x97AB1234
Received 0x97AB1234 1 times
- Sending event 2 times, with a receiver, with info
Throwing event 0x97AB1234 2 times
Received event 0x97AB1234
Received event 0x97AB1234
Received 0x97AB1234 2 times
- Triggering core event: 0xE0600001
Assertion failed at WEST_TOPDIR/zephyr/tests/net/mgmt/src/mgmt.c:267: test_core_event: (func() is false)
func() check failed
FAIL - test_mgmt in 0.469 seconds

there are a lot of error logs, so, if you need more detail, comment it.

**Environment**
OS: Fedora33
Toolchain:zephyr-sdk-0.12.4
Commit ID:  e1a6820e4c470ad56

LeiW000, 2021-04-26:
It seems to me that it is an issue with the test case or the test configuration. Could any networking expert help to look into it? It is blocking our daily testing.

Zhaoningx, 2021-04-26:
The testcases tests/net are not working **mimxrt1050_evk**, **sam_e70b_xplained** and **k64f platforms,**

jukkar, 2021-04-27:
> Booting Zephyr OS zephyr-v1.14.2-intel-rc3-ww02.2 *****
> *** Booting Zephyr OS build zephyr-v2.5.0-2468-ge1a6820e4c47 ***

Why there are two different banners in the description? Which version this issue refers to?

Zhaoningx, 2021-04-28:
> > Booting Zephyr OS zephyr-v1.14.2-intel-rc3-ww02.2 *****
> > *** Booting Zephyr OS build zephyr-v2.5.0-2468-ge1a6820e4c47 ***
> 
> Why there are two different banners in the description? Which version this issue refers to?

Sorry for mistake, I have update the description.

LeiW000, 2021-04-28:
@jukkar , the commit id is e1a6820.

LeiW000, 2021-04-30:
@jukkar , thanks for fixing the issue.

commit 6e1df0d6 (2021-04-30): tests: net: Refactor tests when running in real hw

The network tests were expecting that network interfaces
are in certain order. As we cannot guarantee that, refactor
the tests like this:

* if test is using DUMMY L2 driver, then disable Ethernet L2
  and fetch only DUMMY L2 instead of default interface
* if test is using Ethernet L2 driver, then make sure that the
  test is using the Ethernet interface specified in the test
  instead of the one provided by the DUT

Fixes #34505