zephyrproject-rtos/zephyr #34507: RFC : WiFi : Add support for supplicant

sr1dh48r, 2021-04-23:
## Introduction

Proposal to include a supplicant that can be used for WiFi connections.

### Problem description

Currently Zephy OS does not have a supplicant or WiFi related stack which can be used to establish connections to WiFi peers. Adding a supplicant would help developers to integrate their WiFi stacks faster and help the community share the advanced features.

### Proposed change

We have done some preliminary study and have come with the conclusion that porting wpa_supplicant (https://w1.fi/wpa_supplicant/) is a viable solution (with some optimizations to the size) and something that will help us to keep pulling in changes from the community and also help contribute going forward.

## Detailed RFC

wpa_supplicant is the de-facto choice in Linux community and is actively maintained and has all the advanced features that WiFi connectivity requires. The idea here is to optimize the size of the existing codebase so that we can reduce the footprint to the features required for IOT devices. The intention is to have a supplicant codebase with the following goals in mind.

•	Ease of upgradability to the latest version of wpa_supplicant
•	Ease of interfacing to different WiFi Stacks.
•	Interfacing with Zephyr NET_MGMT interface
•	Suitability of upstreaming to Zephyr
•	Ease of submitting relevant changes back to wpa_supplicant tree

### Proposed change (Detailed)

Additional notes on Crypto:

The upstream wpa_supplicant from what I have seen provides support for a few Crypto engines (openSSL, wolfSSL, gnuTLS) but not mbedTLS. ESP solution (https://github.com/zephyrproject-rtos/hal_espressif/tree/zephyr/components/wpa_supplicant) implements the cryptography with mbedTLS already so it should not be a problem to do the same in the generic solution.

### Dependencies

Since Zephyr currently has fully offloaded design, the use of the supplicant implies the following : 
  - Supplicant will become the default interface for WiFi NET_MGMT APIs for drivers that want to use it
  - WiFi Driver's will need to call into Supplicant APIs to initialize interfaces instead. Both control and data APIs get routed from supplicant.
  - wpa_supplicant has driver ops that need to be implemented based on the WiFi stack design. Linux uses net80211 to talk to mac80211 WiFi stack but for custom stacks, the drivers will need to implement them.

## Alternatives

iwd (https://git.kernel.org/pub/scm/network/wireless/iwd.git/) : 
  - Looks like the license would be an issue with integrating it with Zephyr 
  - Relatively less popular than wpa_supplicant in Linux community
  - Concerns about future maintainability (that it won't be abandoned)

Other 3rd party options but very few to none that are open source maintained and stable like wpa_supplicant.

rlubos, 2021-04-23:
CC @jukkar @carlescufi @sylvioalves

jukkar, 2021-04-23:
CC @dleach02 I remember you mentioning you had wifi experience so we would appreciate any comments / ideas you may have.

sr1dh48r, 2021-05-06:
**Minutes of the meeting : Zephyr Networking Forum : 04-05-2021**

Marcel's comments

- Recommendation is to write from scratch and use IWD code as required and avoid wpa_supplicant.
- Entire IWD cannot be licensed or used but some parts of it can be used (re-licensed ??). (Can the entire WiFi stack code apart from Linux be used that part isn't clear ???)

Sridhar's comments

- We understand that IWD is mainly to solve Linux end user usability issues and to improve linux WiFi experience. And that the code itself has been optimized and more readable. (and wpa_supplicant's school of thought being different)
- wpa_supplicant is still pretty much the choice for all android devices, linux distributions, openWRT and other router projects we have worked on so far. So, we are leaning towards it.
- Concerns about long term support since IWD is not mainstream yet
- How much more optimization can we do to the IWD code since its already optimized (its unclear as of now how much the standalone wifi code can be used as such)
- Any IOT devices out the wild that use IWD so that we can get better confidence for using it

dleach02, 2021-05-07:
Didn't Marcel ultimately suggest that we do from scratch and just use IWD for architectural guidance? He indicated that it was too tightly tied to Linux to effectively port to an RTOS. I felt at the end of the meeting that the recommendation was generally, don't use wpa_supplicant and write one from scratch using IWD as a general guide. That clearly is a larger effort.

I personally think wpa_supplicant is a reasonable starting point. I've used it in the past for a previous wifi project with another RTOS.