zephyrproject-rtos/zephyr #34515: samples: net: syslog_net: hard fault when running on frdm_k64f

aunsbjerg, 2021-04-23:
**Describe the bug**
When building and running the syslog_net sample on a frdm_k64f board with no changes to the sample configuration, the application crashes with nothing printed to the shell besides the standard zephyr booting banner.

Disabling autostart of the log backend yields slightly more information in the form of a bus fault stack trace, see log output.

**To Reproduce**
west build -b frdm_k64f zephyr/samples/net/syslog_net -- -DCONFIG_LOG_BACKEND_NET_AUTOSTART=n
west flash -r jlink

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v2.5.0-2550-gd93297b3d078  ***

[00:00:00.006,000] <inf> net_config: Initializing network
[00:00:00.006,000] <inf> net_config: Waiting interface 1 (0x20001238) to be up...
uart:~$ 
[00:00:03.001,000] <inf> eth_mcux: ETH_0 enabled 100M full-duplex mode.
[00:00:03.002,000] <inf> net_config: IPv4 address: 192.0.2.2
[00:00:03.103,000] <inf> net_config: IPv6 address: 2001:db8::1
[00:00:03.103,000] <dbg> net_syslog.main: Starting
[00:00:04.002,000] <err> os: ***** BUS FAULT *****
[00:00:04.002,000] <err> os:   Imprecise data bus error
[00:00:04.002,000] <err> os:   NXP MPU error, port 3
[00:00:04.002,000] <err> os:     Mode: Supervisor, Data Address: 0x20003b84
[00:00:04.002,000] <err> os:     Type: Write, Master: 0, Regions: 0x8200
[00:00:04.002,000] <err> os: r0/a1:  0x20000c80  r1/a2:  0x20003bc0  r2/a3:  0x20003c4c
[00:00:04.002,000] <err> os: r3/a4:  0x00018ab3 r12/ip:  0x0000000a r14/lr:  0x00011359
[00:00:04.002,000] <err> os:  xpsr:  0x61002000
[00:00:04.002,000] <err> os: Faulting instruction address (r15/pc): 0x00011370
[00:00:04.002,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:04.002,000] <err> os: Current thread: 0x20001ea0 (logging)
[00:00:04.113,000] <err> os: Halting system
```

**Additional context**
Possibly related to #34403, at least in terms of observed behaviour.

jukkar, 2021-04-26:
Increasing the logging thread stack space from 768 (default) to 2048, the crash disappeared.
`-DCONFIG_LOG_PROCESS_THREAD_STACK_SIZE=2048`
probably lower number would be ok too, I did not really try to smaller value.

aunsbjerg, 2021-04-26:
Thanks @jukkar, you are right. I have should tried that before making this issue.
Would it make sense to add this to the sample configuration? I assume this requirement would be the same for all boards, not just the frdm_k64f.

jukkar, 2021-04-26:
@aunsbjerg I think better would be to increase the logging thread default stack size as the 768 seems to be too small. Then we would not need to adjust each and every sample.

nordic-krch, 2021-04-26:
I will look into this. Stack usage should not increase after logging v2.

PR #34573 (2021-04-26): logging: Increase thread stack size when net backend in use
It has been seen that when net backend is enabled stack usage
is around 1080. Setting 1152 as the default.

Fixes #34515.
closed 2021-04-26 by nashif