zephyrproject-rtos/zephyr #34527: Cpp compiling error: expected primary-expression before 'char'.    _Generic macros problem

DariusBabrauskas, 2021-04-23:
**Describe the bug**

Compile error if use call  LOG_DBG(...)  from cpp code.

Error example:
error: expected primary-expression before 'char'.  

**To Reproduce**
Simple try compile cpp file with code:
// Type your code here, or load an example.

// C program to illustrate macro function.
#include <stdio.h>
#define geeks(T) _Generic( (T), char: 1, int: 2, long: 3, default: 0)
int main(void)
{
    // char returns ASCII value which is int type. 
    printf("%d\n", geeks('A')); 
   
    // Here A is a string.
    printf("%d",geeks("A"));
      
    return 0;
} 

You can test it on different compilers her. I found that clang from 3.5 cpp support  _Generic - but result different if compile c file.
https://godbolt.org/z/jf1o41co5

**Expected behavior**

**Impact**

**Logs and console output**
../zephyr/include/sys/cbprintf_internal.h:75:4: error: expected primary-expression before 'char'
   75 |    char * : 1, \
      |    ^~~~
../zephyr/include/sys/util_internal.h:67:26: note: in definition of macro '__DEBRACKET'
   67 | #define __DEBRACKET(...) __VA_ARGS__
      |                          ^~~~~~~~~~~
../zephyr/include/sys/util_internal.h:59:2: note: in expansion of macro '__GET_ARG2_DEBRACKET'
   59 |  __GET_ARG2_DEBRACKET(one_or_two_args _if_code, _else_code)
      |  ^~~~~~~~~~~~~~~~~~~~
../zephyr/include/sys/util_internal.h:56:2: note: in expansion of macro '__COND_CODE'
   56 |  __COND_CODE(_ZZZZ##_flag, _if_0_code, _else_code)
      |  ^~~~~~~~~~~
../zephyr/include/sys/util_macro.h:133:2: note: in expansion of macro 'Z_COND_CODE_0'
  133 |  Z_COND_CODE_0(_flag, _if_0_code, _else_code)
      |  ^~~~~~~~~~~~~
../zephyr/include/sys/cbprintf_internal.h:110:2: note: in expansion of macro 'COND_CODE_0'
  110 |  COND_CODE_0(NUM_VA_ARGS_LESS_1(__VA_ARGS__), \
      |  ^~~~~~~~~~~
../zephyr/include/sys/util_loops.h:1075:2: note: in expansion of macro 'Z_CBPRINTF_IS_PCHAR'
 1075 |  fixed_arg0(x)
      |  ^~~~~~~~~~
../zephyr/include/sys/util_loops.h:28:2: note: in expansion of macro 'Z_FOR_EACH_EXEC'
   28 |  z_call(0, x, fixed_arg0, fixed_arg1)
      |  ^~~~~~
../zephyr/include/sys/util_loops.h:31:2: note: in expansion of macro 'Z_FOR_LOOP_1'
   31 |  Z_FOR_LOOP_1(z_call, sep, fixed_arg0, fixed_arg1, ##__VA_ARGS__) \
      |  ^~~~~~~~~~~~
../zephyr/include/sys/util_loops.h:36:2: note: in expansion of macro 'Z_FOR_LOOP_2'
   36 |  Z_FOR_LOOP_2(z_call, sep, fixed_arg0, fixed_arg1, ##__VA_ARGS__) \
      |  ^~~~~~~~~~~~
../zephyr/include/sys/util_loops.h:23:53: note: in expansion of macro 'Z_FOR_LOOP_3'
   23 |     _57, _58, _59, _60, _61, _62, _63, _64, N, ...) N
      |                                                     ^
../zephyr/include/sys/util_loops.h:1078:2: note: in expansion of macro 'Z_FOR_EACH_ENGINE'
 1078 |  Z_FOR_EACH_ENGINE(Z_FOR_EACH_EXEC, sep, F, _, __VA_ARGS__)
      |  ^~~~~~~~~~~~~~~~~
../zephyr/include/sys/util_macro.h:371:2: note: in expansion of macro 'Z_FOR_EACH'
  371 |  Z_FOR_EACH(F, sep, REVERSE_ARGS(__VA_ARGS__))
      |  ^~~~~~~~~~
../zephyr/include/sys/cbprintf_internal.h:95:3: note: in expansion of macro 'FOR_EACH'
   95 |  (FOR_EACH(Z_CBPRINTF_IS_PCHAR, (+),  __VA_ARGS__))
      |   ^~~~~~~~
../zephyr/include/sys/cbprintf_internal.h:112:7: note: in expansion of macro 'Z_CBPRINTF_HAS_PCHAR_ARGS'
  112 |    (((Z_CBPRINTF_HAS_PCHAR_ARGS(__VA_ARGS__) - skip) > 0)))
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~
../zephyr/include/sys/cbprintf.h:92:2: note: in expansion of macro 'Z_CBPRINTF_MUST_RUNTIME_PACKAGE'
   92 |  Z_CBPRINTF_MUST_RUNTIME_PACKAGE(skip, __VA_ARGS__)
      |  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../zephyr/include/logging/log_msg2.h:400:6: note: in expansion of macro 'CBPRINTF_MUST_RUNTIME_PACKAGE'
  400 |  if (CBPRINTF_MUST_RUNTIME_PACKAGE(_cstr_cnt, __VA_ARGS__)) { \
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../zephyr/include/logging/log_msg2.h:429:3: note: in expansion of macro 'Z_LOG_MSG2_CREATE2'
  429 |   Z_LOG_MSG2_CREATE2(_try_0cpy, _mode, 1, _domain_id, _source, \
      |   ^~~~~~~~~~~~~~~~~~
../zephyr/include/logging/log_core.h:315:3: note: in expansion of macro 'Z_LOG_MSG2_CREATE'
  315 |   Z_LOG_MSG2_CREATE(!IS_ENABLED(CONFIG_USERSPACE), _mode, \
      |   ^~~~~~~~~~~~~~~~~
../zephyr/include/logging/log_core.h:330:2: note: in expansion of macro 'Z_LOG2'
  330 |  Z_LOG2(_level, \
      |  ^~~~~~
../zephyr/include/logging/log.h:71:25: note: in expansion of macro 'Z_LOG'
   71 | #define LOG_DBG(...)    Z_LOG(LOG_LEVEL_DBG, __VA_ARGS__)
      |                         ^~~~~
../modules/libcpp/zephyr/global_allocator_overloads.cpp:13:5: note: in expansion of macro 'LOG_DBG'
   13 |     LOG_DBG("size = %u -> %p", (unsigned)size, ptr);
      |     ^~~~~~~

**Environment (please complete the following information):**
 - OS:  Windows
 - Toolchain :gcc version 10.2.1 20201103 (release) (GNU Arm Embedded Toolchain 10-2020-q4-major)
 - Commit SHA : SHA-1: a379196b48ede6c9e1b308c5ed92e1ce20bec94e
 

**Additional context**
-

galak, 2021-04-23:
Maybe the same issue as #34495

nordic-krch, 2021-04-26:
Yes, same issue. Fix in #34527

galak, 2021-04-26:
Closing this as a duplicate.

PR #34516 (2021-04-27): Logging: Fix issues that leads to failures when compiling for C++
Solving few casting issues in logging headers.

Limiting `_Generic` usage in cbprintf to C only.

I will work towards extending logging tests to validate C++ case.
Note, that for now logging v2 will work slower for C++ because `_Generic` cannot be used. This will also be fixed.

Fixes #34495.
Fixes #34527.
closed 2021-04-26 by galak