zephyrproject-rtos/zephyr #34536: Simple event-driven framework

hongshui3000, 2021-04-24:
**Is your feature request related to a problem? Please describe.**
Hope to add a simple event-driven framework for single-threaded applications. 
In this way, some simple devices do not need to start complex multi-threading.
Usable range
1.mcuboot
2.ble controller
3. Minimal MCU applications.
4. etc.

M1cha, 2021-07-27:
can you give an example of an application that would benefit from this that would otherwise require "complex multi-threading"?
Zephyr already has the (system-)workqueue which you can use for event-driven applications.
In addition to that you can ignore a lot of thread safety as long as you're working with cooperative threads only.

hongshui3000, 2021-07-28:
Event-driven programming is a programming paradigm, where the execution flow of the program is determined by external events. It is characterized by including an event loop, when an external event occurs, a callback mechanism is used to trigger the corresponding processing. Two other common programming paradigms are (single-threaded) synchronous and multi-threaded programming.

Let us use examples to compare and contrast single-threaded, multi-threaded, and event-driven programming models. Figure  shows the work done by the program in these three modes over time. This program has 3 tasks to complete, and each task blocks itself while waiting for I/O operations. The time spent blocking on I/O operations has been marked with a gray box.
[image]
In the single-threaded synchronization model, tasks are executed sequentially. If a task is blocked due to I/O, all other tasks must wait until they are completed before they can be executed in sequence. This clear execution order and serialization behavior is easy to infer. If the tasks are not dependent on each other, but still need to wait for each other, this makes the program unnecessarily slow down the running speed.

In the multi-threaded version, these 3 tasks are executed in separate threads. These threads are managed by the operating system and can be processed in parallel on a multi-processor system or interleaved on a single-processor system. This allows other threads to continue execution while a certain thread is blocked on a certain resource. Compared with a synchronous program that performs similar functions, this method is more efficient, but the programmer must write code to protect shared resources and prevent them from being accessed by multiple threads at the same time. Multi-threaded programs are more difficult to infer, because such programs have to deal with thread safety issues through thread synchronization mechanisms such as locks, reentrant functions, thread-local storage or other mechanisms. Improper implementation will lead to subtle and undesirable bug.

In the event-driven version of the program, the three tasks are executed in a staggered manner, but they are still controlled by a single thread. When dealing with I/O or other expensive operations, register a callback to the event loop, and then continue execution when the I/O operation is complete. The callback describes how to handle an event. The event loop polls all events and assigns them to the callback function waiting to process the event when the event arrives. In this way, the program can be executed as much as possible without the need for additional threads. Event-driven programs are easier to infer behavior than multithreaded programs, because programmers do not need to care about thread safety issues.
When we face the following environment, the event-driven model is usually a good choice:

1.There are many tasks in the program, and...
2.The tasks are highly independent (so they don’t need to communicate with each other or wait for each other) and...
3.While waiting for the event to arrive, some tasks will block.
This is also a good choice when the application needs to share variable data between tasks, because there is no need for synchronization.

Event-driven programming usually only uses one execution process. CPUs are not concurrent. When dealing with multitasking, event-driven programming uses cooperative processing tasks instead of multithreaded preemption. Event-driven is simple and easy to use, you only need to register the events of interest, design logic in the callback, and that's it. In the process of calling, the event loop is waiting for the event to occur, and then the Handler is called. The event Handler is not preemptive, and the Handler generally has a short life cycle.

Advantages of event-driven programming

1. In most application scenarios, event programming is superior to multithreaded programming.

2. Compared with multi-threaded programming, event-driven programming is easier and less complex, which developers are willing to accept.

3. Most GUI frameworks are based on event-driven programming. Each event is bound to a Handler. These events are usually button clicks, menu selections, and so on. Handler to implement specific behavior logic.

4. Event-driven is often used in the IO framework, which can achieve IO reuse. Many high-performance IO frameworks use event-driven models

5. Easy to debug. Time dependence is only related to events, not internal scheduling. The problem is easy to expose.

Disadvantages of event-driven programming

1. If the Handler takes a long time, it will block the response of the application.

2. The local state cannot be maintained by time, because the Handler must return.

3. Generally, it is faster than multi-threaded programming in a single CPU environment, because there is no lock factor and no thread switching loss. The CPU is not concurrent, so it is not suitable for use in some scientific computing applications.

M1cha, 2021-07-28:
Thanks for the detailed explanation but I do know the advantages of single-threaded event-driven programming vs multithreaded programming.

What I was trying to say is that zephyr does support that already using many different mechanisms(system-workqueue, kpoll, ...).

I understand that that's not the classical FSM that some projects want to use but if you're after that you should just use one of the many libraries since they are usually OS independent and totally out of scope for Zephyr.

PS: in case your don't know, I'm not a zephyr maintainer, just a random dude with an opinion. So don't take any of this as a project decision.

henrikbrixandersen, 2022-11-17:
@hongshui3000 Zephyr now also has a state machine framework: https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/smf.h Does this satisfy your requirements?

hongshui3000, 2022-11-18:
yes
I will close it

closed 2022-11-18 by hongshui3000