zephyrproject-rtos/zephyr #34538: STM32 temperature sensor

Delsian, 2021-04-24:
**Is your enhancement proposal related to a problem? Please describe.**
STM32 chips contain an internal temperature sensor, but existing ADC bindings doesn't allow read internal temperature value in a proper way.

**Describe the solution you'd like**
Sensor binding similar to dts\bindings\sensor\nxp,kinetis-temperature.yaml

**Additional context**

```
/* Temperature sensor calibration value address */
#define TEMP130_CAL_ADDR ((uint16_t*) ((uint32_t) 0x1FF8007E))
#define TEMP30_CAL_ADDR ((uint16_t*) ((uint32_t) 0x1FF8007A))
#define VDD_CALIB ((uint16_t) (300))
#define VDD_APPLI ((uint16_t) (330)) // <-- change this to according to your supply voltage

int32_t ComputeTemperature(uint32_t measure)
{
  int32_t temperature;
  temperature = ((measure * VDD_APPLI / VDD_CALIB) - (int32_t)*TEMP30_CAL_ADDR );
  temperature = temperature *(int32_t)(130-30);
  temperature = temperature /(int32_t)(*TEMP130_CAL_ADDR -*TEMP30_CAL_ADDR);
  temperature = temperature + 30;
  return(temperature);
}
```

Other usable sources, for example, **AN3964** "STM32L1x temperature sensor example"

erwango, 2021-04-26:
@Delsian Thanks for this. Don't hesiate to contribute of you have solution in mind.

Delsian, 2021-05-08:
Implemented.

closed 2021-05-08 by Delsian