zephyrproject-rtos/zephyr #34541: per-adv-sync-create doesn't work on nRF52840, ./tests/bluetooth/shell/

kaiser-ren, 2021-04-24:
**Describe the bug**
I tried to make shell sample code of *./tests/bluetooth/shell/* to initial periodic scanner on nRF52840. However, when I typed "bt per-adv-sync-create 00:11:22:33:44:55 random 0 skip 0 timeout 2000", it always return "Failed to set periodic advertising 
 -35". 
I dug into it and found it means Periodic Advertising is not supported. The root cause is that that nRF52840 should define 
CONFIG_BT_CTLR_ADV_EXT=y
CONFIG_BT_CTLR_SYNC_PERIODIC=y
if the developers want to use Periodic Advertising and these are experimental. I suggest that add these two flags on prj.conf. 

**To Reproduce**
Steps to reproduce the behavior:
1. cd ./tests/bluetooth/shell/
2. west build -p auto -b nrf52840dk_nrf52840 ./
3. west flash
4. open a terminal of the right com port
5. type "bt init"
6. type "bt per-adv-sync-create 00:11:22:33:44:55 random 0 skip 0 timeout 2000"

**Expected behavior**
Return an error code, -35. 

**Impact**
Periodic sync doesn't work. 

**Logs and console output**

**Environment (please complete the following information):**

**Additional context**

kaiser-ren, 2021-04-24:
Maybe add board overlay for nrf52840?

joerchan, 2021-04-30:
@kaiser-ren We haven't added them so far because this is not a finished feature, if we add them we would just be getting bug-reports on something that is not finished, which is something we wish to avoid.

closed 2021-04-30 by joerchan

kaiser-ren, 2021-05-01:
> @kaiser-ren We haven't added them so far because this is not a finished feature, if we add them we would just be getting bug-reports on something that is not finished, which is something we wish to avoid.
> 
> You will just have to enable them locally to test instead.

@joerchan , I don't agree with you. There are two samples, ./samples/bluetooth/periodic_adv/ and ./samples/bluetooth/periodic_sync/, they are public and periodic adv are enabled by default.

cvinayak, 2021-05-02:
@kaiser-ren I will go with @joerchan, the periodic_adv and periodic_sync  samples have already started raising github issues for the feature being experimental in the Zephyr controller. I will plan to remove the board specific configuration files until the feature is stable.

kaiser-ren, 2021-05-02:
Okay, no objection to close this issue.