zephyrproject-rtos/zephyr #34543: STM32F1 failed to compile with CONFIG_UART_ASYNC_API

roland-xz, 2021-04-24:
**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?

Testing nucleo_f103rb board, with my own code or the tests/drivers/uart/uart_async_api code, the compile fails.

**To Reproduce**
Steps to reproduce the behavior:
1. Configure a board with STM32F1 SoC, e.g, nucleo_f103rb
2. $ west build -p auto -b nucleo_f103rb --cmake ./zephyr/tests/drivers/uart/uart_async_api/
3. See error
```
/zephyr/drivers/dma/dma_stm32.c: In function 'dma_stm32_configure':
/zephyr/drivers/dma/dma_stm32.c:464:16: error: 'LL_DMA_InitTypeDef' has no member named 'PeriphRequest'
  464 |  DMA_InitStruct.PeriphRequest = config->dma_slot;
      |                ^
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 
```

**Expected behavior**
Compile should pass.

**Impact**
annoyance

**Logs and console output**
as above

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.12.3
 - Zephyr version: 2.5.99

**Additional context**
I believe DMA_InitStruct.PeriphRequest  is only used by SoCs that support dmamux. STM32F1 does not.
Adding the following seems to fix the problem.
```
	#if defined(DMAMUX1)
	DMA_InitStruct.PeriphRequest = config->dma_slot;
	#endif
```

FRASTM, 2021-04-26:
@roland-xz thank you for reporting this issue
The stm32f103 has a DMA1 (dma V2) but no DMAMUX, just like the stm32l476, but their stm32_ll_dma.h differ. It appears that `LL_DMA_InitTypeDef` structure  of the modules `stm32cube/ stm32f1xx` has NO PeriphRequest.

FRASTM, 2021-04-26:
A quick fix is to exclude the CONFIG_SOC_SERIES_STM32F1X in the drivers/dma/dma_stm32.c
```
#if defined(CONFIG_DMA_STM32_V2) || defined(CONFIG_DMAMUX_STM32)
#ifndef CONFIG_SOC_SERIES_STM32F1X
	/*
	 * the with dma V2 and dma mux,
	 * the request ID is stored in the dma_slot
	 */
	DMA_InitStruct.PeriphRequest = config->dma_slot;
#endif
#endif
```

FRASTM, 2021-04-26:
With this exception for the stm32f1x soc serie, the dma binding of the stm32 does not change :  a channel (defined by the ref. Manual "Summary of DMA1 requests for each channel") and a slot (not used). In this configuration, it could be specified that slot equal channels, for convenience. So for testing the _usart2_, the **nucleo_f103rb.overlay** could be as follows:
```
&usart2 {
	dmas = <&dma1 7 7 0x440 0>,
		<&dma1 6 6 0x480 0>;
	dma-names = "tx", "rx";
};

&dma1 {
	status = "okay";
};
```

roland-xz, 2021-04-26:
> With this exception for the stm32f1x soc serie, the dma binding of the stm32 does not change : a channel (defined by the ref. Manual "Summary of DMA1 requests for each channel") and a slot (not [...]

@FRASTM Thanks a lot for the suggested fixes. 
However, this was the other issue I encountered. I configured dma on usart2 this way the other day, but because zephyr console used uart 2 on this board, the test always failed. Not 100% sure if it was due to a conflict with the console. Please see failure log down below.
Moved the test on usart3 and it passed.

Keen to see tests/drivers/uart/uart_mix_fifo_poll work next because it's close to what I want to do.

```
*** Booting Zephyr OS build zephyr-v2.5.0-2384-gfacd0f57db85  ***
Running test suite uart_async_test
===================================================================
START - test_single_read_setup
 PASS - test_single_read_setup in 0.1 seconds
===================================================================
START - test_single_read
test
    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/uart/uart_async_api/src/test_uart_async.c:82: test_single_read: (k_sem_take(&rx_rdy, K_MSEC(100)) not equal to 0)
RX_RDY timeout
 FAIL - test_single_read in 0.218 seconds
===================================================================
START - test_chained_read_setup
 PASS - test_chained_read_setup in 0.1 seconds
===================================================================
START - test_chained_read
W: RX was already enabled
Message 0
    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/uart/uart_async_api/src/test_uart_async.c:164: test_chained_read: (k_sem_take(&rx_rdy, K_MSEC(1000)) not equal to 0)
RX_RDY timeout
 FAIL - test_chained_read in 1.31 seconds
===================================================================
START - test_double_buffer_setup
 PASS - test_double_buffer_setup in 0.1 seconds
===================================================================
START - test_double_buffer
W: RX was already enabled

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/uart/uart_async_api/src/test_uart_async.c:217: test_double_buffer: (uart_rx_enable(uart_dev, double_buffer[0], sizeof(double_buffer[0]), 50) not equal to 0)
Failed to enable receiving
 FAIL - test_double_buffer in 0.24 seconds
===================================================================
START - test_read_abort_setup
 PASS - test_read_abort_setup in 0.1 seconds
===================================================================
START - test_read_abort
W: RX was already enabled

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/uart/uart_async_api/src/test_uart_async.c:292: test_read_abort: (k_sem_take(&rx_rdy, K_MSEC(100)) not equal to 0)
RX_RDY timeout
 FAIL - test_read_abort in 0.120 seconds
===================================================================
START - test_chained_write_setup
 PASS - test_chained_write_setup in 0.1 seconds
===================================================================
START - test_chained_write
W: RX was already enabled
Message 1Message 2
    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/uart/uart_async_api/src/test_uart_async.c:508: test_chained_write: (k_sem_take(&rx_rdy, K_MSEC(100)) not equal to 0)
RX_RDY timeout
 FAIL - test_chained_write in 0.121 seconds
===================================================================
START - test_long_buffers_setup
 PASS - test_long_buffers_setup in 0.1 seconds
===================================================================
START - test_long_buffers
W: RX was already enabled

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/uart/uart_async_api/src/test_uart_async.c:580: test_long_buffers: (k_sem_take(&rx_rdy, K_MSEC(200)) not equal to 0)
RX_RDY timeout
 FAIL - test_long_buffers in 0.263 seconds
===================================================================
START - test_write_abort_setup
 PASS - test_write_abort_setup in 0.1 seconds
===================================================================
START - test_write_abort
W: RX was already enabled

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/uart/uart_async_api/src/test_uart_async.c:366: test_write_abort: (k_sem_take(&rx_rdy, K_MSEC(100)) not equal to 0)
RX_RDY timeout
 FAIL - test_write_abort in 0.120 seconds
===================================================================
START - test_forever_timeout_setup
 PASS - test_forever_timeout_setup in 0.1 seconds
===================================================================
START - test_forever_timeout
W: RX was already enabled

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/uart/uart_async_api/src/test_uart_async.c:442: test_forever_timeout: (k_sem_take(&rx_rdy, K_MSEC(100)) not equal to 0)
RX_RDY timeout
 FAIL - test_forever_timeout in 3.120 seconds
===================================================================
Test suite uart_async_test failed.
===================================================================
PROJECT EXECUTION FAILED
```

FRASTM, 2021-04-26:
On the nucleo_f103rb, the usart 2 is the console, so that you cannot run test on it, but on usart3 or usart1
```
&usart1 {
	dmas = <&dma1 4 4 0x440 0>,
		<&dma1 5 5 0x480 0>;
	dma-names = "tx", "rx";
};
```
```
&usart3 {
	dmas = <&dma1 2 2 0x440 0>,
		<&dma1 3 3 0x480 0>;
	dma-names = "tx", "rx";
};

```

roland-xz, 2021-04-26:
> On the nucleo_f103rb, the usart 2 is the console, so that you cannot run test on it, but on usart3 or usart1
>
> ```
> &usart1 {
> 	dmas = <&dma1 4 4 0x440 0>,
> 		<&dma1 5 5 0x480 0>;
> 	dma-names = "tx", [...]

Thanks again@FRASTM. 
Having trouble configuring the device tree for TIMER_0 with `compatible = "st,stm32-timer-counter"`, and in general DT configuration requires a lot of trials and errors for me.
Do you know of any STM32 specific documentation on this?

erwango, 2021-04-26:
> Thanks again@FRASTM.
Having trouble configuring the device tree for TIMER_0 with compatible = "st,stm32-timer-counter", and in general DT configuration requires a lot of trials and errors for me.
Do you know of any STM32 specific documentation on this?

@roland-xz There is no STM32 specific documentation for this, but you can find the list of supported STM32 bindings here:
https://docs.zephyrproject.org/latest/reference/devicetree/bindings.html#stmicroelectronics-st
Generic infirmation about bindings: https://docs.zephyrproject.org/latest/guides/dts/bindings.html

roland-xz, 2021-04-27:
@erwango Thanks for the links.

erwango, 2021-05-10:
Moving ticket from bug as enhancement as conclusion is DMA is not yet supported on STM32F1 and requires some implementation for support. cf #34666

PR #34666 (2021-07-22): stm32 DMA driver bindings new packaging
With this definition of the dma bindings, a new type of stm32 DMA version is supported.
For stm32F1 and stm32L1 soc, the DMA is of type V2 without dma request : named _V2bis_
A generic stm32-dma is introduced as a basis for all the stm32 dma types. The the `#dma-cells` is specified in the .yaml for each stm32 dma version, with a specific nb of elements.
- For the V1: stm32 F2, F4, F7, H7 : the dma instance has a Fifo the dma-cells has 4 (all) elements
- For the V2: stm32 F0, F3, G0,G4,WB, L0, L4,L5  : the dma instance has no Fifo, the dma-cells has 3 ele [... cut]
closed 2021-07-22 by cfriedt