zephyrproject-rtos/zephyr #34581: Unable to work with SX1276 Lora module.

Andermutu, 2021-04-26:
**Describe the bug**
I am trying to use SX1276 Lora module with nRF9160. I am trying /zephyr/samples/drivers/lora/send example. I can build and flash the example.

The connection between nRF and RF95W(SX1276) is the following:

3V --> 3V
GND --> GND
MOSI --> MOSI
MISO --> MISO
SCK --> SCK
GPIO 19 --> NSS(Chip select)

I added this .overlay file:

`&spi3 {
compatible = "nordic,nrf-spim";
status = "okay";
sck-pin = <20>;
mosi-pin = <21>;
miso-pin = <22>;
cs-gpios = <&gpio0 19 GPIO_ACTIVE_LOW>;

lora: sx1276@0 {
	compatible = "semtech,sx1276";
	reg = <0>;
	label = "sx1276";
	reset-gpios = <&gpio0 30 GPIO_ACTIVE_LOW>;
	dio-gpios = <&gpio0 31 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>,
		<&gpio0 2 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>,
		<&gpio0 3 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>,
		<&gpio0 4 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>,
		<&gpio0 5 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>;
	spi-max-frequency = <1000000>;
	power-amplifier-output = "pa-boost";		
};
};`

**To Reproduce**
Steps to reproduce the behavior:
1. build /zephyr/samples/drivers/lora/send for nRF9160
2. flash nRF9160
3. See error

**Expected behavior**

When the SX1276 is disconnected from SX1276 I expect the code to enter in `if (!lora_dev) {` or at least in `ret = lora_config(lora_dev, &config);` condition. However, that does not happend, it's more, I get "Data Sent" message from console.

When the SX1276 is properly connected I never get sent data in the Lora gateway, because I suspect that data is not being sent.

The only difference between having the module connected or not is that when the module is connected I see in the console this message:

> I: SX1276 Version:12 found 

When it is not properly connected or it is disconnected I see:

> I: SX1276 Version:00 found

or 

> I: SX1276 Version:ff found

**Impact**

We are waiting to be able to use SX1276 with nRF9160 to continiue with a project.

**Logs and console output**

I: SX1276 Version:12 found
*** Booting Zephyr OS build v2.4.0-ncs2  ***

Data sent!
Data sent!
Data sent!
Data sent!

**Environment (please complete the following information):**
 - OS: MacOS
 - Zephyr OS v2.4.0-ncs2

galak, 2021-04-27:
@carlescufi is this a reference to the nordic downstream ( v2.4.0-ncs2?)

Mani-Sadhasivam, 2021-04-28:
> When the SX1276 is disconnected from SX1276 I expect the code to enter in if (!lora_dev) { or at least in ret = lora_config(lora_dev, &config); condition.

This is not happening because the SPI reads are getting succeeded. I think we can do better here by bailing out if the version is `00` or `ff`. Let me do that.

> However, that does not happend, it's more, I get "Data Sent" message from console.

This also depends on the SPI controller of your chipset or its pin configuration.

> When the SX1276 is properly connected I never get sent data in the Lora gateway, because I suspect that data is not being sent.

> The only difference between having the module connected or not is that when the module is connected I see in the console this message:

Looks like the version info is getting fetched correctly but data is not sent. Did you observe a hang? Can you probe the SPI pins to see if there is any traffic?

Andermutu, 2021-04-28:
Thank you for your answer!

> Looks like the version info is getting fetched correctly but data is not sent. Did you observe a hang? Can you probe the SPI pins to see if there is any traffic?

The program keeps running forever and I get "Data Sent" message from console every second. We can not probe SPI pins because we don't have a digital oscilloscope, sorry. 

I also think that the right version is 12, because we get 00 or ff when it is disconnected or not properly connected(one SPI pin unwired for example).

Mani-Sadhasivam, 2021-04-30:
> The program keeps running forever and I get "Data Sent" message from console every second. We can not probe SPI pins because we don't have a digital oscilloscope, sorry.

No, I asked what happens when you connect the lora module?

>I also think that the right version is 12, because we get 00 or ff when it is disconnected or not properly connected(one SPI pin unwired for example).

Since this driver supports both SX1272 and SX1276, correct versions are `0x12` and `0x22`. I'll add check for these.

Andermutu, 2021-05-03:
> No, I asked what happens when you connect the lora module?

When I connect the Lora module I get version 12 found and "Data sent" print from console. However, I can not receive the sent data in my gateway/receiver device.

carlescufi, 2021-05-06:
> @carlescufi is this a reference to the nordic downstream ( v2.4.0-ncs2?)

It is. @Andermutu you might want to try opening a ticket in the Nordic DevZone.

Andermutu, 2021-05-07:
@carlescufi I have already openend a ticket. They told me that I might found help here...

carlescufi, 2021-05-07:
> @carlescufi I have already openend a ticket. They told me that I might found help here...

OK, understood. In that case, can you please try updating to the latest nRF Connect SDK master to see if that fixes your issue?

carlescufi, 2021-05-10:
Closing this issue now.

closed 2021-05-10 by carlescufi