zephyrproject-rtos/zephyr #34583: twister failing: fails platform native_posix, test lib/cmsis_dsp/filtering

sambhurst, 2021-04-26:
**Describe the bug**
Locally executing `twister` to run the lib/cmsis_dsp/filtering test on the native_posix platform fails.

**To Reproduce**
Steps to reproduce the behavior:
In the zephyr directory.

1. set ZEPHYR_TOOLCHAIN_VARIANT=llvm
2. twister -p native_posix -T tests/lib/cmsis_dsp/filtering -v
3. See errors

**Expected behavior**
The test should pass

**Impact**
Push Request is blocked due to Build with Clang/LLVM failure that's caused by this error

**Logs and console output**
```
zephyrproject/zephyr$ twister -p native_posix -T tests/lib/cmsis_dsp/filtering -v
INFO    - Zephyr version: zephyr-v2.5.0-2316-g36d96d2cd9b9
INFO    - JOBS: 8
INFO    - Using 'llvm' toolchain.
INFO    - Building initial testcase list...
INFO    - 4 test scenarios (4 configurations) selected, 1 configurations discarded due to filters.
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - 1/3 native_posix              tests/lib/cmsis_dsp/filtering/libraries.cmsis_dsp.filtering.biquad FAILED Build failure (native)
ERROR   - see: zephyrproject/zephyr/twister-out/native_posix/tests/lib/cmsis_dsp/filtering/libraries.cmsis_dsp.filtering.biquad/build.log
INFO    - 2/3 native_posix              tests/lib/cmsis_dsp/filtering/libraries.cmsis_dsp.filtering.misc FAILED Build failure (native)
ERROR   - see: zephyrproject/zephyr/twister-out/native_posix/tests/lib/cmsis_dsp/filtering/libraries.cmsis_dsp.filtering.misc/build.log
INFO    - 3/3 native_posix              tests/lib/cmsis_dsp/filtering/libraries.cmsis_dsp.filtering.fir FAILED Build failure (native)
ERROR   - see: zephyrproject/zephyr/twister-out/native_posix/tests/lib/cmsis_dsp/filtering/libraries.cmsis_dsp.filtering.fir/build.log

INFO    - 0 of 3 test configurations passed (0.00%), 3 failed, 1 skipped with 0 warnings in 8.02 seconds
INFO    - In total 822 test cases were executed, 274 skipped on 1 out of total 346 platforms (0.29%)
INFO    - 0 test configurations executed on platforms, 3 test configurations were only built.
INFO    - Saving reports...
INFO    - Writing xunit zephyrproject/zephyr/twister-out/twister.xml...
INFO    - Writing xunit zephyrproject/zephyr/twister-out/twister_report.xml...
INFO    - Run completed
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: LLVM

**Additional context**
Error from log file:
```
zephyrproject/modules/hal/cmsis/CMSIS/DSP/Source/FilteringFunctions/arm_biquad_cascade_df2T_f32.c:346:1: error: unknown attribute 'optimize' ignored [-Werror,-Wunknown    -attributes]
LOW_OPTIMIZATION_ENTER
^
zephyrproject/modules/hal/cmsis/CMSIS/DSP/Include/arm_math.h:8883:24: note: expanded from macro 'LOW_OPTIMIZATION_ENTER'
       __attribute__(( optimize("-O1") ))
```

mengxianglinx, 2021-04-27:
clang handles  __attribute__(( optimize("-O1") )) in different way.
add below code before line 8881 #elif defined ( __GNUC__ ) in modules/hal/cmsis/CMSIS/DSP/Include/arm_math.h:
#elif defined ( __clang__ )
  #define LOW_OPTIMIZATION_ENTER \
   __attribute__(( optnone ))
  #define LOW_OPTIMIZATION_EXIT
  #define IAR_ONLY_LOW_OPTIMIZATION_ENTER
  #define IAR_ONLY_LOW_OPTIMIZATION_EXIT

galak, 2021-04-27:
Not sure how you are running into this as we mark the filtering tests as not supported on llvm:

https://github.com/zephyrproject-rtos/zephyr/commit/9614581ebed7f14dbbe938182022630e0d79e675

sambhurst, 2021-04-27:
I needed to sync my repo. The tests are passing now. Thanks

closed 2021-04-27 by sambhurst