zephyrproject-rtos/zephyr #34584: kernel: workqueue thread is occasionally not invoked when kernel is run in cooperative mode only

mnkp, 2021-04-26:
**Describe the bug**

I've been working on a low latency, real time audio processing device which receives audio data over IP network and plays them via an audio codec. The audio data packets are received every 1 ms and passed immediately to the audio codec. Our application is working in cooperative only mode enabled by setting `CONFIG_NUM_PREEMPT_PRIORITIES=0`.

We have noticed that occasionally the network packet containing audio data would be delivered too late to the application. I.e. the networking stack would keep it for an additional 1 ms instead of forwarding it immediately to the application.

In our setup on average 1 packet every 10 s, that is 1 in 10000 packets is delivered too late.

When debugging the issue we've determined that the culprit is a work queue thread that occasionally is not invoked by the kernel.

I have recreated the scenario in a much simplified manner using a counter driver that generates an interrupt every 1 ms and puts  a work item into a work queue. The behavior is the same as in our more complex case.

**To Reproduce**

Steps to reproduce the behavior:
1. copy the file [main.c](https://gist.github.com/mnkp/44269ff359410c519947d458977f5842) to `samples/drivers/counter/alarm/src/main.c`
2. build the application `west build -b <board> -- -DCONFIG_NUM_PREEMPT_PRIORITIES=0 samples/drivers/counter/alarm`
3. check the device log
4. work queue items that stayed in the queue for longer than 600 us will be logged as `d <us delay>`.

**Expected behavior**

As an RTOS the Zephyr's scheduler behavior should be deterministic.

**Impact**

The above behavior pretty much breaks our application. It introduces frequent 1 ms delays and likely longer delays that are less frequent.

**Logs and console output**

Output of the sample application collected during 10 s runtime
```
*** Booting Zephyr OS build zephyr-v2.5.0-2555-gba421d613943  ***
Counter alarm sample

Set alarm in 999 usec (1171 ticks)
d 1002
d 1002
d 1002
d 1002
d 1002
```
In this case 5 out of ~10000 work queue items were processed too late.

**Environment**

 - Tested sam_e70_xplained evaluation board using counter_sam_tc.c driver that is not yet up-streamed.
 - Toolchain zephyr 0.12.4
 - Zephyr v2.5.0

**Additional context**

The issue cannot be observed if the preemptive scheduling is enabled.

ioannisg, 2021-04-27:
@pabigot FYI

pabigot, 2021-04-27:
AFAICT the reproducing code invokes undefined behavior, since the work handler frees the memory that holds the work item data structure.  This is in violation of [the requirement](https://docs.zephyrproject.org/latest/reference/kernel/threads/workqueue.html#work-item-lifecycle) that the work item not be altered outside of the work API until the item has been processed by the work queue thread.  The item is not released by the work queue thread until after the handler function returns and the state flags in the item have been updated to mark it as completed.

https://github.com/zephyrproject-rtos/zephyr/blob/3a4f08cd904701b320e602d2e89840de7be34b37/kernel/work.c#L674-L677

Can you reproduce the behavior without freeing the work item from within the handler?

pabigot, 2021-04-27:
I played around with this more, and by using statically allocated work items I can see the delays show up, but I'm not sure where they're coming from.  It looks like the number of pending items in the work queue is normally zero or one, but at some times so many work items are queued that the expected deadline is exceeded.

In any case I can't see anything about this that is specific to work queues; if this isn't behaving correctly it seems likely to be a problem with scheduling cooperative threads (e.g. the sleeping main thread isn't yielding to the ready work queue thread), or there's some other thread that's just not yielding often enough and so produces backups.

Putting a `k_yield()` in the main loop after `k_sleep()` has a noticeable impact on the frequency and magnitude of the deadline misses.

mnkp, 2021-04-27:
> AFAICT the reproducing code invokes undefined behavior, since the work handler frees the memory that holds the work item data structure. This is in violation of [the [...]

Good point. Though it seems it's a bit tricky to do it in a clean way for non static work items. I could try to pass the memory block that holds the work item to another thread for later processing using one of the available mechanisms (FIFO, message queue) but the queues typically pass the data directly to the waiting thread (if there is a thread waiting). Since handler function itself is executed in the thread context I'm still not sure that the k_work item will not be changed until after the handler function returns. I could pass memory block only to a lower priority thread than that of the handler function. Or there is a better / simpler way?

@pabigot thanks for looking into the issue. Could you add a link to the test code with statically allocated work items to the issue description?

I also believe the observed behavior is not related to work queues but rather the scheduler.

pabigot, 2021-04-28:
https://github.com/pabigot/zephyr/tree/issue/34584 has my test code, based on the original reproducer.  There's an intermediate commit that has additional diagnostics that confirmed to my satisfaction that the work queue is behaving correctly, but are not necessary to reproduce the problem.

Note that I could not reproduce this on Nordic boards, possibly because of the slower cycle clock.  It does reproduce on the atsamd21_xpro.

I've avoided the free-from-handler problem by using a fixed pool of work structures.  I don't have a solution for truly dynamic allocation with an alloc/free pair; 1c7bf9601941b2e76e6f27f070600ebe9535df09 has an example of a similar problem, which was solved by marking the item as complete/not-in-use then finishing the cleanup by detecting that in the allocator.  For the original solution the releasable memory could be put onto a free list that's processed whenever allocations occur---but at that point you might be better using a pool directly.

The behavior I see shows the work queue normally has one or two items, but if something delays the work handler the timer queues a large amount of work, which takes long enough to process that the submit/process interval exceeds expectation.  Whether this is due to a bug in the scheduler is TBD.

Also: I found that the timer callback was submitting items before the work queue was ready to accept them, causing submission to fail.  I will look into that because I don't think it should have happened, but for now I reorganized to give the work queue a chance to run before setting up the timer.

andyross, 2021-05-03:
Have you thought about reconsidering the priority design here?  I don't see what the value in disallowing preemption is, personally.  In general in hard real time situations, you **want** low priority tasks to be preempted so that high priority tasks can run promptly.  With NUM_PREEMPT_PRIORITIES=0, you're guaranteed that high priority tasks will **never** run until the current thread blocks or yields.  Which means that your latency bounds for your highest priority thread are determined by the **worst** code path through **any** other code in the system, app or kernel.

The only question in this particular case is "which code is being slow", because **any** code taking more than 600us, anywhere, doing anything, is going to cause this behavior.

The system can't make the guarantee you're demanding, basically.

mnkp, 2021-05-04:
> Have you thought about reconsidering the priority design here?

Yes, I thought and cooperative only mode seemed like the best choice.

Our application is a fairly simple one, similar to the sample code which I attached to the issue. It reads data packets that arrive via the network, does some minor tweaks here and there and copies it to the audio codec. There are no long tasks. Not in the application and not in the drivers.

The data are arriving in 1 ms tact with possibly large jitter. Processing it is the high priority task. If there were any long, low priority tasks in the system and I had to rely on preemptive scheduling for task switching I would need to set the system tick at 100 us or so. Not a great choice even for a fast MCU that we are using.

> The only question in this particular case is "which code is being slow", because any code taking more than 600us, anywhere, doing anything, is going to cause this behavior.

Yes, that's indeed the question. And so far everything indicates that it's the kernel code.

andyross, 2021-05-04:
> The data are arriving in 1 ms tact with possibly large jitter. Processing it is the high priority task.

But... you picked an architecture where the system will **not** process it with high priority. It's going to wait until whatever it is doing currently (even if it's low priority, like emitting a big log buffer -- do you have non-immediate logging enabled?) finishes up and reaches a schedule point.

Again, any long (>600us) CPU processing at all anywhere in the system is going to cause exactly the behavior you are seeing.  I just don't see why you think this is something to do with the scheduler or workqueue.  It's just the way code works.  "Thread Preemption" is a **feature** invented by operating systems long ago to treat the fact that sometimes you need to stop what you're doing because something more important is happening.  I really think you should try to use it.

andyross, 2021-05-04:
To be clearer: even if there was a circumstance where kernel code[1] was doing some non-trivial CPU work without yielding, that would only constitute a "bug" if the work wasn't necessary or if it was done at an incorrect priority.  Work is work, the OS has work too, and it needs to be able to do it.

[1] Or any code, really.  Again if you have logging enabled, I'm suspicious that this is the culprit.  Log buffers are comparatively large and get drained all at once in a low priority thread.  Which is the intent: flush the log only when you have nothing better to do.

andyross, 2021-05-04:
>  If there were any long, low priority tasks in the system and I had to rely on preemptive scheduling for task switching I would need to set the system tick at 100 us or so. Not a great choice even for a fast MCU that we are using.

Oh, wait.  I think maybe you're confusing preemption with timeslicing.  It's timeslicing that switches between threads (of equal priority!) at ticks to ensure that all runnable threads at a priority get some CPU.

Preemption can happen at any moment.  If, for example, you have a high priority thread blocked on a semaphore and a lower priority preemptible thread spinning on the CPU (computing digits of pi, whatever), and an interrupt arrives that produces some data that results in a k_sem_give() on that semaphore, then the interrupt will return directly into the higher priority thread.  It gets to run immediately.  The lower priority thread is preempted.

mnkp, 2021-05-07:
> But... you picked an architecture where the system will **not** process it with high priority. It's going to wait until whatever it is doing currently (even if it's low priority, like emitting a big log buffer -- do you have non-immediate logging enabled?) finishes up and reaches a schedule point.

But it will process it with a high priority, since at the time point the interrupt arrives the system has been sitting in the idle loop for a long time.

> Again, any long (>600us) CPU processing at all anywhere in the system is going to cause exactly the behavior you are seeing. 

And again, I don't have any long tasks in the system.

> I just don't see why you think this is something to do with the scheduler or workqueue.

Please read fully the issue description. The random long delay I'm observing is 1 ms, always. If it was some long task running in the system we would observe random long delays, but not necessarily of the same length. 1 ms is the time delay between interrupt events. It looks like once in a while the interrupt is processed and a work queue item is submitted to the workqueue thread but the thread is not invoked by the kernel scheduler. The thread will be invoked the next time interrupt arrives and then two work items are processed one after another.

> Again if you have logging enabled, I'm suspicious that this is the culprit.

The logging system is configured with `CONFIG_LOG_MODE_IMMEDIATE=y`, primarily to monitor system startup. Even if immediate mode wasn't enabled I'm certainly not going to use logging system at the time I want to process events arriving at 1 ms interval.

@andyross Rather than trying to find nonexistent problems in the application you don't have access to it would be more productive if you tried to find actual problems in the sample code I've attached to the issue. Please feel free to identify the code responsible for the long tasks (you are so sure are in the system) that cause the delays. Surely it can be something else than the kernel scheduler but at the moment it doesn't look like it.

> Oh, wait. I think maybe you're confusing preemption with timeslicing. It's timeslicing that switches between threads (of equal priority!) at ticks to ensure that all runnable threads at a priority get some CPU.

Indeed, I didn't realize I can configure preemptive mode with time slicing disabled. That's the best option for our application and I'm going to use it. Thanks for the tip. Still, it doesn't change the fact that cooperative only mode seems to have a problem.

andyross, 2021-05-10:
I don't know what to say.  Can you please submit an exerciser that localizes your 600us delay to some code we can optimize?   Again, long priority inversion delays due to code running without hitting scheduling points are the documented behavior of cooperative scheduling.  What you're complaining about is literally what it's supposed to do.

andyross, 2021-05-13:
I pushed a patch that claims to "fix" this bug, though I'm still not convinced it's real. :)  What it actually does is simply remove the "cooperative scheduling only" special cases from the kernel, which even assuming they were working correctly were stale and bitrotten.  Now the kernel builds the same always, and the idle thread is always preemptible by definition.

PR #35276 (2021-05-25): kernel/sched: Remove "cooperative scheduling only" special cases
The scheduler has historically had an API where an application can
inform the kernel that it will never create a thread that can be
preempted, and the kernel and architecture layer would use that as an
optimization hint to eliminate some code paths.

Those optimizations have dwindled to almost nothing at this point, and
they're now objectively a smaller impact than the special casing that
was required to handle the idle thread (which, obviously, must always
be preemptible).

Fix this by eliminating the idea of "cooperative only"  [... cut]
commit 851d14af (2021-05-25): kernel/sched: Remove "cooperative scheduling only" special cases

The scheduler has historically had an API where an application can
inform the kernel that it will never create a thread that can be
preempted, and the kernel and architecture layer would use that as an
optimization hint to eliminate some code paths.

Those optimizations have dwindled to almost nothing at this point, and
they're now objectively a smaller impact than the special casing that
was required to handle the idle thread (which, obviously, must always
be preemptible).

Fix this by eliminating the idea of "cooperative only" [... cut]
closed 2021-05-25 by nashif