zephyrproject-rtos/zephyr #34585: mec15xxevb_assy6853: test_timeout_order in tests/kernel/common assertion failed

yerabolu, 2021-04-27:
**Describe the bug**
 test_timeout_order in tests/kernel/common assertion failed

**To Reproduce**
Steps to reproduce the behavior:
1. twister --hardware-map ~/mec1.file --device-testing -T tests/kernel/common/
2. See error

**Expected behavior**
Test should pass

**Logs and console output**
```
EBUG   - DEVICE: ***** delaying boot 500ms (per build configuration) *****
DEBUG   - DEVICE: *** Booting Zephyr OS build zephyr-v2.5.0-2578-ge366d7948a53  (delayed boot 500ms) ***
DEBUG   - DEVICE: Running test suite common
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_bootdelay
DEBUG   - DEVICE: PASS - test_bootdelay in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_irq_offload
DEBUG   - DEVICE: PASS - test_irq_offload in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_byteorder_memcpy_swap
DEBUG   - DEVICE: PASS - test_byteorder_memcpy_swap in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_byteorder_mem_swap
DEBUG   - DEVICE: PASS - test_byteorder_mem_swap in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sys_get_be64
DEBUG   - DEVICE: PASS - test_sys_get_be64 in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sys_put_be64
DEBUG   - DEVICE: PASS - test_sys_put_be64 in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sys_get_be48
DEBUG   - DEVICE: PASS - test_sys_get_be48 in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sys_put_be48
DEBUG   - DEVICE: PASS - test_sys_put_be48 in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sys_get_be32
DEBUG   - DEVICE: PASS - test_sys_get_be32 in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sys_put_be32
DEBUG   - DEVICE: PASS - test_sys_put_be32 in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sys_get_be24
DEBUG   - DEVICE: PASS - test_sys_get_be24 in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sys_put_be24
DEBUG   - DEVICE: PASS - test_sys_put_be24 in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sys_get_be16
DEBUG   - DEVICE: PASS - test_sys_get_be16 in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sys_put_be16
DEBUG   - DEVICE: PASS - test_sys_put_be16 in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sys_get_le16
DEBUG   - DEVICE: PASS - test_sys_get_le16 in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sys_put_le16
DEBUG   - DEVICE: PASS - test_sys_put_le16 in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sys_get_le24
DEBUG   - DEVICE: PASS - test_sys_get_le24 in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sys_put_le24
DEBUG   - DEVICE: PASS - test_sys_put_le24 in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sys_get_le32
DEBUG   - DEVICE: PASS - test_sys_get_le32 in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sys_put_le32
DEBUG   - DEVICE: PASS - test_sys_put_le32 in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sys_get_le48
DEBUG   - DEVICE: PASS - test_sys_get_le48 in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sys_put_le48
DEBUG   - DEVICE: PASS - test_sys_put_le48 in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sys_get_le64
DEBUG   - DEVICE: PASS - test_sys_get_le64 in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sys_put_le64
DEBUG   - DEVICE: PASS - test_sys_put_le64 in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_atomic
DEBUG   - DEVICE: PASS - test_atomic in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_bitfield
DEBUG   - DEVICE: SKIP - test_bitfield in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_printk
DEBUG   - DEVICE: 22 113 10000 32768 40000 22
DEBUG   - DEVICE: p 112 -10000 -32768 -40000 -22
DEBUG   - DEVICE: 0x1 0x01 0x0001 0x00000001 0x0000000000000001
DEBUG   - DEVICE: 0x1 0x 1 0x   1 0x       1
DEBUG   - DEVICE: 42 42 0042 00000042
DEBUG   - DEVICE: -42 -42 -042 -0000042
DEBUG   - DEVICE: 42 42   42       42
DEBUG   - DEVICE: 42 42 0042 00000042
DEBUG   - DEVICE: 255     42    abcdef        42
DEBUG   - DEVICE: 68719476735 -1 18446744073709551615 ffffffffffffffff
DEBUG   - DEVICE: 0xcafebabe 0xbeef 0x2a
DEBUG   - DEVICE: PASS - test_printk in 0.31 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_timeout_order
DEBUG   - DEVICE:
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/common/src/timeout_order.c:92: test_timeout_order: (poll_events[ii].state not equal to K_POLL_STATE_SEM_AVAILABLE)
DEBUG   - DEVICE:
DEBUG   - DEVICE: FAIL - test_timeout_order in 0.116 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_clock_uptime
DEBUG   - DEVICE: PASS - test_clock_uptime in 0.12 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_clock_cycle
DEBUG   - DEVICE: PASS - test_clock_cycle in 0.3 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_version
DEBUG   - DEVICE: PASS - test_version in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_multilib
DEBUG   - DEVICE: PASS - test_multilib in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_thread_context
DEBUG   - DEVICE: PASS - test_thread_context in 0.31 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_errno
DEBUG   - DEVICE: PASS - test_errno in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_ms_time_duration
DEBUG   - DEVICE: PASS - test_ms_time_duration in 0.182 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_bounds_check_mitigation
DEBUG   - DEVICE: PASS - test_bounds_check_mitigation in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: Test suite common failed.
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: PROJECT EXECUTION FAILED
```
**Environment (please complete the following information):**
 - OS: Fedora 29
 - SDK: zephyr-sdk-0.12.4
 - Commit SHA: e366d7948a53

scottwcpg, 2021-04-27:
Reproduced on MEC152x EVB building only the tests/kernel/common.
Not related to power management. Fails with PM disabled in build.
It will pass if I lower the CPU clock speed from full speed 48 MHz to 12 MHz. This implies this SoC is not able to update fast enough. The build is configured to use a 32 KHz clock for the kernel timer with second 48 MHz timer used for higher resolution.
Were there any changes in the kernel timing logic?

scottwcpg, 2021-04-27:
Narrowed it down to something related to the 32KHz timer we are using for the kernel timer.  If the build is switched to SYSTICK the test passes even at fastest CPU speed.

scottwcpg, 2021-04-27:
@albertofloyd This is an issue. If we have to switch to SYSTICK then deep sleep will require enabling a 32 KHz based timer and the issues converting kernel ticks to 32 KHz units.

scottwcpg, 2021-04-27:
@yerabolu and @ioannisg 
The commit a09b5ade26e97953933753a500d5235d8ba85387 remoted a printk() from the spawned thread speeding up thread completion.  Adding a the printk() back or a delay loop writing a read-only HW register allows the test to pass but causes the test application to trigger a fault after all tests have finished.  Tried increasing thread stack size by setting CONFIG_TEST_EXTRA_STACKSIZE = 512. Tests all pass but still get fault when test program done.
Are other platforms using a 32 KHz kernel timer failing this test?

albertofloyd, 2021-04-28:
@yerabolu was this test passing before?  Do you have any any regression data? 
Does the test have any CPU requirements?

yerabolu, 2021-04-28:
> @yerabolu was this test passing before? Do you have any any regression data?
> Does the test have any CPU requirements?

@albertofloyd and @scottwcpg: This test was passing before and started failing from c22df64b616c. Will bisect to look into which commit it started failing exactly.

albertofloyd, 2021-04-28:
Thanks @yerabolu . That commit seems to be part of this PR https://github.com/zephyrproject-rtos/zephyr/pull/33302 introduced by as part of tickeless idle removal.

I;m not clear in the implications from tickeless removal to the RTOS drivers @andyross @nashif  could you indicate if preliminary analysis from @scottwcpg is going in the direction expected by Zephyr.

nashif, 2021-05-11:
the commit above is not related to tickless idle, tickless idle is not related to tickless kernel. Out kernel is tickless by default, this was the mode we had before introducing tickless kernel.

PR #35275 (2021-05-25): tests: kernel/common: incorrect use of k_poll in timeout order
In the timeout order test, the usage of k_poll() assumes that it
only returns after all events are ready. However, that is not
the case, as k_poll() returns when non-zero number of events are
ready. This means the check for all semaphore being ready after
k_poll() will not always pass. So instead of using k_poll(),
simply wait a bit for timers to fire, then check results.

Also add some bits to clean up at the end of test.

Fixes #34585
commit 8fd3d18b (2021-05-25): tests: kernel/common: incorrect use of k_poll in timeout order

In the timeout order test, the usage of k_poll() assumes that it
only returns after all events are ready. However, that is not
the case, as k_poll() returns when non-zero number of events are
ready. This means the check for all semaphore being ready after
k_poll() will not always pass. So instead of using k_poll(),
simply wait a bit for timers to fire, then check results.

Also add some bits to clean up at the end of test.

Fixes #34585
closed 2021-05-25 by galak