zephyrproject-rtos/zephyr #34590: Functions getopt_long and getopt_long_only from the FreeBSD project

jakub-uC, 2021-04-27:
## Origin

The FreeBSD
https://github.com/lattera/freebsd/blob/master/lib/libc/stdlib/getopt_long.c

## Purpose

Extend existing lib/util/getopt library support with getopt_long and getopt_long_only functions.
Add support for two modes of operation: single instance and multi-instance.

## Mode of integration

It is needed to integrate the module in the main tree. The main module using this library will be a shell. Shell is multi-instance and therefore requires getopt to be multi-instance as well. Because of this, the getopt library from FreeBSD must be modified which makes it impossible to include it as a module.

## Pull Request

https://github.com/zephyrproject-rtos/zephyr/pull/33948

## Description

This library is going to be used mainly by the shell module. Some shell users are not satisfied with subcommands alone and need to use the options for commands as well. A library is needed that allows the developer to parse the arguments string, looking for supported options. Typically, this task is accomplished by the getopt, getopt_long and getopt_long_only functions.
For this purpose, I decided to port the getopt and getopt_long libraries available in the FreeBSD project. I had to modify it so that it could be used by all instances of the shell at the same time.
Originally this function was using global variables which were defining its state. In my implementation, I put those variables in a structure and a pointer to that structure is passed as an additional parameter to getopt functions. In the proposed implementation, each shell backend has its own function state structure which it uses.

Additionally, I implemented full compatibility with the getopt library. Using configuration options:
CONFIG_GETOPT_SINGLE=y
CONFIG_GETOPT_LONG_SINGLE=y
It is possible to use them in getopt single-instance mode. Using single-instance mode has the following consequences:
1. the getopt library has the same API as the standard getopt library, so there is the ease of porting existing code.
2. such usage is not threaded safe because getopt operates on global variables. In other operating systems this problem is solved by processes. Zephyr unfortunately does not support processes, only threads.

In summary, this PR implements the use of the getopt library in several ways:
1. multi-instance API:
z_getopt, z_getopt_only, z_getopt_long_only
These functions correspond to the getopt function except that they have an additional variable to specify their state.

2. shell API:
shell_getopt, shell_getop_long, shell_getopt_long_only
These functions correspond to the getopt function except that they have an additional variable that specifies their state. These functions should be used in the command handler.

3. single-instance API:
getopt, getopt_long, getopt_long_only

# Dependencies

This package does not depend on any other component.
However, you cannot use getopt in the single-instance mode for ztests because there is a name conflict with the GNU implementation of these functions.

## Revision

324e4c082c14aebf00f8dbee0fb7ad285393756a

## License

"BSD 2-Clause "Simplified" License"

gmarull, 2021-04-27:
For reference, in case we prefer using Apache licensed code:

Nuttx has support for longopt, see https://github.com/apache/incubator-nuttx/tree/master/libs/libc/unistd

cfriedt, 2022-03-19:
I believe this was merged with commit d4559f53faf6cdcca95046f0dfbb1d5dd29b8d22

closed 2022-03-19 by cfriedt