zephyrproject-rtos/zephyr #34600: Bluetooth: L2CAP: Deadlock when there are no free buffers while transmitting on multiple channels

KatrineNordic, 2021-04-27:
**Describe the bug**
Transmitting many large SDUs as quickly as possible on multiple L2CAP channels quickly fills a lot of buffers. When no more segments can be queued to be transmitted because there are no more free buffers, or there are no more credits for the channel, queuing of segments is stopped. Whenever a previously queued segment has been transmitted, or more credits are received, segments are again being queued for transmission. If there are still no free buffers, queuing of segments is stopped again right away. If this happens when all segments that have previously been queued on the channel have already been transmitted, and there are no more credits to receive, queuing of segments is stopped indefinitely. 

**Expected behavior**
Queuing of segments continues when there are free buffers available.

**Impact**
Annoyance.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain (e.g Zephyr SDK, ...)
 - https://github.com/nrfconnect/sdk-zephyr/commit/6a1d340b632ec62f5abfb2e0c3e54838bbfd086e

**Additional context**
It is possible to work around this by having a high enough number of buffers (CONFIG_BT_L2CAP_TX_BUF_COUNT or CONFIG_BT_CONN_TX_MAX) that there are always buffers available when queuing of segments is restarted.

carlescufi, 2021-04-27:
@KatrineNordic which Nordic (upstream, not NCS/sdk-zephyr) revision did you test this with?

carlescufi, 2021-05-17:
> @KatrineNordic which Nordic (upstream, not NCS/sdk-zephyr) revision did you test this with?

This seems to be an issue upstream, as per a discussion with @KatrineNordic and @joerchan

carlescufi, 2021-05-20:
Downgrading to low since:

- This is only an issue when using  multiple L2CAP Connection-Oriented Channels (a rare use case in practice)
- There is a workaround (increase `CONFIG_BT_L2CAP_TX_BUF_COUNT`)

carlescufi, 2021-06-17:
@Vudentz FYI, in case you have additional comments to this.

carlescufi, 2021-07-15:
> @Vudentz FYI, in case you have additional comments to this.

@Vudentz ping on this one, would be good to get your input.

alwa-nordic, 2021-09-30:
@KatrineNordic, it sounds like you've already done some digging here to find the cause. Do you have any suspicions on where in the code the bug is? Where does this queuing happen?

cvinayak, 2022-04-27:
@alwa-nordic is following up on possible solutions.

jori-nordic, 2022-08-31:
Working on it, managed to reproduce -a- deadlock (not sure if it's that specific one yet, but does look like it) on https://github.com/jori-nordic/zephyr/tree/l2cap-deadlock

PR #50476 (2022-09-26): Bluetooth: host: l2cap deadlock fix
This combination of commits fixes the symptoms seen in #34600.
It also adds an multilink l2cap test to stress the host in case other bugs like this crop up.

Fixes #34600.

A note on the test:
As it is a special usage of the l2cap CoC API, the stack will throw a lot of errors, this is expected. 
We might want to downgrade those errors (in the near future) to be less scary, as no actual data loss is happening.

Expected errors/warnings:
```
d_00: @00:01:54.482920  [00:01:54.482,910] <wrn> bt_l2cap: No credits to transmit packet
d_00: @00:01:54.518763  [00:01: [... cut]
closed 2022-09-26 by carlescufi