zephyrproject-rtos/zephyr #34601: sample: bluetooth: beacon: USAGE FAULT after few seconds on board b_l4s5i_iot01a

ABOSTM, 2021-04-27:
**Describe the bug**
sample/ bluetooth: beacon crashes on b_l4s5i_iot01a board with USAGE FAULT after about 20 seconds of execution

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p auto -b b_l4s5i_iot01a  samples/bluetooth/beacon/
2. west flash
3. See error

**Logs and console output**
> *** Booting Zephyr OS build zephyr-v2.5.0-2661-g45418dedc0bd  ***
> Starting Beacon Demo
> Bluetooth initialized
> Beacon started, advertising as DA:2A:C2:FC:65:8E (random)
> [00:00:00.000,000] <inf> spi_ll_stm32: CS control inhibited (no GPIO device)
> [00:00:00.008,000] <inf> bt_hci_core: Identity: DA:2A:C2:FC:65:8E (random)
> [00:00:00.008,000] <inf> bt_hci_core: HCI: version 4.1 (0x07) revision 0x3107, manufacturer 0x0030
> [00:00:00.008,000] <inf> bt_hci_core: LMP: version 4.1 (0x07) subver 0x0023
> [00:00:01.000,000] <err> os: ***** USAGE FAULT *****
> [00:00:01.000,000] <err> os:   Illegal use of the EPSR
> [00:00:01.000,000] <err> os: r0/a1:  0x00000000  r1/a2:  0xe000ed00  r2/a3:  0x20000b0c
> [00:00:01.000,000] <err> os: r3/a4:  0x20000460 r12/ip:  0x20676e69 r14/lr:  0x080056bd
> [00:00:01.000,000] <err> os:  xpsr:  0x60000000
> [00:00:01.000,000] <err> os: Faulting instruction address (r15/pc): 0x00000000
> [00:00:01.000,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
> [00:00:01.000,000] <err> os: Current thread: 0x20000460 (unknown)
> [00:00:01.005,000] <err> os: Halting system

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain gnuarmemb
 - Commit SHA 45418dedc0bde40d69b61dbfa6c0449f1098aba8

ABOSTM, 2021-04-27:
I found that doubling `spi_rx_stack` (in file ...\zephyr\drivers\bluetooth\hci\spi.c) fixes the issue
     `static K_KERNEL_STACK_DEFINE(spi_rx_stack, 512);`
But strangely, there is no issue with board nucleo_wb55rg.
So I am not sure increasing stack is the right fix.

ABOSTM, 2021-04-27:
^^ @joerchan @jhedberg @Vudentz

joerchan, 2021-04-27:
@ABOSTM Can you please enable the thread analyzer and report the stack usage for both boards. Run it with the increased stack size that you set, so that we don't get a crash.

Configurations to enable thread analyzer.
```
CONFIG_THREAD_NAME=y
CONFIG_THREAD_ANALYZER=y
CONFIG_THREAD_ANALYZER_AUTO=y
CONFIG_THREAD_ANALYZER_RUN_UNLOCKED=y
CONFIG_THREAD_ANALYZER_USE_PRINTK=y
```

I don't know these boards, what kind of architecture are they?
Can you compare the default board configurations to see if there is additional features selected?

ABOSTM, 2021-04-27:
@joerchan, thanks for your quick answer.
Let's forget about nucleo_wb55rg: it is a dualcore with Cortex M0 dedicated to BLE ... with shared memory. So no hci over spi.

b_l4s5i_iot01a has an STM32L4S5VIT6 chip based on Cortex M4.
It embed a Bluetooth (V4.1 compliant) SPBTLE-RF module, connected to Cortex M4 through SPI.

Below the log of b_l4s5i_iot01a with thread analyzer (and increased stack), please note that thread analyse come just after reset whereas USAGE FAULT comes after about 20 secondes of execution.
```
*** Booting Zephyr OS build zephyr-v2.5.0-2661-g45418dedc0bd  ***
Starting Beacon Demo
Thread analyze:
 0x20000620  Bluetooth initialized
Beacon started, advertising as F7:40:5E:58:FE:9F (random)
        : STACK: unused 328 usage 184 / 512 (35 %); CPU: 1 %
 BT RX               : STACK: unused 872 usage 152 / 1024 (14 %); CPU: 0 %
 BT TX               : STACK: unused 216 usage 200 / 416 (48 %); CPU: 14 %
 thread_analyzer     : STACK: unused 120 usage 392 / 512 (76 %); CPU: 1 %
 sysworkq            : STACK: unused 240 usage 784 / 1024 (76 %); CPU: 49 %
 logging             : STACK: unused 656 usage 112 / 768 (14 %); CPU: 0 %
 idle 00             : STACK: unused 288 usage 32 / 320 (10 %); CPU: 0 %
[00:00:00.000,000] <inf> spi_ll_stm32: CS control inhibited (no GPIO device)
[00:00:00.009,000] <inf> bt_hci_core: Identity: F7:40:5E:58:FE:9F (random)
[00:00:00.009,000] <inf> bt_hci_core: HCI: version 4.1 (0x07) revision 0x3107, manufacturer 0x0030
[00:00:00.009,000] <inf> bt_hci_core: LMP: version 4.1 (0x07) subver 0x0023
```

joerchan, 2021-04-27:
@ABOSTM With the thread analyzer in AUTO mode (`CONFIG_THREAD_ANALYZER_AUTO`) it should run the analyzer once every 60 second (by default, can be changed with `THREAD_ANALYZER_AUTO_INTERVAL`).

Can you get the thread analyzer values once it has had a chance to run for a while?

For convenience, could you also add a name to the thread.
```
--- a/drivers/bluetooth/hci/spi.c
+++ b/drivers/bluetooth/hci/spi.c
@@ -521,6 +521,7 @@ static int bt_spi_open(void)
                        (k_thread_entry_t)bt_spi_rx_thread, NULL, NULL, NULL,
                        K_PRIO_COOP(CONFIG_BT_DRIVER_RX_HIGH_PRIO),
                        0, K_NO_WAIT);
+       k_thread_name_set(&spi_rx_thread_data, "BT SPI RX");
```

ABOSTM, 2021-04-27:
Here it is:
```
Thread analyze:
 BT SPI RX           : STACK: unused 216 usage 296 / 512 (57 %); CPU: 0 %
 BT RX               : STACK: unused 872 usage 152 / 1024 (14 %); CPU: 0 %
 BT TX               : STACK: unused 216 usage 200 / 416 (48 %); CPU: 0 %
 thread_analyzer     : STACK: unused 128 usage 384 / 512 (75 %); CPU: 0 %
 sysworkq            : STACK: unused 240 usage 784 / 1024 (76 %); CPU: 0 %
 logging             : STACK: unused 288 usage 480 / 768 (62 %); CPU: 0 %
 idle 00             : STACK: unused 272 usage 48 / 320 (15 %); CPU: 99 %
```

ABOSTM, 2021-04-27:
By the way, with THREAD_ANALYZER_AUTO_INTERVAL=5 ... it took about 1,5 minutes to get a Thread analyze update.

PR #35065 (2021-05-14): drivers: bluetooth: hci: increase spi_rx_stack
drivers: bluetooth: hci: increase spi_rx_stack

Fixes #34601
commit c3d35d89 (2021-05-14): drivers: bluetooth: hci: increase spi_rx_stack

Thread analyze:
 BT SPI RX : STACK: unused 216 usage 296 / 512 (57 %)

Fixes #34601
closed 2021-05-14 by nashif