zephyrproject-rtos/zephyr #34649: Use spi and dma to refresh the screen very slowly

shuiyihang, 2021-04-28:
I am using the spi of stm32 mini to drive a st7735 screen, but I think it is too slow, so I did the following to enable dma, 
`&spi1 {
	pinctrl-0 = <&spi1_nss_master_pa4 &spi1_sck_master_pa5
		     &spi1_miso_master_pa6 &spi1_mosi_master_pa7>;
	status = "okay";
	dmas = <&dma1 3 1 0x20440 0x00>,
            <&dma1 2 1 0x20480 0x00>;
    dma-names = "tx", "rx";

	cs-gpios = <&gpioa 4 GPIO_ACTIVE_LOW>;
	lcd_st7735:st7735@0{
			compatible = "sitronix,st7735";
			reg=<0>;
			spi-max-frequency = <18000000>; /* 8MHz 36000000*/
			label = "ST7735";
			reset-gpios = <&gpiob 12 GPIO_ACTIVE_HIGH>;
			blk_gpios = <&gpioa 1 GPIO_ACTIVE_HIGH>;
			cmd-data-gpios = <&gpioc 4 GPIO_ACTIVE_HIGH>;
			width = <129>;
			height = <160>;
	};
};`
but there was a problem and a compilation error occurred:

`error:'LL_DMA_InitTypeDef' has no member named'PeriphRequest'
  461 | DMA_InitStruct.PeriphRequest = config->dma_slot;`

So I commented out this line and successfully generated the firmware. After burning, I found that the speed of DMA is much slower than that of pure spi. 

1. Use west build -t menuconfig to open spi's dma transmission
2. Modify the device tree
3. build

I want to make the refreshing speed after using dma faster than pure hardware spi

erwango, 2021-05-04:
Same as #34543, closing

erwango, 2021-05-10:
Removing bug as initial feature is not merged yet (https://github.com/zephyrproject-rtos/zephyr/pull/34666)

erwango, 2021-11-29:
@shuiyihang ^^

erwango, 2022-01-28:
@shuiyihang I'm closing the point, feel free to reopen if required

closed 2022-01-28 by erwango