zephyrproject-rtos/zephyr #34658: TF-M integration samples do not work with GNU ARM Embedded having GCC v10.x.x

ioannisg, 2021-04-28:
**Describe the bug**
I am trying to run the TF-M integration samples (on any supported platform) using the GNU ARM Embedded toolchain, and the samples are failing.

**To Reproduce**
Steps to reproduce the behavior:
1. Build and run any of the tfm integration samples (e.g. psa_level1, tfm_ipc) for any supported platform (including AN521 Qemu)
4. See error. For example, in tfm_ipc the psa_call() service is failing (see the respective log)

**Expected behavior**
The samples should not fail.

**Impact**
We cannot claim that the tf-m integration samples work with GNU ARM Embedded newer versions (10.x)
The Zephyr SDK works fine, probably because it contains a patched version.

**Logs and console output**

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain GNU ARM Embedded with GCC version 10.2.1
 - current master but also v2.5.0 Zephyr

One option is to list this as a known issue for Zephyr v2.6.0.

ioannisg, 2021-04-28:
@galak FYI

microbuilder, 2021-04-28:
Tested on OS X and fails with the following Arm GNU Toolchain releases:

- `10-2020-q2-preview` (GCC 10.1.1)
- `10-2020-q4-major` (GCC 10.2.1)

This seems to be the missing patch, which has been applied in the Zephyr SDK: https://github.com/zephyrproject-rtos/sdk-ng/blob/v0.12.4/patches/gcc/10.2.0/0016-gcc-libgcc-t-arm-Fix-mcmse-compile-test-for-setting-.patch

Arm is aware of this issue, and it should be addressed in the next quarterly release of the toolchain.

The question is if that release will happen before the 2.6.0 code freeze, or before 2.6.0 is tagged. 

Given that it's the end of the quarter this week, hopefully a release is pending soon from Arm(?).

## Steps to reproduce

Build with QEMU (making sure you are using one of the above toolchains):

```bash
$ echo $ZEPHYR_TOOLCHAIN_VARIANT 
gnuarmemb
$ echo $GNUARMEMB_TOOLCHAIN_PATH
/usr/local/prog/gcc-arm-none-eabi-10-2020-q4-major/
$ west build -p auto -b mps2_an521_nonsecure zephyr/samples/tfm_integration/tfm_ipc -t run
...
```
Output:
```
*** Booting Zephyr OS build zephyr-v2.5.0-2745-g4e6cb116f87b  ***
The version of the PSA Framework API is 257.
The minor version is 1.
Connect success!
TFM service support version is 1.
psa_call is failed!
Connect success!
Call failed!
Connect success!
Call failed!
TF-M IPC on mps2_an521_nonsecure
```

galak, 2021-04-28:
the arm gcc is known to be buggy for tfm

galak, 2021-04-28:
See my comments here:

https://lists.trustedfirmware.org/pipermail/tf-m/2021-February/001498.html

galak, 2021-04-28:
This is fixed in upstream gcc 10.3.0 now.

galak, 2021-05-04:
@ioannisg any update here.  Can we close since this isn't an issue with TFM code, but the ARM toolchain that we can't address/fix.

ioannisg, 2021-05-04:
> @ioannisg any update here. Can we close since this isn't an issue with TFM code, but the ARM toolchain that we can't address/fix.

Yeap, sure. Don't recall exactly why I wanted to track this in Zephyr, maybe just raise awareness.