zephyrproject-rtos/zephyr #34659: ASSERTION FAIL [err == 0] @ WEST_TOPDIR/zephyr/subsys/bluetooth/host/hci_core.c:333, k_sem_take failed with err -11

caco3, 2021-04-28:
**Describe the bug**
We have an application based on a nrf52840. We see `k_sem_take failed with err -11` assertion when we stress the device with a lot of notifications and read requests at the same time.

The system is used for a measurement device. We run an external ADC at 10 kSamples/s and read its data through SPI DMA transfers. After 30 samples (=> 333 Hz) we trigger an Interrupt Handler which copies the data into another buffer.
Then we we release a semaphore. A Kernel Thread is waiting for this semaphore so it can grab the data from the buffer and call `bt_gatt_notify()` with it. DLE is enabled and the data length set to 251.

Code snippets:
```
K_SEM_DEFINE(dataReadyToBeSend, 0, 1);
K_THREAD_DEFINE(DataStreaming, 10240, Thread_DataStreaming, NULL, NULL, NULL, 0, 0, 0);

#define TIMER_IRQ_PRIORITY  0

void init() {
    IRQ_DIRECT_CONNECT(TIMER2_IRQn, TIMER_IRQ_PRIORITY, TIMER2_IRQHandler, 0);
}

void TIMER2_IRQHandler(nrf_timer_event_t event_type, void *p_context) {
    [Copy data from DMA buffer to buffer...]
    k_sem_give(&dataReadyToBeSend);
}

static void Thread_DataStreaming(void) {
    [..]
    while (1) {
        semRet = k_sem_take(&dataReadyToBeSend, K_MSEC(1000));
        err = bt_gatt_notify(NULL, &myAttr, data, len);
    }
}
```

Log:
```
[..]
[00:00:04.257,873] <dbg> bt.connected: Update 2MBit PHY
[00:00:04.257,934] <inf> bt: Setting the Connection Parameters: Interval min: 7ms, Interval max: 4000ms, Latency: 0, timout: 5000ms...
[..]
[00:00:04.906,433] <dbg> bt.le_phy_updated: PHY updated!
[00:00:04.906,463] <dbg> bt.showConnectionInfo: Connection Information:
[00:00:04.906,463] <dbg> bt.showConnectionInfo:  - Role: Slave
[00:00:04.906,463] <dbg> bt.showConnectionInfo:  - Connection Type: LE
[00:00:04.906,494] <dbg> bt.showConnectionInfo:    PHY: TX: LE_PHY_2M, RX: LE_PHY_2M
[00:00:04.906,524] <dbg> bt.showDataLengthInfo:    Data Length: TX Max length: 251, TX Max time: 1064us, RX Max length: 251, RX Max time: 1064us
[..]
[00:00:56.955,780] <inf> bt_service:  13212 data notif. sent ( 627 kbit/s), Meas Data Buffer Level:  0.0% (   0 samples in buffer), TX RSSI: -56 dBm
[00:00:57.172,241] <inf> bt_service:  13284 data notif. sent ( 622 kbit/s), Meas Data Buffer Level:  0.0% (   0 samples in buffer), TX RSSI: -52 dBm
[00:00:57.389,099] <inf> bt_service:  13356 data notif. sent ( 625 kbit/s), Meas Data Buffer Level:  0.0% (   0 samples in buffer), TX RSSI: -59 dBm
[00:00:57.605,255] <inf> bt_service:  13428 data notif. sent ( 622 kbit/s), Meas Data Buffer Level:  0.0% (   0 samples in buffer), TX RSSI: -53 dBm
[00:00:57.821,899] <inf> bt_service:  13500 data notif. sent ( 625 kbit/s), Meas Data Buffer Level:  0.0% (   0 samples in buffer), TX RSSI: -54 dBm
[00:00:58.039,550] <inf> bt_service:  13573 data notif. sent ( 627 kbit/s), Meas Data Buffer Level:  0.0% (   0 samples in buffer), TX RSSI: -53 dBm
[00:00:58.255,859] <inf> bt_service:  13645 data notif. sent ( 625 kbit/s), Meas Data Buffer Level:  0.0% (   0 samples in buffer), TX RSSI: -57 dBm
[00:00:58.473,419] <inf> bt_service:  13718 data notif. sent ( 622 kbit/s), Meas Data Buffer Level:  0.0% (   0 samples in buffer), TX RSSI: -69 dBm
[00:00:58.690,795] <inf> bt_service:  13790 data notif. sent ( 627 kbit/s), Meas Data Buffer Level:  0.0% (   0 samples in buffer), TX RSSI: -53 dBm
[00:00:58.907,226] <inf> bt_service:  13862 data notif. sent ( 622 kbit/s), Meas Data Buffer Level:  0.0% (   0 samples in buffer), TX RSSI: -58 dBm
[00:00:59.124,572] <inf> bt_service:  13935 data notif. sent ( 630 kbit/s), Meas Data Buffer Level:  0.0% (   0 samples in buffer), TX RSSI: -63 dBm
[00:01:00.192,047] <wrn> ringbuffer: Buffer full, overwriting oldest sample! (w: 1441, r: 1470, in: 433440, out: 424470)
[00:01:00.192,047] <wrn> ringbuffer: Buffer full, overwriting oldest sample! (w: 1442, r: 1471, in: 433441, out: 424470)

ASSERTION FAIL [err == 0] @ WEST_TOPDIR/zephyr/subsys/bluetooth/host/hci_core.c:311
        k_sem_take failed with err -11
[00:01:09.351,501] <err> os: r0/a1:  0x00000003  r1/a2:  0x00000001  r2/a3:  0x40002000

*** Booting Zephyr OS build zephyr-v2.5.0-2746-g1cc73074d054  ***      // <= WATCHDOG RESET
```

What is the best way to trace this further down?

prj.conf (sanitized):
```
### General Kernel Configuration
CONFIG_LOG=y
CONFIG_UART_CONSOLE=y
CONFIG_ASSERT=y

## Logging (used in the defines)
CONFIG_LOG_STRDUP_BUF_COUNT=50
CONFIG_LOG_BUFFER_SIZE=65536
CONFIG_LOG_STRDUP_MAX_STRING=100

CONFIG_NEWLIB_LIBC=y              # Needed for logging only
CONFIG_NEWLIB_LIBC_FLOAT_PRINTF=y # Needed for logging only

## Optimization
CONFIG_SPEED_OPTIMIZATIONS=y
CONFIG_BT_CTLR_OPTIMIZE_FOR_SPEED=y

## Bluetooth Low Energy
# Increased stack due to settings API usage
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=2048

# Bluetooth
CONFIG_BT=y
CONFIG_BT_DEBUG_LOG=y
CONFIG_BT_SMP=y
CONFIG_BT_SIGNING=y
CONFIG_BT_PERIPHERAL=y
CONFIG_BT_DIS=y
CONFIG_BT_ATT_PREPARE_COUNT=5
CONFIG_BT_BAS=y
CONFIG_BT_PRIVACY=y
CONFIG_BT_DEVICE_APPEARANCE=833
CONFIG_BT_DEVICE_NAME_DYNAMIC=y
CONFIG_BT_DEVICE_NAME_MAX=65

CONFIG_BT_KEYS_OVERWRITE_OLDEST=y # Overwrite oldest keys with new ones if key storage is full

# Below is setup to be able to update DIS at runtime
# There is still a warning "bt_hci_core: Unable to store name" of which I could not get rid off.
# See also https://github.com/zephyrproject-rtos/zephyr/issues/22478
CONFIG_BT_SETTINGS=y # Needed for settings_runtime_set()
CONFIG_SETTINGS=y # Needed for settings_runtime_set()
CONFIG_SETTINGS_RUNTIME=y # Needed for settings_runtime_set()
CONFIG_SETTINGS_NONE=y

CONFIG_BT_GATT_CACHING=n # To avoid error "bt_gatt: Failed to save Database Hash"

# BLE Throughput Maximization
# Adapted from https://github.com/nrfconnect/sdk-nrf/blob/master/samples/bluetooth/throughput/prj.conf
# Not all parameters might be needed/usefull!
CONFIG_HEAP_MEM_POOL_SIZE=2048
CONFIG_BT_RX_BUF_LEN=258
CONFIG_BT_GATT_CLIENT=y
CONFIG_BT_ATT_PREPARE_COUNT=2
CONFIG_BT_CONN_TX_MAX=255

CONFIG_BT_L2CAP_TX_BUF_COUNT=10
CONFIG_BT_L2CAP_TX_MTU=247
CONFIG_BT_L2CAP_RX_MTU=247
CONFIG_BT_L2CAP_DYNAMIC_CHANNEL=y

CONFIG_BT_CTLR_PHY_2M=y
CONFIG_BT_CTLR_TX_BUFFERS=10
CONFIG_BT_CTLR_TX_BUFFER_SIZE=251
CONFIG_BT_USER_PHY_UPDATE=y
CONFIG_BT_USER_DATA_LEN_UPDATE=y

CONFIG_BT_CTLR_DATA_LENGTH_MAX=251
CONFIG_BT_CTLR_ADVANCED_FEATURES=y
CONFIG_BT_CTLR_XTAL_THRESHOLD=500000

CONFIG_BT_PHY_UPDATE=y
CONFIG_BT_DATA_LEN_UPDATE=y

CONFIG_BT_CTLR_CONN_RSSI=y
CONFIG_BT_CTLR_TX_PWR_DYNAMIC_CONTROL=y

# This seems to fix the following assertion:
#     ASSERTION FAIL [!radio_is_ready()] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_conn.c:317
# See https://github.com/zephyrproject-rtos/zephyr/issues/29994#issuecomment-739623736 and https://jira.kistler.com/browse/WRCD-524
CONFIG_ZERO_LATENCY_IRQS=y
CONFIG_BT_CTLR_ZLI=y
```

We also played with the following parameters but did not see an improvement:
```
CONFIG_BT_LLL_VENDOR_NORDIC=y
CONFIG_BT_CTLR_LLL_PRIO=0
CONFIG_BT_CTLR_ULL_HIGH_PRIO=0
CONFIG_BT_CTLR_ULL_LOW_PRIO=0
```
and changed `TIMER_IRQ_PRIORITY ` to `3`

**Environment :**
 - OS: Ubuntu 20.04,
 - Toolchain: zephyr-sdk-0.12.2
 - Zephyr Project (Tag `2.5.0`) but also tested with 1cc73074d0540caa8dd7ea9501afefea2e410791

Reference to code: https://github.com/zephyrproject-rtos/zephyr/blob/fe7c2efca800a0cf1bccf23aefe08b3c4beb88bf/subsys/bluetooth/host/hci_core.c#L333

cvinayak, 2021-04-28:
@caco3 Relates to #25917

@joerchan  can #26057 be used as a fix/workaround for the deadlock?

caco3, 2021-04-28:
Thanks for the reference to this open patch. I ported it to `2.5.0` and gave it a try.
Patch is [here](https://github.com/zephyrproject-rtos/zephyr/pull/34681/files), I just did it quickly but it compiled and BLE communication worked.

Sadly, I still experience the crashes, but it looks a bit different now:
```
[00:03:31.587,738] <inf> bt_wrcd_service_helper:  63270 data notif. sent (   0 kbit/s), Meas Data Buffer Level: 99.7% (8970 samples in buffer), TX RSSI: -54 dBm
[00:03:31.804,016] <inf> bt_wrcd_service_helper:  63270 data notif. sent (   0 kbit/s), Meas Data Buffer Level: 99.7% (8970 samples in buffer), TX RSSI: -51 dBm
[00:03:32.020,172] <inf> bt_wrcd_service_helper:  63270 data notif. sent (   0 kbit/s), Meas Data Buffer Level: 99.7% (8970 samples in buffer), TX RSSI: -53 dBm
[00:03:32.236,389] <inf> bt_wrcd_service_helper:  63270 data notif. sent (   0 kbit/s), Meas Data Buffer Level: 99.7% (8970 samples in buffer), TX RSSI: -53 dBm
[00:03:32.452,575] <inf> bt_wrcd_service_helper:  63270 data notif. sent (   0 kbit/s), Meas Data Buffer Level: 99.7% (8970 samples in buffer), TX RSSI: -76 dBm
[00:03:32.668,701] <inf> bt_wrcd_service_helper:  63270 data notif. sent (   0 kbit/s), Meas Data Buffer Level: 99.7% (8970 samples in buffer), TX RSSI: -65 dBm
[00:03:32.884,887] <inf> bt_wrcd_service_helper:  63270 data notif. sent (   0 kbit/s), Meas Data Buffer Level: 99.7% (8970 samples in buffer), TX RSSI: -51 dBm
[00:03:33.112,030] <inf> bt_wrcd_service_helper:  63270 data notif. sent (   0 kbit/s), Meas Data Buffer Level: 99.7% (8970 samples in buffer), TX RSSI: -55 dBm
[00:03:33.153,564] <err> os: ***** USAGE FAULT *****
ASSERTION FAIL [out_ctx->contro*** Booting Zephyr OS build zephyr-v2.5.0  ***
=> Watchdog reset
```

Another time it looked as following:
```
[00:01:48.125,000] <inf> bt_wrcd_service_helper:  27321 data notif. sent (   0 kbit/s), Meas Data Buffer Level: 99.7% (8970 samples in buffer), TX RSSI: -52 dBm
[00:01:48.341,217] <inf> bt_wrcd_service_helper:  27321 data notif. sent (   0 kbit/s), Meas Data Buffer Level: 99.7% (8970 samples in buffer), TX RSSI: -51 dBm
[00:01:48.557,525] <inf> bt_wrcd_service_helper:  27321 data notif. sent (   0 kbit/s), Meas Data Buffer Level: 99.7% (8970 samples in buffer), TX RSSI: -51 dBm
[00:01:48.773,773] <inf> bt_wrcd_service_helper:  27321 data notif. sent (   0 kbit/s), Meas Data Buffer Level: 99.7% (8970 samples in buffer), TX RSSI: -51 dBm
[00:01:48.989,929] <inf> bt_wrcd_service_helper:  27321 data notif. sent (   0 kbit/s), Meas Data Buffer Level: 99.7% (8970 samples in buffer), TX RSSI: -72 dBm
[00:01:49.206,237] <inf> bt_wrcd_service_helper:  27321 data notif. sent (   0 kbit/s), Meas Data Buffer Level: 99.7% (8970 samples in buffer), TX RSSI: -50 dBm
[00:01:49.422,393] <inf> bt_wrcd_service_helper:  27321 data notif. sent (   0 kbit/s), Meas Data Buffer Level: 99.7% (8970 samples in buffer), TX RSSI: -53 dBm
[00:01:49.638,519] <inf> bt_wrcd_service_helper:  27321 data notif. sent (   0 kbit/s), Meas Data Buffer Level: 99.7% (8970 samples in buffer), TX RSSI: -65 dBm
[00:01:49.795,806] <err> os: ***** USAGE FAULT *****
[00:01:49.795,837] <err> os:   Illegal use o*** Booting Zephyr OS build zephyr-v2.5.0  ***
=> CPU-Lockup Reset
```

joerchan, 2021-04-30:
> joerchan can 26057 be used as a fix/workaround for the deadlock?

I think if this is the fix needed the symptom should be possible to see in a sniffer log, it should be locked with NACKS at the LL.

The USAGE fault could mean a stack overflow, or calling of an invalid function pointer.

caco3, 2021-05-02:
Thanks for the tip. I did a trace using an nRF52840-DK and Wireshark.
The trace is attached: [Wireshark.pcapng.gz](https://github.com/zephyrproject-rtos/zephyr/files/6411478/Wireshark.pcapng.gz)
How ever I do not see any error in it.
Below some screenshots for further explaining the happening.

### Measure Mode: Data notifications (handle 0x28, 241 bytes of payload) every 1.4 ms
[image]
This works reliable for hours!

### Read other Characteristic (Handle 0x39) while in Measure Mode
The first and the 3th black lines are the read request, the 2nd black and the blue lines are the responses.
(For testing purpose we added a log back-end which we can poll entry by entry through this characteristic).
[image]

### Crash after some minutes
Reading the log Characteristic (Handle 0x39) while in Measure Mode works for some minutes but the crashes with a Watchdog Reset (60 seconds window):
[image]

### Device Log
```[00:02:19.096,557] <inf> bt_service_helper:  38449 data notif. sent ( 622 kbit/s), Meas Data Buffer Level:  0.0% (   0 samples in buffer), TX RSSI: -75 dBm
[00:02:19.312,286] <inf> bt_service_helper:  38521 data notif. sent ( 625 kbit/s), Meas Data Buffer Level:  0.0% (   0 samples in buffer), TX RSSI: -49 dBm
[00:02:19.528,533] <inf> bt_service_helper:  38593 data notif. sent ( 625 kbit/s), Meas Data Buffer Level:  0.0% (   0 samples in buffer), TX RSSI: -65 dBm
[00:02:19.745,971] <inf> bt_service_helper:  38666 data notif. sent ( 630 kbit/s), Meas Data Buffer Level:  0.0% (   0 samples in buffer), TX RSSI: -47 dBm
[00:02:19.964,141] <inf> bt_service_helper:  38739 data notif. sent ( 627 kbit/s), Meas Data Buffer Level:  0.0% (   0 samples in buffer), TX RSSI: -52 dBm
[00:02:20.194,580] <inf> bt_service_helper:  38815 data notif. sent ( 622 kbit/s), Meas Data Buffer Level:  0.0% (   0 samples in buffer), TX RSSI: -49 dBm
[00:02:20.411,682] <inf> bt_service_helper:  38888 data notif. sent ( 625 kbit/s), Meas Data Buffer Level:  0.0% (   0 samples in buffer), TX RSSI: -48 dBm
[00:02:20.627,929] <inf> bt_service_helper:  38960 data notif. sent ( 625 kbit/s), Meas Data Buffer Level:  0.0% (   0 samples in buffer), TX RSSI: -47 dBm
[00:02:20.846,069] <inf> bt_service_helper:  39033 data notif. sent ( 622 kbit/s), Meas Data Buffer Level:  0.0% (   0 samples in buffer), TX RSSI: -51 dBm
[00:02:21.063,720] <inf> bt_service_helper:  39105 data notif. sent ( 625 kbit/s), Meas Data Buffer Level:  0.0% (   0 samples in buffer), TX RSSI: -47 dBm
[00:02:21.279,815] <inf> bt_service_helper:  39177 data notif. sent ( 625 kbit/s), Meas Data Buffer Level:  0.0% (   0 samples in buffer), TX RSSI: -47 dBm
[00:02:21.497,283] <inf> bt_service_helper:  39250 data notif. sent ( 622 kbit/s), Meas Data Buffer Level:  0.0% (   0 samples in buffer), TX RSSI: -61 dBm
[00:02:21.714,752] <inf> bt_service_helper:  39322 data notif. sent ( 627 kbit/s), Meas Data Buffer Level:  0.0% (   0 samples in buffer), TX RSSI: -50 dBm
[00:02:21.933,135] <inf> bt_service_helper:  39395 data notif. sent ( 625 kbit/s), Meas Data Buffer Level:  0.0% (   0 samples in buffer), TX RSSI: -49 dBm
[00:02:22.151,397] <inf> bt_service_helper:  39468 data notif. sent ( 622 kbit/s), Meas Data Buffer Level:  0.0% (   0 samples in buffer), TX RSSI: -58 dBm
[00:02:22.369,323] <inf> bt_service_helper:  39540 data notif. sent ( 625 kbit/s), Meas Data Buffer Level:  0.0% (   0 samples in buffer), TX RSSI: -49 dBm
[00:02:22.431,549] <dbg> peripherals.peripherals_getBatteryPercentage: Battery Voltage: 4018 mV => 100%
[00:02:22.585,083] <inf> bt_service_helper:  39612 data notif. sent ( 627 kbit/s), Meas Data Buffer Level:  0.0% (   0 samples in buffer), TX RSSI: -47 dBm
[00:02:22.812,744] <inf> bt_service_helper:  39688 data notif. sent ( 625 kbit/s), Meas Data Buffer Level:  0.0% (   0 samples in buffer), TX RSSI: -48 dBm
[00:02:23.029,174] <inf> bt_service_helper:  39760 data notif. sent ( 622 kbit/s), Meas Data Buffer Level:  0.0% (   0 samples in buffer), TX RSSI: -48 dBm
[00:02:23.246,063] <inf> bt_service_helper:  39833 data notif. sent ( 625 kbit/s), Meas Data Buffer Level:  0.0% (   0 samples in buffer), TX RSSI: -50 dBm
[00:02:23.463,378] <inf> bt_service_helper:  39905 data notif. sent ( 627 kbit/s), Meas Data Buffer Level:  0.0% (   0 samples in buffer), TX RSSI: -49 dBm
[00:02:23.678,894] <inf> bt_service_helper:  39977 data notif. sent ( 627 kbit/s), Meas Data Buffer Level:  0.0% (   0 samples in buffer), TX RSSI: -57 dBm
[00:02:23.895,965] <inf> bt_service_helper:  40049 data notif. sent ( 622 kbit/s), Meas Data Buffer Level:  0.0% (   0 samples in buffer), TX RSSI: -62 dBm
[00:02:24.112,579] <inf> bt_service_helper:  40121 data notif. sent ( 622 kbit/s), Meas Data Buffer Level:  0.0% (   0 samples in buffer), TX RSSI: -47 dBm
[00:02:24.328,491] <inf> bt_service_helper:  40193 data notif. sent ( 625 kbit/s), Meas Data Buffer Level:  0.0% (   0 samples in buffer), TX RSSI: -48 dBm
[00:02:24.545,440] <inf> bt_service_helper:  40266 data notif. sent ( 630 kbit/s), Meas Data Buffer Level:  0.0% (   0 samples in buffer), TX RSSI: -50 dBm
[00:02:24.762,939] <inf> bt_service_helper:  40338 data notif. sent ( 622 kbit/s), Meas Data Buffer Level:  0.0% (   0 samples in buffer), TX RSSI: -49 dBm
[00:02:24.978,851] <inf> bt_service_helper:  40410 data notif. sent ( 625 kbit/s), Meas Data Buffer Level:  0.0% (   0 samples in buffer), TX RSSI: -61 dBm
[... cut ...]
I: Swap type: none
I: Bootloader chainload address offset: 0x12000
I: Jumping to the first image slot
*** Booting Zephyr OS build zephyr-v2.5.0  ***
[..]
[00:00:00.004,272] <dbg> watchdog.initWatchdog: Configured the Watchdog to reset the device if it does not get feed withing 60 seconds
[00:00:00.004,302] <inf> misc: Reset Reason: Watchdog
[00:00:00.004,302] <err> main: ####################
[00:00:00.004,302] <err> main: ## WATCHDOG RESET ##
[00:00:00.004,302] <err> main: ####################
```

carlescufi, 2021-05-05:
@caco3 so to summarize:

- You see the issue both in v2.5.0 and also current master
- [This patch](https://github.com/zephyrproject-rtos/zephyr/pull/26057/files) did not help

Is this correct?

caco3, 2021-05-10:
@carlescufi 

I am not 100% sure if the patch did help. Fact is, running stock 2.5.0 or 1cc73074d054 leads to 
```
ASSERTION FAIL [err == 0] @ WEST_TOPDIR/zephyr/subsys/bluetooth/host/hci_core.c:311
        k_sem_take failed with err -11
=> Watchdog Reset
```

Then I ported the mentioned patch to 2.5.0 and gave it another try. The system still crashes but the trace looks a bit different, so I am not sure if it is the same issue:
```
[00:01:49.795,806] <err> os: ***** USAGE FAULT *****
[00:01:49.795,837] <err> os:   Illegal use o*** Booting Zephyr OS build zephyr-v2.5.0  ***
=> CPU-Lockup Reset
```

Is there a way to get more output (printk) in case of such a crash?

Note 1:
I also can see such crashes when I do *not* stress the system with permanent read requests but instead just send notifications at a weak RSSI!

Note 2: 
Unluckely I will be away for the next couple of weeks.
I will tell my co-workers to have a look on this issue.

cvinayak, 2021-05-12:
> I also can see such crashes when I do not stress the system with permanent read requests but instead just send notifications at a weak RSSI!

@joerchan will tx thread timeout if num complete does not arrive for long time until supervision timeout?

joerchan, 2021-05-12:
No, TX thread waits for num completes indefinitely.

carlescufi, 2021-05-20:
@caco3 we will need more information in order to try and figure out the root cause. Ideally we'd need a small sample that reproduces this on an Nordic DK so we can then analyze the issue.

carlescufi, 2021-05-27:
@caco3 when you come back please do comment on this issue, since Zephyr 2.6.0 is currently being stabilizied and ideally we'd like to include a fix for this.

caco3, 2021-06-01:
@carlescufi I will look into this after I get back to work. How ever this will only be at beginning of July, most likely to late for the `2.6.0` release, sry!

caco3, 2021-07-05:
Just a little update:
I am now back @ work and migrated to `2.6.0`. As expected the issue still is the same.

I will now try to build a minimal example to reproduce the issue.

carlescufi, 2021-07-15:
> I will now try to build a minimal example to reproduce the issue.

Thanks @caco3. Please do let us know when you have the sample so we can analyze it.

caco3, 2021-07-16:
We are trying intensively to trace the issue down as deep that we can provide a minimal example.

I ported the mentioned patch to Zephyr `2.6.0` (see https://github.com/caco3/zephyr/commit/434370ba2eef3f9a3f16ca0fb9258da88968b629) and applied it to the measuring device as well as the BLE receiver dongle (a nrf52840 dongle running the Zephyr HCI_USB example) connected to the host system. This brings quite some improvement; **The measurement device now no longer crashes but only disconnects!**

It will need further testing but I strongly believe that this patch helps against the crashing!

We still have stability issues but maybe they are no longer directly related to this issue.
In the setup as we use it, the host system is an ARM Cortex-A8 device on which we run a Yocto based Linux System. As mentioned earlier, we only have the stability issue when we stress the BLE connection additionally with extra GATT reads while we send notifications at a high rate from the measurement device to the host.
I tried to reproduce the issue on a more powerful host (x86 PC running Ubuntu 20.04) but haven't been able to get it crashing in the same way, althrough the software is almost identical (same dongle, same bluez, similair kernel). This leads me to the impression, that the sudden disconnects might be performance issue on receiver side. How ever we will need to investigate more for this.

carlescufi, 2021-07-29:
>the host system is an ARM Cortex-A8

I feel we need to clarify the setup here, because we are confused.
If the Host is a Cortex-A8 running Linux, then what Bluetooth Host are you using? Because the patch you applied is for the Zephyr Bluetooth Host, but if the Host is Linux then I don't see how that can help.
Could you please describe your setup in detail, including all ICs present, the physical connections among them as well as the exact software stacks that you are using?

caco3, 2021-08-04:
Sorry for the confusion. I try to describe it better below:

**Setup A:**
```
 --------------------           --------------------           --------------------
| Measurement Device |   BLE   | BLE Receiver       |   USB   | Processing Host    |
| Running patched    |  <--->  | Running patched    |  <--->  | Yocto based Linux  |
| Zephyr 2.6 (nrf52) |         | Zephyr 2.6 (nrf52) |         | (Cortex-A8)        |
 --------------------           --------------------           --------------------
```
**Setup B:**
Same as Setup A, but we replaced the Processing Host by a x86 PC running Ubuntu 20.04

---

The Measurement Device sends data at around 625 kbit/s to the Processing Host using GATT Notify.
This works fine until the Processing Host (Cortex-A8) starts to do GATT reads excessively while still receiving the notifications!

**Without** the patch the `Measurement Device` **crashes**.
**With** the patch, it **no longer crashes**, but it simply **stops transmitting** any notification or read response.
We do not see any error message or disconnect, just no more communication.
<details>
<summary>Output of `btmon` (run on the Cortex-A8):</summary>

```
> ACL Data RX: Handle 0 flags 0x02 dlen 248                                                                                                                                                                                #1433 [hci0] 18.701375
      ATT: Handle Value Notification (0x1b) len 243
        Handle: 0x0028
          Data: b6499dddd0f4e9d8b6369dc3d0e8e9beb6199d9bd0d5e997b6059d83d0c9e97eb60f9d90d0d0e98db6009d7bd0c6e97ab5f09d67d0bce966b5df9d51d0b1e952b5cc9d37d0a5e938b5bc9d22d09ae923b5a49d03d08ce904b58c9ce3d07de8e5b5769cc5d06ee8c8b5609ca8d060e8abb5499c8ad051e88fb5349c6ed044e873b5219c53d036e859b50b9c35d028e83cb4f59c18d01ae820b4e19bfdd00ee807b4cd9be2d001e7edb4b49bc1cff1e7ccb49b9ba0cfe0e7acb4879b85cfd4e793b46a9b5fcfc2e76cb44e9b38cfafe747b4379b1dcfa1e72bb4069adccf81e6e9b3d69a9ccf62e6a9b3c59a84cf57e6948a
> ACL Data RX: Handle 0 flags 0x02 dlen 248                                                                                                                                                                                #1434 [hci0] 18.702721
      ATT: Handle Value Notification (0x1b) len 243
        Handle: 0x0028
          Data: b3b39a6bcf4be67bb39d9a4ecf3de65fb3849a2ccf2ce63eb36b9a0acf1ce61eb35999f2cf12e607b34699d8cf06e5eeb33299bfcef9e5d4b32f99bbcef7e5d2b33599c2cefbe5dcb31f99a5ceede5c0b317999bcee7e5b7b3079986cedee5a2b2ee9966cecfe583b2df9952cec4e56eb2cc9937ceb6e554b2b8991cceabe53cb2a49903ce9ee523b28e98e5ce8fe506b27a98c9ce83e4edb26898b0ce77e4d5b2529892ce69e4b8b238986fce59e496b2229853ce4ae47ab210983ace3ee462b1fb981fce31e447b1e69803ce23e42bb1cf97e5ce15e40eb1bb97c9ce08e3f4b1a697aecdfae3dab1919793cdede3c18b
> ACL Data RX: Handle 0 flags 0x02 dlen 248                                                                                                                                                                                #1435 [hci0] 18.716373
      ATT: Handle Value Notification (0x1b) len 243
        Handle: 0x0028
          Data: aabf8e79c98bdb0baa868e2dc966dac0aa518de6c944da7baa258daac928da41aa178d97c91eda30a9f98d6fc90cda0ba9d78d43c8f6d9dfa9b88d1ac8e3d9b8a99c8cf7c8d1d994a97e8ccfc8bed96da95d8ca2c8a8d940a9398c73c891d911a9168c44c87bd8e3a8fc8c20c86ad8c2a8e08bfac858d89ea8c68bd8c847d87ea8ae8bb7c838d85ea8928b90c825d839a8788b6ec814d817a85f8b4fc805d7f8a8448b2cc7f5d7d7a8298b06c7e3d7b2a80d8ae0c7d1d78ea7f58ac1c7c2d770a7d88a9bc7aed74aa7b78a70c799d71fa79e8a4fc789d6fea76b8a09c768d6baa74189d0c74cd682a72789adc73cd6608f
> ACL Data RX: Handle 0 flags 0x02 dlen 248                                                                                                                                                                                #1436 [hci0] 18.720519
      ATT: Handle Value Notification (0x1b) len 243
        Handle: 0x0028
          Data: a1d98298c3d2cf91a1c2827cc3c4cf74a1ac825ec3b6cf57a19d824bc3accf45a18c8236c3a2cf30a1728213c392cf10a15d81f8c385cef5a14981dcc378cedba13381bdc36acebfa11a819cc35ace9ea0fc8174c346ce76a0e38152c336ce55a0a78100c30ece06a08380d1c2f7cdd7a07480bdc2edcdc5a06280a4c2e1cdaea04f808ac2d6cd96a03b8070c2c9cd7da0238050c2b9cd5fa0108038c2adcd479ff9801bc29fcd2c9fe28000c290cd0e9fda8000c28ccd059fea8006c296cd1b9fd58000c288cd009fbb8000c277ccdf9f878000c255cc9c9f688000c241cc739f508000c232cc549f398000c223cc3692
< ACL Data TX: Handle 0 flags 0x00 dlen 7                                                                                                                                                                                  #1437 [hci0] 18.731080
      ATT: Read Request (0x0a) len 2
        Handle: 0x002b
@ MGMT Command: Get Connection Information (0x0031) plen 7                                                                                                                                                              {0x0003} [hci0] 19.058409
        LE Address: xx:xx:xx:4B:43:B4 (XXX)
< HCI Command: Read RSSI (0x05|0x0005) plen 2                                                                                                                                                                              #1438 [hci0] 19.058508
        Handle: 0
@ MGMT Command: Get Connection Information (0x0031) plen 7                                                                                                                                                              {0x0 [... cut ...]
@ MGMT Event: Command Status (0x0002) plen 3                                                                                                                                                                           {0x0002} [hci0] 149.199051
      Get Connection Information (0x0031)
        Status: Permission Denied (0x14)
```
</details>

Connection Parameters (Measurement Device): `Interval min: 7ms, Interval max: 7ms, Latency: 0, timout: 100ms`
Connection Information (Measurement Device):
```
- Role: Slave
- Connection Type: LE
  PHY: TX: LE_PHY_2M, RX: LE_PHY_2M
  Data Length: TX Max length: 251, TX Max time: 1064us, RX Max length: 251, RX Max time: 1064us
```

carlescufi, 2021-08-05:
@caco3 Thanks for the clarification. This issue seems to always happen when your ring buffer fills up:

```
[00:02:29.626,739] <wrn> ringbuffer: Buffer full, overwriting oldest sample! (w: 2011, r: 2040, in: 1271010, out: 1262040)
```

I have a feeling that the fix you backported (https://github.com/zephyrproject-rtos/zephyr/pull/26057) is only masking the real issue, which seems to be related to your ring buffer being full and then somehow preventing the scheduler from running the controller thread.  Can you confirm that the issue only happens when your ring buffer is full?

caco3, 2021-08-06:
I see your point about the ringbuffer, how ever I have to disappoint you there. For minimized latency, the ringbuffer is always tried to kept empty (See log in my initial post: `[00:00:57.389,099] <inf> bt_service:  13356 data notif. sent ( 625 kbit/s), Meas Data Buffer Level:  0.0% (   0 samples in buffer), TX RSSI: -59 dBm`). 

To prove the innocence of the ringbuffer, I started to toggle some GPIO pins on certain events and recorded it on an oscilloscope:
 * Yellow: Set high just before `bt_gatt_notify()` and cleared again afterwards:
   ```
        setTp0();
        err = bt_gatt_notify(NULL, &wrcdService.attrs[ATTR_DATA_INDEX], data, len);
        resetTp0();
   ```

 * Green: Set on start of a GATT read callback and reset on its end:
    ```
    BT_GATT_SERVICE_DEFINE(myService,BT_GATT_PRIMARY_SERVICE(&service),
       [..]

        /* Data Characteristic */
        BT_GATT_CHARACTERISTIC(
               &dataCharacteristic.uuid,
               BT_GATT_CHRC_NOTIFY,
               BT_GATT_PERM_READ,
               NULL, NULL, 0),
        BT_GATT_CCC(data_ccc_cfg_changed, BT_GATT_PERM_READ | BT_GATT_PERM_WRITE),

        /* Log Characteristic */
        BT_GATT_CHARACTERISTIC(
               &logCharacteristic.uuid,
               BT_GATT_CHRC_READ | BT_GATT_CHRC_NOTIFY,
               BT_GATT_PERM_READ,
               readLog, NULL, 0),
       BT_GATT_CCC(btLog_ccc_cfg_changed, BT_GATT_PERM_READ | BT_GATT_PERM_WRITE),

       [..]
    );
    
    static ssize_t readLog(struct bt_conn *conn, const struct bt_gatt_attr *attr, void *buf, uint16_t len, uint16_t offset) {
    setTp1();
    [..]
    ssize_t s = bt_gatt_attr_read(conn, attr, buf, len, offset, &logBuffer,
                sizeof(logBuffer.remainingMessages) + strlen(logBuffer.msg));
    resetTp1();
    return s;
   ```
 * Blue: Raised high when the ringbuffer overflows (firing of the `<wrn> ringbuffer: Buffer full, overwriting oldest sample! [..]` warning).

Screenshot 1:
Normal operation, around 300 Notifications per second.
![grafik](https://user-images.githubusercontent.com/1783586/128539104-856bbf72-63ce-420f-9b95-be688b98babb.png)

Screenshot 2:
Start of excessive GATT reads while still sending notifications.
And the issue occurring after a few seconds...
![grafik](https://user-images.githubusercontent.com/1783586/128539221-58f949c8-8ac0-4b2d-b024-3d6372d5a354.png)

Screenshot 3:
Zoomed into notify + GATT read.
![grafik](https://user-images.githubusercontent.com/1783586/128539458-eb9fd4c6-1750-4000-9f40-924410516060.png)

As you can see on screenshot 2 and 3, it works all fine for some seconds. Then the GATT read callback does not get called anymore and a bit later the notifications stop.
Also we can see that the buffer overflow warning only gets fired around 900 ms **after** the last notification. The ringbuffer fits 9000 samples which equals this time at the given 10 kSamples/s! This also means that the buffer only starts to fill up at the time the notifications stop!

As we can see, the yellow line stays high, meaning it entered the `bt_gatt_notify()` but did not leave it again!

---

Also I observed some other things, see the flow below:
1. Connect to the measurement device (Using setup 1 documented some post ago).
1. Start Data transmission (333 notifications/s).
    This would now run stable forever (tested >60h).
1. Start reading a GATT characteristic excessively.

After some seconds the transmission stops, and the ringbuffer starts filling up.
The sensor will continue running (no crash, no watchdog reset).
Looking into `bluetoothctl` on the host, the sensor is listed as connected. Trying to disconnect it from host side does not work (no warning, just stays listed as connected).
Also still no crash on sensor or watchdog reset.

Now, when I physically disconnect the BLE-Receiver (USB) from the host, following happens immediately:
 - `bluetoothctrl` immediately lists the sensor as disconnected.
 - Sometimes (maybe 1 in 10 times) the measurement device still thinks it is connected! Then itruns into a watchdog reset.
 - But mostly (maybe 9/10 times) the sensor immediately shows that it is disconnected. It stays running without issues.

In case the sensor continues running, I can connect the RBLE-Receiver to the host again, connect to the sensor and start data transmission again as if nothing happened!
This leads me to the believe, that the initial problem could lay in the the firmware of the BLE-Receiver!
As mentioned before, this is basically a configured Zephyr USB-HCI sample (https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/bluetooth/hci_usb).

main.c:
```
#include <zephyr.h>
#include <sys/printk.h>
#include <usb/usb_device.h>

void main(void)
{
	int ret;

	ret = usb_enable(NULL);
	if (ret != 0) {
		printk("Failed to enable USB");
		return;
	}
}
```

prj.conf (sanitized):
```
########################################
# System
########################################
# Some command handlers require a large stack.
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=2048

########################################
# Bluetooth
########################################
CONFIG_BT=y
# - RAW HCI access
CONFIG_BT_HCI_RAW=y

# - Maximum supported HCI RX buffer length
CONFIG_BT_BUF_ACL_RX_SIZE=254
CONFIG_BT_BUF_EVT_RX_SIZE=255
CONFIG_BT_BUF_CMD_TX_SIZE=255

# Bluetooth -> Bluetooth Controller
# - Maximum data length supported
CONFIG_BT_CTLR_DATA_LENGTH_MAX=251
# - Coded PHY
CONFIG_BT_CTLR_PHY_CODED=n
# - Tx Power
CONFIG_BT_CTLR_TX_PWR_PLUS_8=y

########################################
# USB device stack
########################################
CONFIG_USB=y
CONFIG_USB_DEVICE_STACK=y

# USB HCI
CONFIG_USB_DEVICE_BLUETOOTH=y
CONFIG_USB_DEVICE_BLUETOOTH_VS_H4=y
```

caco3, 2021-08-11:
A minor addition:
I replaced the receiver dongle with a nrf52840 DK board, enabled log output and added in `main()` a `while(1) { printk(". "); k_msleep(500); }` loop. Using this I can see that the receiver dongle does not crash!

How ever, it still does not respond anymore!
When I try to do a BLE scan using `bluetoothctrl`, I get `Failed to start discovery: org.bluez.Error.NotReady`.
Restarting `bluez `does not help.

cvinayak, 2021-08-20:
@caco3 Some Controller bugs were fixed last week related to an implementation handling race condition in disabling advertising, these fixes are upstream now.

There is a lot of information you have provided in this issue here. But correct my understanding:

> The measurement device now no longer crashes but only disconnects!

You were observing hci_core assertion on timeout taking the semaphore, which you no longer observe when using hci_usb sample with a host system?

To avoid further delay,
1. A simple sample be provided in a PR/branch and precise steps (hopefully, simple manual instructions) to reproduce the issue, or
2. I am available for a live video and desktop sharing, day time CET, so that we can hold a debug session to identity the cause in the Controller (You will need nRF52840 DK, logic analyzer to capture trace from 8 GPIO pins)

caco3, 2021-08-23:
I tried it with commit `8c31f31` but no improvement.

> You were observing hci_core assertion on timeout taking the semaphore, which you no longer observe when using hci_usb sample with a host system?

No, it no longer crashes because I applied the patch in https://github.com/zephyrproject-rtos/zephyr/pull/26057 to both the peripheral and the BLE host.

I am only able to provoke the issue when using the weak-powered `Cortex-A8` **Processing Host** (thats where the receiver-dongle is connected to).
If I use a PC as **Processing Host**, it works stable.

@cvinayak  I also think a live video conference would be a good way to continue. Can you provide me some direct contact details for easier communication?

cvinayak, 2021-08-23:
@caco3 I am available on slack with display name vich. Otherwise, my email id is available in Controller commit sign-off. Please ping me or send me an email.

caco3, 2021-08-25:
@jfischer-no , @carlescufi FYI

Thanks to the great support of  @cvinayak , we could trace it further down to a stalling in the USB subsystem.

It seems to hang at the `ret = k_sem_take(&pdata.sem, K_MSEC(USB_TRANSFER_SYNC_TIMEOUT));` in `usb_transfer_sync()` in `/zephyr/subsys/usb/usb_transfer.c`

I added `printk ` calls on various places, see the log below:

`/zephyr/subsys/usb/class/bluetooth.c`:
[image]

`/zephyr/subsys/usb/usb_transfer.c`:
[image]

HCI_USB Log:
[image]

I tested this with commit `8c31f31`, run on a nrf52840 DK board.
Below a sanitized `proj,conf` of the HCI_USB project:
```
CONFIG_STDOUT_CONSOLE=y
CONFIG_GPIO=y
CONFIG_SERIAL=y
CONFIG_UART_INTERRUPT_DRIVEN=y

CONFIG_BT=y
CONFIG_BT_HCI_RAW=y

CONFIG_USB_DEVICE_STACK=y
CONFIG_USB_DEVICE_BLUETOOTH=y
CONFIG_USB_DEVICE_BLUETOOTH_VS_H4=n

# Workaround: Unable to allocate command buffer when using K_NO_WAIT since
# Host number of completed commands does not follow normal flow control.
CONFIG_BT_BUF_CMD_TX_COUNT=10
   
CONFIG_BT_CTLR_DEBUG_PINS=y

########################################
# System
########################################
# Some command handlers require a large stack.
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=2048

########################################
# Bluetooth
########################################
CONFIG_BT=y
# - RAW HCI access
CONFIG_BT_HCI_RAW=y

# - Peripheral Role support
#CONFIG_BT_PERIPHERAL=n

# - Maximum supported HCI RX buffer length
CONFIG_BT_BUF_ACL_RX_SIZE=254
CONFIG_BT_BUF_EVT_RX_SIZE=255
CONFIG_BT_BUF_CMD_TX_SIZE=255

# Bluetooth -> Bluetooth Controller
# - Maximum supported ACL size for outgoing data
CONFIG_BT_BUF_ACL_TX_SIZE=251
CONFIG_BT_BUF_ACL_TX_COUNT=255

# - Maximum data length supported
CONFIG_BT_CTLR_DATA_LENGTH_MAX=251
# - Coded PHY
CONFIG_BT_CTLR_PHY_CODED=n
# - Tx Power
CONFIG_BT_CTLR_TX_PWR_PLUS_8=y

########################################
# USB device stack
########################################
#CONFIG_USB=y
CONFIG_USB_DEVICE_STACK=y

# USB HCI
CONFIG_USB_DEVICE_BLUETOOTH=y
CONFIG_USB_DEVICE_BLUETOOTH_VS_H4=y
```

jfischer-no, 2021-08-25:
> I am only able to provoke the issue when using the weak-powered `Cortex-A8` **Processing Host** (thats where the receiver-dongle is connected to).
> If I use a PC as **Processing Host**, it works stable.

Seems to depend on USB host controller.
Is there a possibility on 'Cortex-A8' to connect external USB host controller via PCI Express?
@cvinayak Could it be a HCI events / ACL Data out of order on some USB host controllers issue?

> It seems to hang at the `ret = k_sem_take(&pdata.sem, K_MSEC(USB_TRANSFER_SYNC_TIMEOUT));` in `usb_transfer_sync()` in `/zephyr/subsys/usb/usb_transfer.c`

I guess it timeouts in a loop, not sure why you do not see a 'p', this lines are changed by cd7461495f1aeb4561d6a7e96c9f37b029805694, before that is was just  `k_sem_take(&pdata.sem, K_FOREVER);`.
However usb_transfer_sync() would block until endpoint buffer is transferred to the host (HCI_IN_EP). The obvious reason is that the host (host application) does not request an IN transfer, so the host controller does not issue IN tokens, and usb_transfer_sync() blocks. On the host side, does the Linux Kernel complain about something related?

> I added `printk ` calls on various places, see the log below:

printk (over UART?) is too slow (synchronous if not via logging) and not suitable for tracing something in USB device support.

jfischer-no, 2021-08-25:
> CONFIG_USB_DEVICE_BLUETOOTH=y
> CONFIG_USB_DEVICE_BLUETOOTH_VS_H4=y

you do not need `CONFIG_USB_DEVICE_BLUETOOTH_VS_H4=y`

cvinayak, 2021-08-25:
> Could it be a HCI events / ACL Data out of order on some USB host controllers issue?

@jfischer-no The is not much HCI events, only ACL Data is being received as fast as possible in the scenario.

Unrelated, `pdata` is auto variable in the `usb_transfer_sync` function, what will happen if the function returns after `k_sem_take` timeout and `k_sem_give` is called by `usb_transfer`? Wouldn't it be out-of-bound access?

jfischer-no, 2021-08-30:
> How ever from then on, it always takes `100ms` until `ret = k_sem_take(&pdata.sem, K_MSEC(USB_TRANSFER_SYNC_TIMEOUT));` resp. `trans = usb_ep_get_transfer(ep);` get called again!

usb_ep_get_transfer() is there to check if the transfer is still valid and not cancelled, for example by detach from the host (see cd74614, before that is was just k_sem_take(&pdata.sem, K_FOREVER); .

> Because `D3` no longer gets toggled, I interprete that it stays in the `while` loop, how ever I do not understand where it blocks.

See my comment above:
> However usb_transfer_sync() would block until endpoint buffer is transferred to the host (HCI_IN_EP). The obvious reason is that the host (host application) does not request an IN transfer, so the [...]

Depending on what tools you are equipped with, you could listen on the bus e.g. with USB protocol analyzer, if the host is still sending IN tockens for the endpoint some time after the stall, then the problem is on Zephyr OS side. 
On the host side, does the Linux Kernel complain about something related?
Maybe it helps to run bluetoothd daemon in foreground with debuging, see https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/HACKING#n83

caco3, 2021-09-01:
I do not have an USB-Analyzer at hand, but we ordered one. How ever it will take some weeks until I will have it in my hands.

I run bluetoothd in foreground, debug mode. I can see messages while the device gets connected, but during the streaming and around the stalling not a single message is shown!

<details>
<summary>bluetoothd -n -d | tee bluetooth.log</summary>
```

# Start and connecting

Sep  1 07:03:43 wlreceiver daemon.debug bluetoothd[1076]: ../bluez-5.54/src/adapter.c:start_discovery() sender :1.11
Sep  1 07:03:43 wlreceiver daemon.debug bluetoothd[1076]: ../bluez-5.54/src/adapter.c:update_discovery_filter() 
Sep  1 07:03:43 wlreceiver daemon.debug bluetoothd[1076]: ../bluez-5.54/src/adapter.c:discovery_filter_to_mgmt_cp() 
Sep  1 07:03:43 wlreceiver daemon.debug bluetoothd[1076]: ../bluez-5.54/src/adapter.c:trigger_start_discovery() 
Sep  1 07:03:43 wlreceiver daemon.debug bluetoothd[1076]: ../bluez-5.54/src/adapter.c:cancel_passive_scanning() 
Sep  1 07:03:43 wlreceiver daemon.debug bluetoothd[1076]: ../bluez-5.54/src/adapter.c:start_discovery_timeout() 
Sep  1 07:03:43 wlreceiver daemon.debug bluetoothd[1076]: ../bluez-5.54/src/adapter.c:start_discovery_timeout() adapter->current_discovery_filter == 0
Sep  1 07:03:43 wlreceiver daemon.debug bluetoothd[1076]: ../bluez-5.54/src/adapter.c:start_discovery_complete() status 0x00
Sep  1 07:03:43 wlreceiver daemon.info btman[536]: INFO: Devices discoverying started
Sep  1 07:03:43 wlreceiver daemon.debug bluetoothd[1076]: ../bluez-5.54/src/adapter.c:discovering_callback() hci0 type 6 discovering 1 method 0
Sep  1 07:03:43 wlreceiver daemon.debug bluetoothd[1076]: ../bluez-5.54/src/adapter.c:device_found_callback() hci0 addr 00:07:29:4B:43:B4, rssi -30 flags 0x0000 eir_len 33
Sep  1 07:03:43 wlreceiver daemon.debug bluetoothd[1076]: ../bluez-5.54/src/device.c:device_set_legacy() legacy 0
Sep  1 07:03:43 wlreceiver daemon.debug bluetoothd[1076]: ../bluez-5.54/src/device.c:device_set_rssi_with_delta() rssi -30
Sep  1 07:03:43 wlreceiver daemon.debug bluetoothd[1076]: ../bluez-5.54/src/device.c:device_set_flags() flags 6
Sep  1 07:03:43 wlreceiver daemon.debug bluetoothd[1076]: ../bluez-5.54/src/adapter.c:stop_discovery() sender :1.11
Sep  1 07:03:43 wlreceiver daemon.debug bluetoothd[1076]: ../bluez-5.54/src/adapter.c:stop_discovery_complete() status 0x00
Sep  1 07:03:43 wlreceiver daemon.info btman[536]: INFO: Devices discoverying stopped
Sep  1 07:03:43 wlreceiver daemon.debug bluetoothd[1076]: ../bluez-5.54/src/adapter.c:trigger_passive_scanning() 
Sep  1 07:03:43 wlreceiver daemon.debug bluetoothd[1076]: ../bluez-5.54/src/adapter.c:discovery_remove() owner :1.11
Sep  1 07:03:43 wlreceiver daemon.debug bluetoothd[1076]: ../bluez-5.54/src/device.c:device_set_rssi_with_delta() rssi 0
Sep  1 07:03:43 wlreceiver daemon.debug bluetoothd[1076]: ../bluez-5.54/src/adapter.c:discovering_callback() hci0 type 6 discovering 0 method 0
Sep  1 07:03:44 wlreceiver daemon.debug bluetoothd[1076]: ../bluez-5.54/src/device.c:device_connect_le() Connection attempt to: 00:07:29:4B:43:B4
Sep  1 07:03:44 wlreceiver daemon.debug bluetoothd[1076]: ../bluez-5.54/src/adapter.c:connected_callback() hci0 device 00:07:29:4B:43:B4 connected eir_len 21
Sep  1 07:03:44 wlreceiver daemon.debug bluetoothd[1076]: ../bluez-5.54/attrib/gattrib.c:g_attrib_ref() 0x584568: g_attrib_ref=1 
Sep  1 07:03:44 wlreceiver daemon.debug bluetoothd[1076]: ../bluez-5.54/src/gatt-client.c:btd_gatt_client_connected() Device connected.
Sep  1 07:03:44 wlreceiver daemon.debug bluetoothd[1076]: ../bluez-5.54/src/device.c:gatt_debug() MTU exchange complete, with MTU: 247
Sep  1 07:03:45 wlreceiver daemon.debug bluetoothd[1076]: ../bluez-5.54/src/device.c:gatt_debug() Primary services found: 7
Sep  1 07:03:45 wlreceiver daemon.debug bluetoothd[1076]: ../bluez-5.54/src/device.c:gatt_debug() start: 0x0001, end: 0x0004, uuid: 00001801-0000-1000-8000-00805f9b34fb
Sep  1 07:03:45 wlreceiver daemon.debug bluetoothd[1076]: ../bluez-5.54/src/device.c:gatt_debug() start: 0x0005, end: 0x000b, uuid: 00001800-0000-1000-8000-00805f9b34fb
Sep  1 07:03:45 wlreceiver daemon.debug bluetoothd[1076]: ../bluez-5.54/src/device.c:gatt_debug() start: 0x000c, end: 0x000f, uuid: 0000180f-0000-1000-8000-00805f9b34fb
Sep  1 07:03:45 wlreceiver daemon.debug bluetoothd[1076]: ../bluez-5.54/src/device.c:gatt_debug() start: 0x0010, end: 0x001c, uuid: 0000180a-0000-1000-8000-00805f9b34fb
Sep  1 07:03:45 wlreceiver daemon.debug bluetoothd[1076]: ../bluez-5.54/src/device.c:gatt_debug() start: 0x001d, end: 0x003f, uuid: 00000100-f5bf-58d5-9d17-172177d1316a
Sep  1 07:03:45 wlreceiver daemon.debug bluetoothd[1076]: ../bluez-5.54/src/device.c:gatt_debug() start: 0x0040, end: 0x0056, uuid: ffffff00-f5bf-58d5-9d17-172177d1316a
Sep  1 07:03:45 wlreceiver daemon.debug bluetoothd[1076]: ../bluez-5.54/src/device.c:gatt_debug() start: 0x0057, end: 0x005a, uuid: 8d53dc1d-1db7-4cd3-868b-8a527460a4
Sep  1 07:03:45 wlreceiver daemon.debug bluetoothd[1076]: ../bluez-5.54/src/device.c:gatt_debug() Characteristics found: 37
Sep  1 07:03:45 wlreceiver daemon.debug bluetoothd[1076]: ../bluez-5.54/src/device.c:gatt_debug() start: 0x0002, end: 0x0005, value: 0x0003, props: 0x20, uuid: 00002a05-0000-1
Sep  1 07:03:45 wlreceiver daemon.debug bluetoothd[1076]: ../bluez-5.54/src/device.c:gatt_debug() start: 0x0006, end: 0x0007, value: 0x0007, props: 0x0a, uuid: 00002a00-0000-1
[... cut ...]
Sep  1 07:04:17 wlreceiver daemon.debug bluetoothd[1076]: ../bluez-5.54/src/gatt-client.c:unregister_characteristic() Removing GATT characteristic: /org/bluez/hci0/dev_00_07_29_4B_43_B4/service0057/char0058
Sep  1 07:04:17 wlreceiver daemon.debug bluetoothd[1076]: ../bluez-5.54/src/gatt-client.c:unregister_descriptor() Removing GATT descriptor: /org/bluez/hci0/dev_00_07_29_4B_43_B4/service0057/char0058/desc005a
Sep  1 07:04:17 wlreceiver daemon.debug bluetoothd[1076]: ../bluez-5.54/src/gatt-database.c:btd_gatt_database_att_disconnected() 
Sep  1 07:04:17 wlreceiver daemon.debug bluetoothd[1076]: ../bluez-5.54/attrib/gattrib.c:g_attrib_unref() 0x584568: g_attrib_unref=0 
```
</details>

carlescufi, 2021-09-02:
>We have an application based on a nrf52840. We see k_sem_take failed with err -11 assertion when we stress the device with a lot of notifications and read requests at the same time.

To add information to this issue for future reference:
- The embedded side (peripheral) sends data at high throughput and the central fails to receive it due to the USB issue currently being described
- This turns into a stall in the BLE connection, with both sides NACKing continuously
- That causes no HCI Number of Completed Packets to be sent, blocking the TX thread on waiting for those num complete packets that will never come, thus reproducing the scenario in https://github.com/zephyrproject-rtos/zephyr/issues/25917
- If the issue on the central side is resolved, then the scenario in #25917 will not be triggered

caco3, 2021-09-03:
👍 

Just for clarification:
>     The embedded side (peripheral) sends data at high throughput and the central fails to receive it due to the USB issue currently being described

The stalling issue only occures, once I start to do excessive GATT reads from the central to the peripheral while the peripheral still transmitts data at the high throughput

carlescufi, 2021-09-08:
> I do not have an USB-Analyzer at hand, but we ordered one

Can you please attach the trace once you have it?
Also, could you please add more information about the SBC/Linux board that shows the symptoms? SoC, kernel version, bluez version USB controller, etc. @jfischer-no suspects the Host USB controller to be at fault, so it'd be good to rule that out if it's not the case.

caco3, 2021-09-14:
Sorry for the late response.

Below some information about the system:

```
> cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 2 (v7l)
BogoMIPS        : 597.60
Features        : half thumb fastmult vfp edsp thumbee neon vfpv3 tls vfpd32 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x3
CPU part        : 0xc08
CPU revision    : 2
Hardware        : Generic AM33XX (Flattened Device Tree)
Revision        : 0000
Serial          : 1
```
Our system is based on a [AM335x](https://www.ti.com/lit/gpn/AM3359). The USB-Controller is built in.
Also there is no other USB device connected to the system:
```
> lsusb
Bus 002 Device 006: ID 1545:0001 *** *** BLE Adapter     < This is our BLE receiver
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

I went through the AM335x [Errata](https://www.ti.com/lit/pdf/sprz360) and found one issue on page 18 that might be related:
 * Advisory 1.0.11 USB: Attached Non-compliant USB Device that Responds to Spurious Invalid Short Packet May Lock Up Bus

I hope to be able to provide some USB traces in the next 1..2 weeks.

jfischer-no, 2021-09-14:
> Our system is based on a [AM335x](https://www.ti.com/lit/gpn/AM3359). The USB-Controller is built in.

I have an AM335x SBC somewhere. Can you please tell who is the manufacturer of your SoM/SBC,  gladly privately if it is confidential.

caco3, 2021-09-15:
The module is designed by us, thus very specific and proprietary.

carlescufi, 2021-09-15:
Downgrading this from Medium to Low since it now looks it could be a USB Host issue on the Linux SBC side. The actual Bluetooth bug that started this issue is already covered by #25917 already, and is listed as a known issue for now, with a workaround in #26057. Discussion can continue here to try and narrow down whether the USB issue can be attributed exclusively to the Linux side or whether there is potentially an issue on the Zephyr side.

caco3, 2021-09-29:
@jfischer-no Sorry for the late response 😒 . I finally got the logic analyzer. How ever I had to realize it only supports USB 1.1 but the BLE dongle runs with USB 2.0. So at the moment, I am not able to provide a logic analyzer trace.

Where you able to investigate something with your board?

jfischer-no, 2021-10-04:
> Where you able to investigate something with your board?

No, TBH it would take a lot of time which I do not currently have.

carlescufi, 2021-10-06:
@caco3 it would be good if you could narrow down the problem to find out whether this can indeed be Zephyr related or not. At the moment this doesn't look like it, so I would like to close the issue if possible.

caco3, 2021-10-06:
The only way I see is to get an USB 2.0 sniffer. Sadly we don't have one and this issue atm. is not very high on the priority list on our side. 😒 
Because of this I agree to close it for now.

closed 2021-10-06 by caco3