zephyrproject-rtos/zephyr #34662: many udp networking cases fails on nxp platforms

hakehuang, 2021-04-28:
**Describe the bug**
many net work cases fails on nxp platforms

tests/net/udp
tests/net/dhcpv4
tests/net/icmpv4
tests/net/ethernet_mgmt

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=mimxrt1050_evk / mimxrt1060_evk / mimxrt1064_evk / frdmk64f 
3. make
4. make flash

**Expected behavior**
test pass

**Impact**
udp related protocol test features

**Logs and console output**

- test_udp

```
  W: You have 1 IPv6 net_if addresses but 2 network interfaces
    W: Consider increasing CONFIG_NET_IF_MAX_IPV6_COUNT value.
    W: You have 1 IPv4 net_if addresses but 2 network interfaces
    W: Consider increasing CONFIG_NET_IF_MAX_IPV4_COUNT value.
    *** Booting Zephyr OS build zephyr-v2.5.0-2637-ga3469a049785  ***
    Running test suite test_udp_fn
    ===================================================================
    START - test_udp
    Cannot add 2001:db8::1 to interface 0x80000e40
        Assertion failed at WEST_TOPDIR/zephyr/tests/net/udp/src/main.c:484: test_udp: (0 is false)
    exiting
     FAIL - test_udp in 0.14 seconds
    ===================================================================
    Test suite test_udp_fn failed.
    ===================================================================
    PROJECT EXECUTION FAILED
```

- test_dhcp

```
W: You have 1 IPv4 net_if addresses but 2 network interfaces

W: Consider increasing CONFIG_NET_IF_MAX_IPV4_COUNT value.

*** Booting Zephyr OS build zephyr-v2.5.0-2637-ga3469a049785  ***

Running test suite test_dhcpv4

===================================================================

START - test_dhcp
W: iface 0x20000a00 is down
I: ETH_0 enabled 100M full-duplex mode.
    Assertion failed at WEST_TOPDIR/zephyr/tests/net/dhcpv4/src/main.c:443: test_dhcp: (false is false)
Timeout while waiting
 FAIL - test_dhcp in 3.12 seconds
===================================================================
Test suite test_dhcpv4 failed.
===================================================================
PROJECT EXECUTION FAILED
```

- ethernet_mgmt_test

```
    W: You have 1 IPv6 net_if addresses but 2 network interfaces

    W: Consider increasing CONFIG_NET_IF_MAX_IPV6_COUNT value.

    W: You have 1 IPv4 net_if addresses but 2 network interfaces

    W: Consider increasing CONFIG_NET_IF_MAX_IPV4_COUNT value.

    *** Booting Zephyr OS build zephyr-v2.5.0-2637-ga3469a049785  ***

    Running test suite ethernet_mgmt_test

    ===================================================================

    START - test_change_mac_when_up

    

        Assertion failed at WEST_TOPDIR/zephyr/tests/net/ethernet_mgmt/src/main.c:319: test_change_mac_when_up: (ret equal to 0)

    [******** Truncated '23' Lines Log ********]

    ===================================================================

    START - test_change_unsupported_link

     PASS - test_change_unsupported_link in 0.1 seconds

    ===================================================================

    START - test_change_duplex

    

        Assertion failed at WEST_TOPDIR/zephyr/tests/net/ethernet_mgmt/src/main.c:428: test_change_duplex: (ret not equal to 0)

    invalid duplex change

     FAIL - test_change_duplex in 0.14 seconds

    ===================================================================

    START - test_change_same_duplex

     PASS - test_change_same_duplex in 0.1 seconds

    ===================================================================

    START - test_change_qav_params

    

        Assertion failed at WEST_TOPDIR/zephyr/tests/net/ethernet_mgmt/src/main.c:460: test_change_qav_params: (ret not equal to 0)

    could not get the number of priority queues

     FAIL - test_change_qav_params in 0.16 seconds

    ===================================================================

    START - test_change_promisc_mode_on

    

        Assertion failed at WEST_TOPDIR/zephyr/tests/net/ethernet_mgmt/src/main.c:632: test_change_promisc_mode: (ret not equal to 0)

    invalid promisc mode change

     FAIL - test_change_promisc_mode_on in 0.15 seconds

    ===================================================================

    START - test_change_to_same_promisc_mode

    

        Assertion failed at WEST_TOPDIR/zephyr/tests/net/ethernet_mgmt/src/main.c:656: test_change_to_same_promisc_mode: (ret not equal to -EALREADY)

    invalid change to already set promisc mode

     FAIL - test_change_to_same_promisc_mode in 0.17 seconds

    ===================================================================

    START - test_change_promisc_mode_off

    

        Assertion failed at WEST_TOPDIR/zephyr/tests/net/ethernet_mgmt/src/main.c:632: test_change_promisc_mode: (ret not equal to 0)

    invalid promisc mode change

     FAIL - test_change_promisc_mode_off in 0.15 seconds

    ===================================================================

    Test suite ethernet_mgmt_test failed.

    ===================================================================

    PROJECT EXECUTION FAILED
```

- test_icmpv4_fn

```

 W: You have 1 IPv4 net_if addresses but 2 network interfaces

    W: Consider increasing CONFIG_NET_IF_MAX_IPV4_COUNT value.

    *** Booting Zephyr OS build zephyr-v2.5.0-2637-ga3469a049785  ***

    Running test suite test_icmpv4_fn

    ===================================================================

    START - test_icmpv4

     PASS - test_icmpv4 in 0.1 seconds

    ===================================================================

    START - test_icmpv4_send_echo_req

    W: iface 0x80000988 is down

    

        Assertion failed at WEST_TOPDIR/zephyr/tests/net/icmpv4/src/main.c:445: test_icmpv4_send_echo_req: (false is false)

    Failed to send

     FAIL - test_icmpv4_send_echo_req in 0.15 seconds

    ===================================================================

    START - test_icmpv4_send_echo_rep

     PASS - test_icmpv4_send_echo_rep in 0.1 seconds

    ===================================================================

    START - test_icmpv4_send_echo_req_opt

    W: iface 0x80000988 is down

    

        Assertion failed at WEST_TOPDIR/zephyr/tests/net/icmpv4/src/main.c:481: test_icmpv4_send_echo_req_opt: (false is false)

    Failed to send

     FAIL - test_icmpv4_send_echo_req_opt in 0.15 seconds

    ===================================================================

    START - test_icmpv4_send_echo_req_bad_opt

     PASS - test_icmpv4_send_echo_req_bad_opt in 0.1 seconds

    ===================================================================

    Test suite test_icmpv4_fn failed.

    ===================================================================

    PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.

hakehuang, 2021-04-28:
@MaureenHelm , @dleach02

jukkar, 2021-04-28:
Shoudl be fixed by #34655, please try that fix and report back.